siemens/meta-iot2050 #1: mraa do_dpkg_build error on building example image

davimaf, 2020-05-12:
**System:**
Win10 + Virtual Machine VirtualBox 6.0.8 r130520
**VM:**
Kernel: 5.3.0-51-generic x86_64 bits: 64
Desktop: Gnome 3.28.4 Distro: Ubuntu 18.04.4 LTS
The machine is the same used for building IOT2040 images.

I'm trying to build the example image as specified in README:
`./kas-docker --isar build kas-iot2050-example.yml`
but I cannot generate the image only because I get one error on `mraa do_dpkg_build`. Below the building result log lines:

```
ERROR: Logfile of failure stored in: /work/build/tmp/work/isar-arm64/mraa/2.0-r0/temp/log.do_dpkg_build.2094
ERROR: Task (/repo/recipes-app/mraa/mraa.bb:do_dpkg_build) failed with exit code '1'
NOTE: Tasks Summary: Attempted 200 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
/repo/recipes-app/mraa/mraa.bb:do_dpkg_build
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

Analyzing the logfile ` log.do_dpkg_build.2094` seems that the error starts from Java compilation. 
Below the last lines from the logfile:

```
Compiling java..
cd /home/builder/mraa/git/obj-aarch64-linux-gnu/src/java && cmake -E make_directory /home/builder/mraa/git/obj-aarch64-linux-gnu/src/java/mraa
cd /home/builder/mraa/git/obj-aarch64-linux-gnu/src/java && /usr/lib/jvm/default-java/bin/javac *.java -d /home/builder/mraa/git/obj-aarch64-linux-gnu/src/java
/bin/sh: 1: /usr/lib/jvm/default-java/bin/javac: not found
make[3]: *** [src/java/CMakeFiles/mraajava.dir/build.make:107: src/java/libmraajava.so] Error 127
make[3]: *** Deleting file 'src/java/libmraajava.so'
make[3]: Leaving directory '/home/builder/mraa/git/obj-aarch64-linux-gnu'
make[2]: *** [CMakeFiles/Makefile2:473: src/java/CMakeFiles/mraajava.dir/all] Error 2
make[2]: Leaving directory '/home/builder/mraa/git/obj-aarch64-linux-gnu'
make[1]: *** [Makefile:144: all] Error 2
make[1]: Leaving directory '/home/builder/mraa/git/obj-aarch64-linux-gnu'
dh_auto_build: cd obj-aarch64-linux-gnu && make -j1 PKG_CONFIG=aarch64-linux-gnu-pkg-config CXX=aarch64-linux-gnu-g\+\+ CC=aarch64-linux-gnu-gcc returned exit code 2
make: *** [debian/rules:9: build] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
WARNING: exit code 2 from a shell command.
```
What can I do?
Thank you

Attached there is the **complete logfile** of the failure:

```
DEBUG: Executing python function do_dpkg_build
DEBUG: Executing shell function dpkg_do_mounts
DEBUG: Shell function dpkg_do_mounts finished
DEBUG: Executing shell function dpkg_runbuild
patching file javascript/v8/javascriptcode.swg
patching file javascript/v8/javascripthelpers.swg
patching file javascript/v8/javascriptinit.swg
patching file javascript/v8/javascriptrun.swg
dpkg-buildpackage: warning:     debian/changelog(l5): badly formatted trailer line
LINE:  -- le.jin@siemens.com  Mon, 11 May 2020 10:23:58 +0000
dpkg-buildpackage: warning:     debian/changelog(l5): found end of file where expected more change data or trailer
dpkg-buildpackage: info: source package mraa
dpkg-buildpackage: info: source version 2.0
dpkg-buildpackage: info: source distribution UNRELEASED
dpkg-buildpackage: info: source changed by 
dpkg-architecture: warning: specified GNU system type aarch64-linux-gnu does not match CC system type x86_64-linux-gnu, try setting a correct CC environment variable
 dpkg-source -I --before-build .
dpkg-buildpackage: info: host architecture arm64
dpkg-source: warning: git/debian/changelog(l5): badly formatted trailer line
LINE:  -- le.jin@siemens.com  Mon, 11 May 2020 10:23:58 +0000
dpkg-source: warning: git/debian/changelog(l5): found end of file where expected more change data or trailer
 fakeroot debian/rules clean
dh clean
[... 1668 lines cut ...]
/home/builder/mraa/git/obj-aarch64-linux-gnu/src/java/CMakeFiles/mraajava.dir/mraajavaJAVA_wrap.cxx:6417:11: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   arg1 = *(mraa::Gpio **)&jarg1;
           ^~~~~~~~~~~~~~~~~~~~~
/home/builder/mraa/git/obj-aarch64-linux-gnu/src/java/CMakeFiles/mraajava.dir/mraajavaJAVA_wrap.cxx: In function 'jint Java_mraa_mraaJNI_Gpio_1isrExit(JNIEnv*, jclass, jlong, jobject)':
/home/builder/mraa/git/obj-aarch64-linux-gnu/src/java/CMakeFiles/mraajava.dir/mraajavaJAVA_wrap.cxx:6446:11: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   arg1 = *(mraa::Gpio **)&jarg1;
           ^~~~~~~~~~~~~~~~~~~~~
/home/builder/mraa/git/obj-aarch64-linux-gnu/src/java/CMakeFiles/mraajava.dir/mraajavaJAVA_wrap.cxx: In function 'jint Java_mraa_mraaJNI_Gpio_1mode(JNIEnv*, jclass, jlong, jobject, jint)':
/home/builder/mraa/git/obj-aarch64-linux-gnu/src/java/CMakeFiles/mraajava.dir/mraajavaJAVA_wrap.cxx:6474:11: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   arg1 = *(mraa::Gpio **)&jarg1;
           ^~~~~~~~~~~~~~~~~~~~~
/home/builder/mraa/git/obj-aarch64-linux-gnu/src/java/CMakeFiles/mraajava.dir/mraajavaJAVA_wrap.cxx: In function 'jint Java_mraa_mraaJNI_Gpio_1dir(JNIEnv*, jclass, jlong, jobject, jint)':
/home/builder/mraa/git/obj-aarch64-linux-gnu/src/java/CMakeFiles/mraajava.dir/mraajavaJAVA_wrap.cxx:6503:11: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   arg1 = *(mraa::Gpio **)&jarg1;
           ^~~~~~~~~~~~~~~~~~~~~
/home/builder/mraa/git/obj-aarch64-linux-gnu/src/java/CMakeFiles/mraajava.dir/mraajavaJAVA_wrap.cxx: In function 'jint Java_mraa_mraaJNI_Gpio_1readDir(JNIEnv*, jclass, jlong, jobject)':
/home/builder/mraa/git/obj-aarch64-linux-gnu/src/java/CMakeFiles/mraajava.dir/mraajavaJAVA_wrap.cxx:6531:11: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   arg1 = *(mraa::Gpio **)&jarg1;
           ^~~~~~~~~~~~~~~~~~~~~
/home/builder/mraa/git/obj-aarch64-linux-gnu/src/java/CMakeFiles/mraajava.dir/mraajavaJAVA_wrap.cxx: In function 'jint Java_mraa_mraaJNI_Gpio_1read(JNIEnv*, jclass, jlong, jobject)':
/home/builder/mraa/git/obj-aarch64-linux-gnu/src/java/CMakeFiles/mraajava.dir/mraajavaJAVA_wrap.cxx:6558:11: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   arg1 = *(mraa::Gpio **)&jarg1;
           ^~~~~~~~~~~~~~~~~~~~~
/home/builder/mraa/git/obj-aarch64-linux-gnu/src/java/CMakeFiles/mraajava.dir/mraajavaJAVA_wrap.cxx: In function 'jint Java_mraa_mraaJNI_Gpio_1write(JNIEnv*, jclass, jlong, jobject, jint)':
/home/builder/m

jan-kiszka, 2020-05-12:
Please also provide the console log where kas-docker was running. And probably also `build/tmp/work/isar-arm64/mraa/2.0-r0/temp/log.do_install_builddeps` because this looks like something was missed when installing the build deps. I find that javac binary here.

davimaf, 2020-05-12:
> Please also provide the console log where kas-docker was running. And probably also `build/tmp/work/isar-arm64/mraa/2.0-r0/temp/log.do_install_builddeps` because this looks like something was missed when installing the build deps. I find that javac binary here.

Hi jan-kiska,
Attached there are the kas docker console output and the specific builddeps log files.
thank you.

[kas-docker.txt](https://github.com/siemens/meta-iot2050/files/4618126/kas-docker.txt)
[log.do_install_builddeps.txt](https://github.com/siemens/meta-iot2050/files/4618129/log.do_install_builddeps.txt)

jan-kiszka, 2020-05-12:
No luck in understanding or even reproducing yet. I just ran
```
./kas-docker --iser build kas-iot2050-example.yml --target mraa
```
and that worked.

But my Ubuntu VM is a bit older (18.04.2), and so is the docker.io (18.09.7).  Will have to refresh that setup, but that may take some days.

Last thing to compare upfront: I was giving 4G RAM to the VM (and 8 cores, but that should not make the difference). How much does yours have?

bergmanu, 2020-05-13:
I can confirm that 
`./kas-docker --isar build kas-iot2050-example.yml`
works on my machine: 
Ubuntu: 18.04.4
Kernel: 5.3.0-51
docker: 19.03.8

My VM has 4GB of RAM and two cores.

jan-kiszka, 2020-05-13:
We likely need a reference description for the setup of such a system.

Still, I'd like to understand eventually where this issue comes from. Apparently, it was not only seen once.

jan-kiszka, 2020-05-13:
Another factor to specify, besides recommended RAM size, is free storage size. Both values should also go into our README.

bergmanu, 2020-05-13:
I'll try to setup a VM from scratch with Ubuntu 20.04 to see what steps are needed to setup such a system. 
For now I was using an already prepared VM for docker.

davimaf, 2020-05-13:
The machine with problems was:
System:
Win10 + Virtual Machine VirtualBox 6.0.8 r130520
VM:
Kernel: 5.3.0-51-generic x86_64 bits: 64
Desktop: Gnome 3.28.4 Distro: Ubuntu 18.04.4 LTS
CPU(s):  4 core of Intel Xeon IPC- 8GB RAM
Docker version 19.03.6
The machine is the same used for building IOT2040 images.

I've tried with another VM on VMWare on my Notebook:
Host: ubuntu Kernel: 4.15.0-47-generic x86_64 
Distro: Ubuntu 16.04 xenial
CPU(s):  2 core of Intel Core i7-6820EQs  - 8GB RAM
Docker version 18.09.6

And its working. I'm able to generate example image correctly.

jan-kiszka, 2020-05-18:
I suppose we can close the issue then.

FWIW, I've scheduled some safety measure for kas-docker (https://github.com/siemens/kas/commit/385e42ff11beb505df20acd0d2238814a5522687) to detect and reject execution as root so that at least one source of trouble will be eliminated in the future.

tammerOfTheGods, 2020-06-01:
I ran into this same issue (/usr/lib/jvm/default-java/bin/javac: not found).  It was corrected by building on a debian 10 VB as opposed to Ubuntu 18 or by upgrading the MRAA patch to use python 3 exclusively (remove references to python 2.6).

jan-kiszka, 2020-06-01:
Do you mean you dropped `libpython-dev` from `DEBIAN_BUILD_DEPENDS` in recipes-app/mraa/mraa.bb? Or what exactly did you do? How did you find out what was causing the problem for you?

tammerOfTheGods, 2020-06-01:
Yes... initially I tried pulling a more recent version of mraa but the problem remained and was during the cmake.  IIRC I did the cmake locally to resolve the build problems.  This diff is based on a different version of mraa.bb but indicates the different DEBIAN_BUILD_DEPENDS (also turned off the cmake tests).  Didn't get a change to test it as we determined that it would build on debian 10.
  
17,20c17,23
<             file://0002-feat-iot2050-add-iot2050-platform-support.patch \
<             file://0003-feat-iot2050-add-some-example-code-for-testing.patch \
<             file://rules"
< SRC_URI[sha256sum] = "15783b4c4431a36d44ba95daf134318a04ff44a8190ba3f19abbda89ede35a26"
---
>             file://0002-feat-iot2050-add-iot2050-platform-support.patch  \
>             file://0003-feat-iot2050-add-some-example-code-for-testing.patch "
> 
> #
> #           file://rules"
> #
> #SRC_URI[sha256sum] = "15783b4c4431a36d44ba95daf134318a04ff44a8190ba3f19abbda89ede35a26"
22c25,26
< MRAA_REV = "967585c9ea0e1a8818d2172d2395d8502f6180a2"
---
> #MRAA_REV = "967585c9ea0e1a8818d2172d2395d8502f6180a2"
> MRAA_REV = "e15ce6fbc76148ba8835adc92196b0d0a3f245e7"
30,39d33
< DEBIAN_BUILD_DEPENDS = " \
<     cmake, \
<     swig3.0, \
<     libpython-dev, \
<     libpython3-dev, \
<     nodejs, \
<     libnode-dev, \
<     libjson-c-dev, \
<     default-jdk"
< 
40a35
> DEB_BUILD_GNU_TYPE = " aarch64-linux-gnu "
45a41,43
>     sed -i -e 's/Build-Depends: /Build-Depends: cmake, swig3.0, libpython3-dev, nodejs, libnode-dev, libjson-c-dev, openjdk-11-jdk, /g' ${S}/debian/control
>     sed -i -e 's/option (BUILDTESTS "Override the addition of tests" ON)/option (BUILDTESTS "Override the addition of tests" OFF)/g' ${S}/CMakeLists.txt

jan-kiszka, 2020-06-02:
It's a bit tricky to find the root change here. My feeling is that one of your chances disabled the java binding support and, thus, no longer needs the javac during build. That would obviously also avoid the issue. Or can you rule that out?

tammerOfTheGods, 2020-06-02:
javac is required for the cmake.  I think that the default javac is not set up correctly with libpython-dev,  libpython3-dev and default-jdk on some platforms.  Removing libpython-dev and using the specific openjdk-11-jdk (as opposed to default-jdk) appears to set up javac in a more consistent manner across platforms.

jan-kiszka, 2020-06-02:
cmake (with the mraa-provided setup) probes for java, it does not require it. You can perfectly build mraa without any java installed. However, our recipe enforces it (`-DBUILDSWIGJAVA=ON`).

applea9, 2020-08-24:
I also ran into the same issue, my machine is 
Ubuntu: 16.04
Kernel: 4.15.0-112
docker: 18.09.7

Actually I have compiled it successfully before on same machine, but when I cleaned the cache and try to rebuild, this issue occured.

applea9, 2020-08-26:
I changed -DBUILDSWIGJAVA=ON to -DBUILDSWIGJAVA=OFF, the issue is solved.

jan-kiszka, 2020-08-26:
I think I found the solution:
```
diff --git a/recipes-app/mraa/mraa.bb b/recipes-app/mraa/mraa.bb
index 0c2f39c..03921dd 100644
--- a/recipes-app/mraa/mraa.bb
+++ b/recipes-app/mraa/mraa.bb
@@ -35,7 +35,7 @@ DEBIAN_BUILD_DEPENDS = " \
     nodejs, \
     libnode-dev, \
     libjson-c-dev, \
-    default-jdk"
+    default-jdk:native"
 
 DEBIAN_DEPENDS = "python3, nodejs "
 
```
Could you retry with that change and java enabled?

applea9, 2020-08-26:
With configuration of default-jdk:native and -DBUILDSWIGJAVA=ON, the image is successfully compiled.

What is difference between default-jdk and default-jdk:native?

jan-kiszka, 2020-08-26:
From the commit log if the change:

    Otherwise, also the arm64 jdk is installed, and
    /usr/lib/jvm/default-java is set up to point to it. That will make the
    build reference the arm64 javac which may work on some systems
    inheriting binfmt-misc settings from the host, but not on all.

commit 244732d8 (2020-09-01): mraa: Build-depend on native jdk

Otherwise, also the arm64 jdk is installed, and
/usr/lib/jvm/default-java is set up to point to it. That will make the
build reference the arm64 javac which may work on some systems
inheriting binfmt-misc settings from the host, but not on all.

Fixes #1.
closed 2020-09-01 by johnson325