siemens/meta-iot2050 #11: Shutdown Panic: k3-system-off patch

tammerOfTheGods, 2020-06-02:
Migrated issue #103 from c.s.c.  Please advise when the following patch is picked up in this repo...

https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/4110

jan-kiszka, 2020-06-05:
Fix merged upstream: https://github.com/ARM-software/arm-trusted-firmware/commit/42d9b3aaf473a1e01475a020644b504ecc3d4443. We will pick it up when moving the firmware to https://github.com/siemens/meta-iot2050/commits/jan/upstream-u-boot.

tammerOfTheGods, 2020-08-27:
Any word on the firmware upgrade schedule?

jan-kiszka, 2020-08-27:
Would love to be concrete, but we are currently struggling with sysfw ABI changes (breakages) that were just introduced and need to find a reasonable upgrade path through that. Ideally a path that avoids having to bump firmware and image at the same time. Discussions with TI ongoing, but that naturally impacts the schedule.

Ben-WD, 2021-02-05:
@jan-kiszka Any news here? Were also running an application with some kind of watchdog that reboots the device in case the server is not reachable. Unfortunately, the reboot never works which is really a problem.

jan-kiszka, 2021-02-05:
Are you on a Basic or Advanced? For the former, you could already replace all parts needed to get HW watchdog support using ´:kas/opt/upstream.yml` builds. But system tests are ongoing ATM, regressions not excluded.

Ben-WD, 2021-02-05:
I'm using the Basic devices, the first device was installed for production use this week. This there any ETA for a release that contains the upstream fixes? I need a stable image so that device still boots up successfully after a reboot. I noticed that sometimes after the reboot eth0 is not recognized. Have to disconnect the device from power, wait a few minutes to make it work again.

jan-kiszka, 2021-02-05:
I have no date to share yet, specifically as the tests only started and nasty integration issues can always delay the schedule. Specifically the tricky transition phase - first update the base image to a version that works with both old and new firmware, then update the firmware - requires careful testing and create a larger test matrix.

Regarding that Ethernet issue: Please open a separate ticket, describing as much details as you have about it (e.g. kernel console logs when the error is present). TIA.

Ben-WD, 2021-02-16:
I tried to boot an image that contains the upstream changes, but the devices (basic and advanced) fail to boot. The state led turns red and blinks a few seconds after the device has been connected to power.

jan-kiszka, 2021-02-16:
Do you have debug console logs?

Ben-WD, 2021-02-17:
ATM I cannot provide debug logs but  I think I'll order an USB UART cable later today.

The firmware was not updated, at least I didn't do anything to update it. Can you point me into the right direction?

jan-kiszka, 2021-02-17:
See https://github.com/siemens/meta-iot2050/blob/master/recipes-bsp/u-boot/README.md. Append `:kas/opt/upstream.yml` there as well.

But note the big-fat warning: If the device should no longer boot afterwards, a flash programmer will be needed to recover.

kumajaya, 2021-04-25:
Is there an alternative flash programmer besides Dediprog SF100 or SF600? Something like widely available CH341A + flashrom like https://github.com/Ragebone/nr64_supermicro_x10sle-f/blob/master/firmware_flashing.md

fr0st61te, 2021-12-24:
> Is there an alternative flash programmer besides Dediprog SF100 or SF600? Something like widely available CH341A + flashrom like https://github.com/Ragebone/nr64_supermicro_x10sle-f/blob/master/firmware_flashing.md

It seems like an off-topic. Yes, it's possible to use ch341a for flashing with flashrom.

BaochengSu, 2022-01-17:
> It seems like an off-topic. Yes, it's possible to use ch341a for flashing with flashrom.

#235

jan-kiszka, 2022-01-17:
BTW, I think we can close this issue, now that the new firmware version is available.

closed 2022-01-17 by BaochengSu