siemens/meta-iot2050 #13: Add kernel-headers to example Image

gramss, 2020-06-07:
Hey,

would it be possible to add the kernel headers to the example image?
I was not able to install a kernel module that is depended on these packages:

 ```
The following packages have unmet dependencies:
 gasket-dkms : Depends: linux-headers-686-pae but it is not installable or
                        linux-headers-amd64 but it is not installable or
                        linux-headers-generic but it is not installable or
                        linux-headers but it is not installable
```

Installing linux-headers-arm64 did not help.
Also installing via 

```
apt install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-4.19.59
E: Couldn't find any package by glob 'linux-headers-4.19.59'
E: Couldn't find any package by regex 'linux-headers-4.19.59'
```
did also not help as packages for these kernels are currently not released..?
As the kernels (RT and non-RT) are directly included into this git, I assume that I would not be able to generate them through interacting with `kas`?

I found some similar "issues" around raspbian and they seemed to release these kernel-headers through a custom package.

Appreciate your help for my thesis.. :)

jan-kiszka, 2020-06-08:
If you want to build a module on the device, you need linux-headers-iot2050[-rt] which is generated during build but only for the build environment. You have now two options:
- hook up your module with the a custom image build (see https://github.com/ilbers/isar/tree/master/meta-isar/recipes-kernel/example-module for an example)
- add linux-headers-iot2050 to the image (`IMAGE_INSTALL += "linux-headers-${KERNEL_NAME}"` in https://github.com/siemens/meta-iot2050/blob/master/recipes-core/images/iot2050-image-example.bb) AND switch off cross-building, at least for the kernel (will be lengthy)

The latter is needed due to an open issue with the Isar build system: We only build a package for one architecture, and that can be either the building machine or the target.

jan-kiszka, 2020-06-08:
You need to build the driver, that is for sure. Therefore, you need a customization. But I've missed another option: Add the headers package to our SDK. Then you can build the module cross on a Linux machine. Checking...

The reason for not installing the headers package on the target image is simply the aforementioned complication. That may eventually be resolved, though.

A general note: Installing out-of-tree or even closed-source drivers is OK for experiments but you need to be aware of the (generally) negative impact on quality and possible legal complications when doing that for production use. That's why we are working on cleanup up our own work for upstream as well. If you have the choice, pick an upstream-support hardware.

gramss, 2020-06-08:
I‘m not 100% sure we were on the same page.

I have the latest example image up and running on my iot2050. Now I want to install a kernel module through an `apt install`.
I‘m was pretty sure that the deb packages containing the new kernel module was just installing it. But thinking about it. I need the headers so the desired kernel module can be build with the custom kernel currently on my board.
Okay.. I got it.
But I still want to integrate the headers in the target-image.
Can you point me out to exactly where I do need to turn off the ISAR_CROSS_COMPILING in this repo / .bb file?
Adding the flag for the kernel-headers is not the problem. But I do not yet know how to interact with ISAR properly.
I‘m totally fine with letting it compile a few hours..

But ultimately when you figured a way out to add it to the sdk it would be great to make it available as an apt installable deb package. Not for now but the future..

I can also wait till you figured the SDK way out. If this is working, I should be able to manually copy them into my target image for now..? This would be totally fine for me as well :)

Thanks for the help!

jan-kiszka, 2020-06-08:
Kernel modules are typically not just installed by "apt install". They are build along with the kernel or via Isar-specific recipes along with the image. In regular Debian, there is nothing like a module package.

I've just pushed https://github.com/siemens/meta-iot2050/commits/jan/sdk-improvements with the SDK enhancement to build modules outside of the regular build. The topmost commit contains a brief how-to. Then you can manually copy the module over and load it.

gramss, 2020-06-08:
Okay, that is very interesting for me. Probably, I just happen to know the prototype site of kernel modules. I learned to build basic modules on a beagleboard right on the target system with linux-headers enabled. After building them we "installed" them via `insmod` (at least that is my rough memory of this).
And also from my few maker projects on raspis, but also on other IPCs running mostly Ubuntu, it quite often happened that I needed to install kernel modules through `apt install`. Mostly they involved some device drivers like USB->CAN or GPU cards (via PCIe). With the beagleboard we interacted directly with gpios and SPI/I2C etc. Only in another project I needed to rebuild the whole kernel with some extra enabled upstream modules (terminal based gui) for some custom touch display drivers.
(As I worked now a bit with the SDK I remembered my Prof saying, that she wasn't able to provide us with a cross-build setup for our machines so we needed to develop inside the target-os directly on the BeagleBoard Black. So I just missed the opportunity to learn it the right way..)

So, I just build the image and the latest SDK. My first approach using the SDK was to just follow the apt way of creating this module. But this failed as the chroot does not have apt properly setup (cannot reach the deb.debian.org servers) and might be stripped pretty hefty to just include the buildchain-tools (I guess so?). (no ping, wget, curl,.. etc available in chroot)
Okay, so I tried downloading the source of the module I want to install: https://coral.googlesource.com/gasket-dkms-debian
I cloned it into my host, mounted it into my chroot and tried a `make` inside the src-folder of the git. But this failed as this is not a full MakeFile..?
It seems to me that this is a sort of patch of an existing kernel module (gasket). I'm coming here sadly to the end of my knowledge.
This is the original documentation I wanted to follow: https://coral.ai/docs/m2/get-started#1-install-the-pcie-driver

I refrained from manually copying the header files into my target-os as I think I won't get any further as this guy with another board but a very similar problem: https://e2e.ti.com/support/legacy_forums/embedded/linux/f/354/t/393523

And from some other github issues I saw that the apt-get way is issuing a make "under the hood" to compile the module as you suggested. Thank you for that learning curve! :)

gramss, 2020-06-08:
So, I managed to get the linux-headers-iot2050 to be installed on my IOT2050.
I added the `IMAGE_INSTALL += "linux-headers-${KERNEL_NAME}"` to an already build image, crossed my fingers and got it out correctly. Atleast, the image boots and lists the package installed via `apt list --installed | grep linux-headers`.

But sadly this does not satisfy the needs of the desired driver.
I tried adding the iot2050 linux headers to the control section of this package via:
https://unix.stackexchange.com/questions/138188/easily-unpack-deb-edit-postinst-and-repack-deb#138190
But this only brought me to the point of the kernel failing during compile and not complaining about missing deps..
```
dpkg -i fixed.deb 
(Reading database ... 77834 files and directories currently installed.)
Preparing to unpack fixed.deb ...

------------------------------
Deleting module version: 1.0
completely from the DKMS tree.
------------------------------
Done.
Unpacking gasket-dkms (1.0-10) over (1.0-10) ...
Setting up gasket-dkms (1.0-10) ...
Loading new gasket-1.0 DKMS files...
Building for 4.19.59
Building initial module for 4.19.59
Error! Bad return status for module build on kernel: 4.19.59 (aarch64)
Consult /var/lib/dkms/gasket/1.0/build/make.log for more information.
dpkg: error processing package gasket-dkms (--install):
 installed gasket-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 gasket-dkms
```
This is the error log:
```
DKMS make.log for gasket-1.0 for kernel 4.19.59 (aarch64)
Mon 08 Jun 2020 03:23:19 PM UTC
make: Entering directory '/usr/src/linux-headers-4.19.59'
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_core.o
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_ioctl.o
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_interrupt.o
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_page_table.o
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[1]: *** [scripts/Makefile.build:303: /var/lib/dkms/gasket/1.0/build/gasket_ioctl.o] Error 2
make[1]: *** Waiting for unfinished jobs....
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[1]: *** [scripts/Makefile.build:303: /var/lib/dkms/gasket/1.0/build/gasket_interrupt.o] Error 2
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[1]: *** [scripts/Makefile.build:303: /var/lib/dkms/gasket/1.0/build/gasket_page_table.o] Error 2
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[1]: *** [scripts/Makefile.build:303: /var/lib/dkms/gasket/1.0/build/gasket_core.o] Error 2
make: *** [Makefile:1517: _module_/var/lib/dkms/gasket/1.0/build] Error 2
make: Leaving directory '/usr/src/linux-headers-4.19.59'
```

When I additionally install `linux-headers-arm64` it looks a little different while compiling for the wrong kernel-version:
```
dpkg -i fixed.deb 
(Reading database ... 77837 files and directories currently installed.)
Preparing to unpack fixed.deb ...

-------- Uninstall Beginning --------
Module:  gasket
Version: 1.0
Kernel:  4.19.0-9-arm64 (aarch64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

gasket.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.0-9-arm64/updates/dkms/
rmdir: failed to remove 'updates/dkms': Directory not empty
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

apex.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.0-9-arm64/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

Warning: Unable to find an initial ram disk that I know how to handle.
Will not try to make an initrd.

DKMS: uninstall completed.

------------------------------
Deleting module version: 1.0
completely from the DKMS tree.
------------------------------
Done.
Unpacking gasket-dkms (1.0-10) over (1.0-10) ...
Setting up gasket-dkms (1.0-10) ...
Loading new gasket-1.0 DKMS files...
Building for 4.19.59
Building initial module for 4.19.59
Error! Bad return status for module build on kernel: 4.19.59 (aarch64)
Consult /var/lib/dkms/gasket/1.0/build/make.log for more information.
dpkg: error processing package gasket-dkms (--install):
 installed gasket-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 gasket-dkms
```

This leeds me to the question if there could be a `linux-headers-generic` for this specific 4.19.59 kernel, that is missing.

I appreciate the help! :)

jan-kiszka, 2020-06-08:
DKMS: if we want/need to use that approach, you indeed need the headers on the target image.

For that, you found one piece in the puzzle already (IMAGE_INSTALL), but one is missing. That is compiling the kernel and, thus, also linux-headers natively, rather than cross. Try to add `ISAR_CROSS_COMPILE = "0"` to linux-iot2050_4.19.59+.bb. May 10-40 times the cross-build time for the kernel.

gramss, 2020-06-08:
I need to use it, yes.

I did recompile the kernel/image with the suggested flag. But sadly the headers are still the same..? Did took about 2 hours on my machine.

I erased my eMMC and reflashed my usb stick with the latest generated image..
Have I missed something?

jan-kiszka, 2020-06-09:
Does dkms build against the wrong kernel version if you do not patch it? That would mean you couldn't use it also in upstream to build against Debian's RT kernel, eg.

jan-kiszka, 2020-06-09:
Just checked: By default, dkms builds for all installed kernels, at least for all Debian kernels.
```
Setting up dahdi-dkms (1:2.11.1.0.20170917~dfsg-7) ...
Loading new dahdi-DEB_VERSION DKMS files...
Building for 4.19.0-9-amd64 and 4.19.0-9-rt-amd64
```
However, I suspect it will do so for our kernel as well, provided we have the linux-headers package installed.

gramss, 2020-06-09:
It did build against our kernel.

I was only patching the dependencies of that package. (The check that prevents running the package if the dependencies are not met)
So if we create an alias package called `linux-headers[-generic|-common]` we should be good to go.

The module works 100% as expected.
Tested it with the userspace libraries and had no issues with it.

I can either close this now or wait till the ‚alias‘ (?) package gets added to the example image?

When do you expect a patch for isar to generate those headers with the cross toolchain?

jan-kiszka, 2020-06-09:
I'm unsure about the alias yet - upstream does not use that packages anymore. Also, there are no unfulfilled dependencies of dkms, only recommendations. Thus, there should be no need to patch anything.

Regarding the availability of an arm64 linux-headers-iot2050: That may take a while unless I find a shortcut. So we should leave this issue open for now.

jan-kiszka, 2020-06-09:
IOW: Try `apt install --no-install-recommends dkms` instead in order to get away without pulling other kernel headers or even more. If it refuses to install on arm64 otherwise, that might even be a Debian upstream issue (there is no -amd64 kernel for arm64...).

gramss, 2020-06-09:
The problem is not with `dkms`. This worked right form the beginning.
The problem is with the overlay/patch of `dkms` called `gasket-dkms`.
```
root@iot2050-debian:~# apt install --no-install-recommends gasket-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gasket-dkms : Depends: linux-headers-686-pae but it is not installable or
                        linux-headers-amd64 but it is not installable or
                        linux-headers-generic but it is not installable or
                        linux-headers but it is not installable
E: Unable to correct problems, you have held broken packages.
```
Here is the info about this package:
```
root@iot2050-debian:~# apt info gasket-dkms
Package: gasket-dkms
Version: 1.0-10
Maintainer: Coral <coral-support@google.com>
Installed-Size: 240 kB
Depends: dkms (>= 2.1.0.0), linux-headers-686-pae | linux-headers-amd64 | linux-headers-generic | linux-headers
Homepage: https://coral.withgoogle.com/
Download-Size: 44.7 kB
APT-Sources: https://packages.cloud.google.com/apt coral-edgetpu-stable/main arm64 Packages
Description: DKMS source for the gasket driver
```

Those are hard dependencies. I was not able to ignore them with apt flags. That's why I went down the rabbit whole of patching .deb files manually.
But this package can be fixed by its maintainers. At least I can try create an issue at the coral repos.
What would you recommend to have as dependencies?
`dkms` seems to have the `linux-headers` only as Recommend and not depends..

But if the gasket-dkms does not have the headers installed it does fail installing (see above comments). So it is right to have them as dependencies.
If the linux-headers-common are not used at upstream anymore. What should they use instead then..? Isn't there any name that a package can refer to, platform independent? I'm pretty sure that adding `linux-headers-iot2050` won't get merged into their package.. Or at least it looks pretty strange to list all "compatible" boards if the package claims to be compatible with all armv7 and armv8 (aarch64 ?) debian based platforms/operating-systems.
Wouldn't it be okay to have an "legacy" package for those headers to be compatible to such situations. I can believe that there are more packages using this constellation.

jan-kiszka, 2020-06-09:
OK, got it - downstream issue then, and you must patch. Upstream does not have such issues, see e.g. https://packages.debian.org/buster/bbswitch-dkms.

That's the problem with out-of-tree modules, including their packaging...

jan-kiszka, 2020-12-14:
We have workarounds, just the full integration of kernel headers on the image is not in sight due to Isar limitations. Closing for now.

closed 2020-12-14 by jan-kiszka