siemens/meta-iot2050 #14: Example Ethernet Configuration

simonabler, 2020-06-08:
> The example image starts with the IP 192.168.200.1 preconfigured on the first Ethernet interface, and use DHCP at another. You can use ssh to connect to the system.

It was the other way around. P2 was preconfigured with 192.168.200.1

simonabler, 2020-06-08:
Interface X1 P2 Port is eth0.
eth1 is accessable on X1 P1. 
It is confusing.

jan-kiszka, 2020-06-08:
That is a good point... I'm running my box without cover so I didn't notice this "detail" yet. @johnson325, can we simply flip that for the next release or is that too late now?

applea9, 2020-06-09:
In IOT2040, X1P1LAN is preconfigured with 192.168.200.1, and X2P1LAN is preconfigured with DHCP mode.

johnson325, 2020-06-09:
> That is a good point... I'm running my box without cover so I didn't notice this "detail" yet. @johnson325, can we simply flip that for the next release or is that too late now?

Technically, it's possible. I'll try to do the alignment internally.

closed 2020-06-22 by johnson325