siemens/meta-iot2050 #17: Userspace interaction with u-boot is not working

gramss, 2020-06-09:
Hey,

I needed to switch from my eMMC boot image back to usb0 boot image.
And I think I'll need to do this in the future as well, when I want to upgrade my image for example.

So I heard that I could use `fw_setenv` to change the parameters in u-boot, as I thought I did not have any TTL cables around here. This is the issue I had with them:

```
root@iot2050-debian:~# fw_setenv boot_targets usb0
Cannot parse config file '/etc/fw_env.config': No such file or directory
Error: environment not initialized
root@iot2050-debian:~# fw_printenv boot_targets
Cannot parse config file '/etc/fw_env.config': No such file or directory
root@iot2050-debian:~#
```

Could this be fixed in the future?

I found a TTL cable that did the job and I formatted the eMMC afterwards, so I'm back at usb0 boot for now. But the UART line at 115200 baud was not working nicely. (I'm pretty sure that is due to my setup. Some cheap ttl board and some really flanky "adapter" cables)
Would be cool to do this in the future without being dependent on a TTL cable :)

jan-kiszka, 2020-06-09:
The official recommendation so far is doing that from the U-Boot command line, but what you did shall work as well. It used to work, but don't recall if that was only prior to dde86de6b2688883939c31bcb43eaee11ba80a4e which switched to upstream u-boot-tools. We will inject a fitting /etc/fw_env.config like this:
```
# MTD device name       Device offset   Env. size       Flash sector size       Number of sectors
/dev/mtd3               0x0000          0x20000         0x10000
/dev/mtd4               0x0000          0x20000         0x10000
```

gramss, 2020-06-09:
I went ahead and created the config file with your content:
```
root@iot2050-debian:~# fw_printenv boot_targets
boot_targets=mmc1 mmc0 usb0 usb1 usb2 
```
Worked like a charm. Also writing works now:
```
root@iot2050-debian:~# fw_printenv test
## Error: "test" not defined
root@iot2050-debian:~# fw_setenv test 1
root@iot2050-debian:~# fw_printenv test
test=1
```
Please ping this issue, when you included it into the example image 🙂

jan-kiszka, 2020-06-09:
Pushed a branch for early testing.

commit 44f3b640 (2020-06-09): customizations-base: Install fw_env.config

That used to come via self-built u-boot-tools. As we only need this
config, it would be overkill to build U-Boot for that purpose again.

Closes: #17
commit 191b0703 (2020-06-10): customizations-base: Install fw_env.config

That used to come via self-built u-boot-tools. As we only need this
config, it would be overkill to build U-Boot for that purpose again.

Closes: #17
closed 2020-06-10 by johnson325