siemens/meta-iot2050 #18: node-red-gpio

simonabler, 2020-06-09:
It seems that the node-red-gpio module will not be modified during the image build process.
Test image was made with:
`./kas-docker --isar build kas-iot2050-example.yml` 
but mraa-gpio-led.js is not in the directory
> /usr/lib/node_modules/node-red/node_modules/node-red-node-intel-gpio

However:
If i change the bitbake directory in: recipes-app/node-red-gpio/node-red-gpio_0.0.6-IOT2050.bb
for example to 
``` shell
do_install() {
    install -v -d ${D}/usr/lib/node_modules/node-red/node_modules/node-red-node-intel-gpio-em
    install -v -m 644 ${S}/hardware/intel/* ${D}/usr/lib/node_modules/node-red/node_modules/node-red-node-intel-gpio-em
}
```
the directory node-red-node-intel-gpio-em on the image has all the modifications.

jan-kiszka, 2020-06-09:
Ouch, that an ugly mistake in the recipe: patching comes after do_install. This should work around it for now:
```
diff --git a/recipes-app/node-red-gpio/node-red-gpio_0.0.6-IOT2050.bb b/recipes-app/node-red-gpio/node-red-gpio_0.0.6-IOT2050.bb
index a0af085..ac0687f 100644
--- a/recipes-app/node-red-gpio/node-red-gpio_0.0.6-IOT2050.bb
+++ b/recipes-app/node-red-gpio/node-red-gpio_0.0.6-IOT2050.bb
@@ -28,3 +28,4 @@ do_install() {
     install -v -d ${D}/usr/lib/node_modules/node-red/node_modules/node-red-node-intel-gpio
     install -v -m 644 ${S}/hardware/intel/* ${D}/usr/lib/node_modules/node-red/node_modules/node-red-node-intel-gpio
 }
+addtask install after do_patch before do_prepare_build
```
But that rather belongs into Isar.

@johnson325, that affects our tagged release unfortunately.

jan-kiszka, 2020-06-09:
Upstream: https://groups.google.com/d/msgid/isar-users/9ded6977-71ed-3ef1-a1fe-4cb54fed78ed%40siemens.com

commit fd3b2d8b (2020-06-09): Add Isar patch to permit patching of dpgk-raw inputs

Needed to fix node-red-gpio build.

Closes: #18
commit f5f5d045 (2020-06-10): Add Isar patch to permit patching of dpgk-raw inputs

Needed to fix node-red-gpio build.

Closes: #18
closed 2020-06-10 by johnson325