siemens/meta-iot2050 #24: Expose user button

jan-kiszka, 2020-06-15:
This should either happen as GPIO via mraa, i.e. by enhancing the IOT2050 platform support. Or we expose it as gpio-keys (input device), see patch below. Advantage of the latter is broader availability under Linux. Disadvantage is that mraa seems to lack support for input devices and will need a larger patch in order to ensure also binding to node-red & Co.

Patch for exporting the button via gpio-keys:
```
diff --git a/arch/arm64/boot/dts/siemens/iot2050-common.dtsi b/arch/arm64/boot/dts/siemens/iot2050-common.dtsi
index b3082a481f5a..a7ea32b986e3 100644
--- a/arch/arm64/boot/dts/siemens/iot2050-common.dtsi
+++ b/arch/arm64/boot/dts/siemens/iot2050-common.dtsi
@@ -106,6 +106,19 @@ dp_refclk: clock {
 		#clock-cells = <0>;
 		clock-frequency = <19200000>;
 	};
+
+	gpio-keys {
+		compatible = "gpio-keys";
+		autorepeat;
+		pinctrl-names = "default";
+		pinctrl-0 = <&push_button_pins_default>;
+
+		sw5 {
+			label = "GPIO Key USER1";
+			linux,code = <BTN_0>;
+			gpios = <&wkup_gpio0 25 GPIO_ACTIVE_LOW>;
+		};
+	};
 };
 
 &wkup_pmx0 {
@@ -361,7 +374,6 @@ &wkup_gpio0 {
 	pinctrl-0 = <&arduino_io_d2_to_d3_pins_default
 				&arduino_i2c_aio_switch_pins_default
 				&arduino_io_oe_pins_default
-				&push_button_pins_default
 				&db9_com_mode_pins_default>;
 };
 ```

@BaochengSu, @johnson325, any comments?

BaochengSu, 2020-06-16:
This gpio is not only used as a input pin. It is connected to a "Y" circuit, the other ends are the user button and the arduino pin which is coresponding to the uno PC6/RESET pin.

So when used as input, it is connected to user botton, and when used as output, it is connected to arduino pin, to reset the sheild.

For the arduino pin, it is used as the reset pin for arduino uno, but on iot2000, it is used for reseting the daughter sheild, and iot2050 design follows iot2000.

jan-kiszka, 2020-06-16:
On the IOT2000, the user reset button was always available, monitored by the arduino management software to trigger a reset of the sketch software. Additionally, it could be read (polled) by mraa and was thus available to node-red. Do you say that this is no longer possible because the pin has to be deactivated in order to issue shield resets (which happened electrically on the IOT2000)?

BaochengSu, 2020-06-16:
The same use case is still workable on IoT2050, the difference between IoT2000 and IoT2050 is that for IoT2000, there are 2 GPIOs related to these functions, one is used as input to read the user button, and the other is used as output to send reset signal to shield.

But for IoT2050, these 2 GPIOs are combined to one, i.e. the <wkup_gpio0 25>.

BaochengSu, 2020-06-16:
And as for your question, it is workable, but not simultaneously. You have to pick a functionality at a time, there is no way to use them together.

And from HW's design's point of view, they think the use cases for use them together would be very rare.

jan-kiszka, 2020-06-16:
OK, so we need to stick with GPIO modeling of the pin. Still, we need mraa binding for that, like we had in later versions of the IOT2000 image.

jan-kiszka, 2021-01-18:
First prototype for mraa available in https://github.com/siemens/mraa/commits/jan/iot2050

commit 3e8c4947 (2021-01-30): mraa: Update to 2.2.0 and new GPIO line location scheme

This brings a number of updates to mraa. Specifically, it switches to
using GPIO line names for discovering their sysfs numbers. This is
needed to support upstream kernels now and systems with different
gpiochip probing orders in the future.

Furthermore, it adds support the the user button.

By moving to 2.2.0, we now need to explicitly enable javascript support
during configure.

Fixes #23
Fixes #24
commit 6460a578 (2021-01-30): node-red-gpio: Update to new mraa and add support for all digital inputs

The board ID changed with mraa 2.2.0. Furthermore, align the LED names
to the real ones and add digital inputs D14-D20, including the USER
button.

Fixes #24
closed 2021-01-31 by BaochengSu