siemens/meta-iot2050 #28: Use of DS18B20 temperature sensor with onewire bus

Tom-RCNL-DI-FOA, 2020-07-01:
Hello,

I'm trying to use a DS18B20 sensor straight via the terminals of the IOT2050.

I've changed the defonfig_base file [https://github.com/siemens/meta-iot2050/blob/2a63d44e4eb97930c9dc7172bc4e202ef299a30d/recipes-kernel/linux/files/iot2050_defconfig_base](url) by adding the following lines:
CONFIG_W1_SLAVE_THERM=m
CONFIG_W1_MASTER_GPIO=m

After this I rebuild the image and now I can do 'modprobe w1-therm && modprobe w1-gpio'. So far so good.

Now I need to add the onewire devices to the devicetree but it seems not in the image, is that right?

Could you please check what I need to do to use these temperature sensors?

Thank you very much,

Tom Visser 
RC-NL FOA

jan-kiszka, 2020-07-01:
Yes, there is nothing in the device tree.

If I interpret https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/w1/w1-gpio.txt correctly, you just need to add such snippet with the right GPIO number to the existing DT. Or use a DT overlay, but that is likely not enabled in our kernel yet.

Do you already have the right snippet available?

Tom-RCNL-DI-FOA, 2020-07-01:
Hello Jan,

Some webpage I found does the following:

First add the device definition in your devices devicetree (assuming 1-wire device is connected to PD2 pin):

onewire_device {

   compatible = "w1-gpio";

   gpios = <&pio 3 2 GPIO_ACTIVE_HIGH>; /* PD2 */

   pinctrl-names = "default";

   pinctrl-0 = <&my_w1_pin>;

};

And then define the pins under &pio label:

&pio {

   ...

   my_w1_pin: my_w1_pin@0 {

     allwinner,pins = "PD2";

     allwinner,function = "gpio_in";

     allwinner,drive = <SUN4I_PINCTRL_10_MA>;

     allwinner,pull = <SUN4I_PINCTRL_PULL_UP>;

   };

   ...

};

Is this what you mean with snippet?

Met vriendelijke groet,
Tom Visser

Siemens Nederland N.V.
RC-NL DI FA FOA
Prinses Beatrixlaan 800
2595 BN Den Haag, Nederland
Tel.: +31 612320728
mailto:tom.visser@siemens.com
www.siemens.com/ingenuityforlife<https://siemens.com/ingenuityforlife>

jan-kiszka, 2020-07-08:
Yes, this looks similar, though it will not fit as-is (it is referring to an Allwinner SoC).

So you need to solve two issues:
 - inject a snippet into the DT that is built for the IOT2050
 - figure out what exactly is needed here

For the first point, create a patch that targets arch/arm64/boot/dts/siemens/iot2050-common.dtsi, then hook up that patch with https://github.com/siemens/meta-iot2050/blob/master/recipes-kernel/linux/linux-iot2050_4.19.59%2B.bb. The second point might be tricker, though, will likely involve a bit of trial and error. Which GPIO pin are your targeting exactly?

Tom-RCNL-DI-FOA, 2020-07-09:
Hello Jan,

Thank you for your reply.

I’m not a linux/RPI/microcontroller-specialist but I guess I’ll try to figure out what you told me haha.

Btw I did add the config-parts for w1-therm, w1-gpio and ch341 (for usb serial comm) to my built-files.

After this  ‘sudo modprobe w1-therm || modprobe w1-gpio’ was possible, unfortunately the w1-therm/w1-gpio didn’t recognize the sensors, while my Arduino nano immediately sees the sensor.

In the meantime the customer decided to use different sensors (PT100) but we’d still like to be able to use these sensors because they are pretty good and a lot of people use them (on RPI).

We’d like to use any port as may possible, is there a default port to use for this?

RPI uses GPIO4 (pin 7) for this kind of sensors by default.

With kind regards,

Tom Visser

jan-kiszka, 2020-07-09:
There is no default GPIO for this. You need to pick one, specify that in the device tree node that describes the w1-gpio, and then start the device with that DT. Just loading the drivers does not work because Linux drivers are configured via DT, not via hard-coded parameters like it is the case on the more simplistic Arduinos.

Tom-RCNL-DI-FOA, 2020-07-10:
Hello Jan,

Thank you for your response, unfortunately I’ve been searching for this devicetree but I can’t find it.
Is this something I need to install?

Met vriendelijke groet,
Tom Visser

Siemens Nederland N.V.
RC-NL DI FA FOA
Prinses Beatrixlaan 800
2595 BN Den Haag, Nederland
Tel.: +31 612320728
mailto:tom.visser@siemens.com
www.siemens.com/ingenuityforlife<https://siemens.com/ingenuityforlife>

jan-kiszka, 2020-12-14:
Oops, didn't reply back then. I suppose this issue is obsolete now. Feel free to reopen if not.

closed 2020-12-14 by jan-kiszka