siemens/meta-iot2050 #29: GPIO accessible via gpio group

simonabler, 2020-07-06:
can you put the arduino interface DI, DO, user leds , ..  in a gpio group. 

i managed to put the led via the udev rules into the gpio group but failed with the DOs of the arduino shield.

cat >/etc/udev/rules.d/98-comLed.rules  << EOF1
SUBSYSTEM=="gpio*", PROGRAM="/bin/sh -c 'chown -R root:gpio /sys/class/leds && chmod -R 770 /sys/class/leds && chown -R root:gpio /sys/devices/platform/leds/leds && chmod -R 770 /sys/devices/platform/leds/leds'"
EOF1 

can you set this per default?

jan-kiszka, 2020-07-08:
What is the requirement behind this request? Is it about access control management?

The example image is a reference for creating own customization on top. Adding own users, managing their permissions would be one of the possible customizations. If a pattern is generic enough, we are happy to consider including it also in the example image. The best approach to propose that is by submitting a pull request with the desired changes. If that is too tricky, we need at least some more details on the desired use case.

simonabler, 2020-07-09:
if you using the example image. you are forced to use the root user for node-red.
otherwise you cannot use the gpios.

So in my view it is generic to put the leds, gpios and user buttons into a special "gpio" group.

Therefore someone can put own customization on top, adding new user, and give them easily access to the hardware via the gpio group.

jan-kiszka, 2020-07-09:
Ok, so the actual requirement is "run Node Red as non-root user". Permitting non-root access to GPIOs is a building block for that, but it is not reasonable to only do that without a use case, thus also a test case.

jan-kiszka, 2021-01-18:
I just learned that there are limits so far in de-privileging anything that needs mraa to access IOs - and that included node-red. The reason is the way the pin mux controller is accessed from mraa: as directly mapped MMIO via /dev/mem, and that is naturally a highly privileged access. Currently, we are missing an alternative. And nothing would be won by simply adding /dev/mem to such a GPIO group as well. Any user in that group would still have effective root rights.

simonabler, 2021-01-18:
ok, i understand. That makes sense.

Thank you for your researching.

jan-kiszka, 2021-01-18:
We can reopen this topic when the dependency on /dev/mem is resolved. But we have no good idea for that yet, though it's clearly desirable.

mwesenjak, 2022-11-21:
@jan-kiszka what about creating a second chardev which provides access to the pin mux controller's memory only (i.e. /dev/gpiomem) and adding this one to a GPIO group?

It appears that the raspberry-pi guys do it like this as well:
https://github.com/eq-3/RaspberryMatic/blob/master/linux-4.1/drivers/char/broadcom/bcm2835-gpiomem.c

jan-kiszka, 2022-11-21:
We are in fact looking into overcoming /dev/mem for pinmux'ing right now, for the purpose of enable unprivileged mraa usage.

@mwesenjak thanks for the pointer. However, it's a not a good example as it is an out-of-tree approach. We are particularly looking for a solution that permits upstreaming it, see also https://marc.info/?l=linux-gpio&m=166850640920120&w=1.

jan-kiszka, 2023-06-19:
A root-free mraa concept is merged into master by now, testers welcome. Reopen if there is anything remaining in this context, or create a new issue.

closed 2023-06-19 by jan-kiszka

Int3rC3pt0r, 2024-02-21:
@jan-kiszka I've been trying to get this to work but I'm not quite sure what permission a user/group needs to be able to access pins with for example `mraa-gpio get 20`. Any info of where to start with this would be really appreciated!

Int3rC3pt0r, 2024-02-23:
Documentation really would be great 😅. What I've tried so far is adding the following udev rules:
```
SUBSYSTEM=="gpio*", PROGRAM="/bin/sh -c '\
        chown -R root:gpiouser /sys/class/gpio && chmod -R 770 /sys/class/gpio;\
        chown -R root:gpiouser /sys/devices/virtual/gpio && chmod -R 770 /sys/devices/virtual/gpio;\
        chown -R root:gpiouser /sys$devpath && chmod -R 770 /sys$devpath;\
        chown -R root:gpiouser /sys/kernel/debug && chmod -R 770 /sys/kernel/debug;\
'"

KERNEL=="gpiochip*", GROUP="gpiouser", MODE="0660"
```
Access to `/sys/kernel/debug` allows my user that is part of the gpiouser group to run:
```
$ mraa-gpio get 1
Pin 1 = -1
$ mraa-gpio get 1
Pin 1 = 0
```
The problem here is that the value of the pin is only given the second time i run the command. This is somehow also related to NodeRed because i can only access the value of pins that I've run the command for (in this case D1).
[image]
Another obervation I've made is that i can only set the value of a pin via NodeRed after setting it before with `mraa-gpio set 1 1`.

benbrenson, 2024-04-09:
It seems, that mraa is currently not completely root-free. 
The problem here is that mraa is still relying on sysfs for controlling GPIOs. 
The sysfs creates new GPIO files on-the-fly when exporting/initializing a GPIO pin. Unfortunately the new files are created with default root permissions.

I think you controlled the pin as root somewhere in between:

```bash
$ mraa-gpio get 1
Pin 1 = -1

# somewhere here you ran mraa as root ...

$ mraa-gpio get 1
Pin 1 = 0
```
.... am I right here?

MRAA currently supports general gpio-chardev access but I have to adapt the iot2050 layer to support it.
I think the step towards using the chardev interface is the way to go.

jan-kiszka, 2024-04-09:
But the export only happens on first access, and there is no unexport, right? Couldn't we do those upfront, set the rights accordingly, and then only use the existing ones in mraa?

benbrenson, 2024-04-12:
Unfortunately I was definitely wrong with the assumption that the sysfs files for GPIO are created with default root permissions, in case the above udev rules from @Int3rC3pt0r were used.
 
After adding a udev rules file that adjusts permissions I see:
```bash
ls -l /sys/class/gpio/gpio490/
total 0
-rwxrwx--- 1 root gpio 4096 Apr 12 13:11 active_low
lrwxrwxrwx 1 root root    0 Apr 12 13:11 device -> ../../../gpiochip1
-rwxrwx--- 1 root gpio 4096 Apr 12 13:11 direction
drwxrwx--- 2 root gpio    0 Apr 12 13:11 power
lrwxrwxrwx 1 root gpio    0 Apr 12 13:11 subsystem -> ../../../../../../../../../../class/gpio
-rwxrwx--- 1 root gpio 4096 Apr 12 13:11 uevent
-rwxrwx--- 1 root gpio 4096 Apr 12 13:11 value
```

The actual problem is that the udev commands are executed asynchronously to MRAA. The required user permissions are not completely setup while MRAA accesses those files, hence the errors.
 
Even though the chardev implementation may avoid these problems, the PWM implementation still relies on almost the same sysfs export mechanism as GPIO does.