siemens/meta-iot2050 #30: Binaries built for genereic Debian Buster Arm64 work on Industrial OS 2.1, but not on the example image

Chefkoch81, 2020-07-08:
Hi,

First of all, thanks for providing a ready to go image for download.

In general it works, but I have some issues with my binaries.

I use a more generic approach for building to Debian Buster Arm64. I use a toolchain from [here ](https://releases.linaro.org/components/toolchain/binaries/latest-7/aarch64-linux-gnu/) and have a sysroot generated from a stock Debian Buster arm64 docker container.

I do my own Qt builds and based on that my own project in cmake.

Running the resulting binary results in the error message:
`libQt5Core.so.5: ELF load command alignment not page-aligned`

Doing an ldd:
```
ldd libQt5Core.so.5.12.5
        not a dynamic executable
```

The same binaries work perfectly normal in Industrial OS 2.1 on the IOT2050.

I have found that the kernel in the example image (4.19.59+) is older than in Industrial OS (4.19.94-rt39). Upon a search I found the pagesize in the ELF header could be the reason. Could the reason be the deactivation of 64kByte page sizes in the kernel of the example image?
https://github.com/siemens/meta-iot2050/commit/fb1d72677fb7443d851b032517d693bbd0e1d1e4

Or could it be the older kernel?

I tried to build the example image with support for 64k pages, but was not successful:
> ERROR: tcf-agent-1.7.0-dc373fa9-r0 do_dpkg_build: Error executing a python function in exec_python_func() autogenerated:
Full log attached. [build_log.txt](https://github.com/siemens/meta-iot2050/files/4890851/build_log.txt)

Hope I can find a way to make the generic Debian Arm64 build working on the example image and Industrial OS.

BR, Andreas

jan-kiszka, 2020-07-08:
While using a different toolchain than the original one provided by Debian may work - I'm sometimes doing that myself for prototyping purposes -, you are clearly in the "at your own risk" area then.

Regarding the concrete issue you see: If you are using the released example image V01.00.02, you will still get a kernel with 64K pages. With the industrial OS, you should get an RT kernel, and that with 4K pages. The reason for this deviation was inherited from TI. They choose slightly faster 64K pages for general purpose while going back to normal 4K for RT due to https://github.com/siemens/meta-iot2050/blob/master/recipes-kernel/linux/files/iot2050-rt.cfg#L39.

We've dropped 64K also from the example image in https://github.com/siemens/meta-iot2050/commit/fb1d72677fb7443d851b032517d693bbd0e1d1e4, after the release, because it caused crashes in Chromium. I also provided feedback to TI that their 64K is likely too risky for general purposes. I wouldn't be surprised if your case is similar. Try building the latest example image from source so that you get a kernel without 4K.

Chefkoch81, 2020-07-09:
Thanks for the input and the insight, I now have switched to the Debian [crossbuild compiler packages](https://packages.debian.org/buster/crossbuild-essential-arm64). You are right, that would be the more stable solution.

I have downloaded and use the example from [here](https://support.industry.siemens.com/cs/document/109780231/simatic-iot2050-sd-card-example-image?dti=0&lc=en-WW), as far as I see its 1.0.2.

So you mean, if it works with Industrial OS, it should work with a newer example image too? Because I tried building a new version, but failed with the error mentioned above:
> ERROR: tcf-agent-1.7.0-dc373fa9-r0 do_dpkg_build: Error executing a python function in exec_python_func() autogenerated:
> Full log attached. [build_log.txt](https://github.com/siemens/meta-iot2050/files/4890851/build_log.txt)

jan-kiszka, 2020-07-09:
Debian complains that tcf-agent.install is executable, but that is not true, at least not when looking at the repository. When I provoke this locally (chmod a+x), I get the same error. So, I suspect you checked out over a non-unix filesystem or otherwise caused this corruption. Try with a clean checkout again.

Chefkoch81, 2020-07-13:
I did try the image build as root first, which I stoppped afterwards and switched to a non-root user. It seems I forgot to reset some of the ownerships.
Long Story short -> after doing a clean checkout the image build worked.
I was also able to run my binary.

Thank you for your help, will close this issue.

closed 2020-07-13 by Chefkoch81