siemens/meta-iot2050 #32: Un-Autostart tcf-agent

bergmanu, 2020-07-14:
Is it possible to add tcf-agent to _Manage Autostart Options_ in iot2050setup? 
Would be to have the possibility to disable it, like it was possible in iot2000setup.

jan-kiszka, 2020-07-14:
Does this work?
```
diff --git a/recipes-app/iot2050setup/files/iot2050setup.py b/recipes-app/iot2050setup/files/iot2050setup.py
index b6503f8..dfcb355 100755
--- a/recipes-app/iot2050setup/files/iot2050setup.py
+++ b/recipes-app/iot2050setup/files/iot2050setup.py
@@ -100,22 +100,22 @@ class SoftwareMenu:
             return
         selection()
 
+    @staticmethode
+    def serviceEnabled(service):
+        return 'enabled' in subprocess.Popen('systemctl is-enabled %s' % service,
+                                             shell=True,
+                                             stdout=subprocess.PIPE,
+                                             stderr=open(os.devnull, 'wb')).stdout.read().decode('utf-8')
+
     def changeAutostart(self):
-        sshEnabled = 'enabled' in subprocess.Popen('systemctl is-enabled ssh',
-                                                   shell=True,
-                                                   stdout=subprocess.PIPE,
-                                                   stderr=open(os.devnull, 'wb')).stdout.read().decode('utf-8')
-        mosquittoAutostartEnabled = 'enabled' in subprocess.Popen('systemctl is-enabled mosquitto',
-                                                                  shell=True,
-                                                                  stdout=subprocess.PIPE,
-                                                                  stderr=open(os.devnull, 'wb')).stdout.read().decode('utf-8')
-        noderedAutostartEnabled = 'enabled' in subprocess.Popen('systemctl is-enabled node-red',
-                                                                shell=True,
-                                                                stdout=subprocess.PIPE,
-                                                                stderr=open(os.devnull, 'wb')).stdout.read().decode('utf-8')
+        sshEnabled = SoftwareMenu.serviceEnabled('ssh')
+        tcfAgentEnabled = SoftwareMenu.serviceEnabled('tcf-agent')
+        mosquittoAutostartEnabled = SoftwareMenu.serviceEnabled('mosquitto')
+        noderedAutostartEnabled = SoftwareMenu.serviceEnabled('node-red')
         buttonbar = ButtonBar(screen=self.topmenu.gscreen, buttonlist=[('Ok', 'ok'), ('Cancel', 'cancel', 'ESC')])
         ct = CheckboxTree(height=4, scroll=0)
-        ct.append('SSH Server Enabled', selected=sshEnabled)
+        ct.append('SSH Server enabled', selected=sshEnabled)
+        ct.append('TCF Agent enabled', selected=tcfAgentEnabled)
         ct.append('Auto Start Mosquitto Broker', selected=mosquittoAutostartEnabled)
         ct.append('Auto Start node-red', selected=noderedAutostartEnabled)
         g = GridForm(self.topmenu.gscreen, 'Advanced Options', 1, 2)
@@ -125,11 +125,14 @@ class SoftwareMenu:
         if buttonbar.buttonPressed(result) == 'cancel':
             return
         selectedOptions = ct.getSelection()
-        sshEnabledNew = 'SSH Server Enabled' in selectedOptions
+        sshEnabledNew = 'SSH Server enabled' in selectedOptions
+        tcfAgentEnabledNew = 'TCF Agent enabled' in selectedOptions
         mosquittoAutostartEnabledNew = 'Auto Start Mosquitto Broker' in selectedOptions
         noderedAutostartEnabledNew = 'Auto Start node-red' in selectedOptions
         if sshEnabled != sshEnabledNew:
             self.changeServiceSetting('ssh', sshEnabledNew)
+        if tcfAgentEnabled != tcfAgentEnabledNew:
+            self.changeServiceSetting('tcf-agent', tcfAgentEnabledNew)
         if mosquittoAutostartEnabled != mosquittoAutostartEnabledNew:
             self.changeServiceSetting('mosquitto', mosquittoAutostartEnabledNew)
         if noderedAutostartEnabled != noderedAutostartEnabledNew:
```
Untested and intermixed with refactorings.

bergmanu, 2020-07-15:
@staticmethode needs to be replaced with **@staticmethod**.
The iot2050setup opens, but if I go to _Software_ it closes automatically

jan-kiszka, 2020-07-16:
Perfect. Will send a PR with this soon.

commit 14254e43 (2020-07-17): iot2050setup: Add control for TCF agent

Allows to disable the TSC agent, just like on the IOT2020/40.

Closes: #32
closed 2020-07-20 by johnson325