siemens/meta-iot2050 #34: Support users with WIFI firmware

jan-kiszka, 2020-07-20:
We should either preinstall more firmware packages, e.g. firmware-brcm80211, or offer some logic via iot2050setup to identify and install them from there, when the user tries to configure a network device.

BaochengSu, 2020-07-20:
Due to the long clearing process, I perfer the iot2050setup way.

jan-kiszka, 2020-07-20:
Well, it targets the next release anyway, thus also a general package update. I'm slightly optimistic that modern firmware packages should not trigger any licensing concern. IIRC, we only had findings in legacy firmware. And Debian splits this up, thus we can cherry-pick.

bergmanu, 2020-07-24:
Hi, 
I'd really appreciate the support of more WiFi firmware. 
We have to clear the new version anyway, so maybe preinstalling more firmware does not blow up the clearing process.

But of course I'd also be happy with "semi-manual" installing via setup tool. But I guess this is means more effort in development, right?

commit 7c08906e (2020-12-01): iot2050-image-example: Extend set of pre-installed WIFI firmware

Closes: #34
commit 2bacc245 (2020-12-01): iot2050-image-example: Extend set of pre-installed WIFI firmware

Closes: #34
commit 9f97a575 (2020-12-02): iot2050-image-example: Extend set of pre-installed WIFI firmware

Closes: #34
commit 99200e0f (2020-12-03): iot2050-image-example: Extend set of pre-installed WIFI firmware

Closes: #34
closed 2020-12-04 by BaochengSu