siemens/meta-iot2050 #37: Write isar recipe with both c file and scripts

applea9, 2020-08-07:
Hello,

I would like to create a isar recipe which contains both a single c file and some scripts. I tried with dpkg-raw class, but examples only show how to install the scripts but nothing about how to compile a single c file in the recipe(like do_compile() in yocto). 

Now my solution is to precompiled the c file in iot2050 and install it in the dpkg-raw recipe

Could you give some advice about it, thanks a lot.

johnson325, 2020-08-07:
Please check [this](https://github.com/siemens/meta-iot2050/blob/master/recipes-app/switchserialmode/switchserialmode_0.1.bb) file as a reference.

applea9, 2020-08-07:
> Please check [this](https://github.com/siemens/meta-iot2050/blob/master/recipes-app/switchserialmode/switchserialmode_0.1.bb) file as a reference.

Thanks for the reply. 

I have checked this file, but in my case I have several shell scripts to install in addtion to compilation of the c file. The problem is do_install() only works in dpkg-raw instead of dpkg. 

I am not clear how to do both in one recipe, thanks.

AsuraZeng, 2020-08-10:
Here is a reference, maybe it could help you
# architecture:
```
├── customdemo_0.1.bb
└── files
    └── src
        ├── CMakeLists.txt
        ├── hello.sh
        └── helloworld.c

```
# bb file reference
```
DESCRIPTION = "Custom Package"
MAINTAINER = "chao.zeng@siemens.com"

SRC_URI = "file://src"

inherit dpkg-raw

do_install() {
    install -v -d ${D}/usr/bin/
    install -v -m 755 ${WORKDIR}/src/hello.sh ${D}/usr/bin/
}

S = "${WORKDIR}/src"

DEBIAN_BUILD_DEPENDS = "cmake"

do_prepare_build[cleandirs] += "${S}/debian"

do_prepare_build() {
    deb_debianize
}
```
# CMakeLists reference
```
cmake_minimum_required(VERSION 2.6)
project(customtest)
add_executable(helloworld helloworld.c)
set(CMAKE_INSTALL_PREFIX "/usr")
install(TARGETS helloworld RUNTIME DESTINATION bin)
```

applea9, 2020-08-11:
Thanks a lot!

closed 2020-08-11 by applea9

jan-kiszka, 2020-08-11:
FWIW, while this is working, it's not the cleanest approach as `dpkg-raw` should not be used for compiling. The cleaner solutions would have been `inherit dpkg` + some `debian/<package>.install` file that places the extra script into the package.

applea9, 2020-08-13:
I will have a try, thanks!