siemens/meta-iot2050 #40: add libgpiod with tools to example image

gramss, 2020-08-24:
Since Linux kernel version 4.8 gpios can be controlled via an chardev interface, see here: https://embeddedbits.org/new-linux-kernel-gpio-user-space-interface/

I was able to install libgpiod from apt, but this seems to be build without the tools.

So I was able to install libgpiod from source https://github.com/brgl/libgpiod/tree/v0.1.x with version v0.1.x (as the version installed as binary from apt with standard sources).

Needed to install `autoconf autoconf-archive libtool` but then it worked.

I see a lot of pins, but only some of them are named. Would it be possible to name them in future releases?

Here is the output I get:

```
root@iot2050-debian:~# gpioinfo 
gpiochip6 - 4 lines:
	line   0:      unnamed       unused   input  active-high 
	line   1:      unnamed       unused   input  active-high 
	line   2:      unnamed       unused   input  active-high 
	line   3:      unnamed       unused   input  active-high 
gpiochip5 - 90 lines:
	line   0:      unnamed       unused   input  active-high 
	line   1:      unnamed       unused   input  active-high 
	line   2:      unnamed       unused   input  active-high 
	line   3:      unnamed       unused   input  active-high 
	line   4:      unnamed       unused   input  active-high 
	line   5:      unnamed       unused   input  active-high 
	line   6:      unnamed       unused   input  active-high 
	line   7:      unnamed       unused   input  active-high 
	line   8:      unnamed       unused   input  active-high 
	line   9:      unnamed       unused   input  active-high 
	line  10:      unnamed       unused   input  active-high 
	line  11:      unnamed       unused   input  active-high 
	line  12:      unnamed       unused   input  active-high 
	line  13:      unnamed       unused   input  active-high 
	line  14:      unnamed       unused   input  active-high 
	line  15:      unnamed       unused   input  active-high 
	line  16:      unnamed       unused   input  active-high 
	line  17:      unnamed       unused   input  active-high 
	line  18:      unnamed       unused   input  active-high 
	line  19:      unnamed       unused   input  active-high 
	line  20:      unnamed       unused   input  active-high 
	line  21:      unnamed       unused   input  active-high 
	line  22:      unnamed       unused   input  active-high 
	line  23:      unnamed       unused   input  active-high 
	line  24:      unnamed       unused   input  active-high 
	line  25:      unnamed       unused   input  active-high 
	line  26:      unnamed       unused   input  active-high 
	line  27:      unnamed       unused   input  active-high 
	line  28:      unnamed       unused   input  active-high 
	line  29:      unnamed       unused   input  active-high 
	line  30:      unnamed       unused   input  active-high 
	line  31:      unnamed       unused   input  active-high 
	line  32:      unnamed       unused   input  active-high 
	line  33:      unnamed       unused   input  active-high 
	line  34:      unnamed       unused   input  active-high 
	line  35:      unnamed       unused   input  active-high 
	line  36:      unnamed       unused   input  active-high 
	line  37:      unnamed       unused   input  active-high 
	line  38:      unnamed       unused   input  active-high 
	line  39:      unnamed       unused   input  active-high 
	line  40:      unnamed       unused   input  active-high 
	line  41:      unnamed       unused   input  active-high 
	line  42:      unnamed       unused   input  active-high 
	line  43:      unnamed       unused   input  active-high 
	line  44:      unnamed       unused   input  active-high 
	line  45:      unnamed       unused   input  active-high 
	line  46:      unnamed       unused   input  active-high 
	line  47:      unnamed       unused   input  active-high 
	line  48:      unnamed       unused   input  active-high 
	line  49:      unnamed       unused   input  active-high 
[... 231 lines cut ...]
	line   2:      unnamed       unused   input  active-high 
	line   3:      unnamed       unused   input  active-high 
	line   4:      unnamed       unused   input  active-high 
	line   5:      unnamed       unused   input  active-high 
	line   6:      unnamed       unused   input  active-high 
	line   7:      unnamed       unused   input  active-high 
	line   8:      unnamed      "sysfs"  output  active-high [kernel]
	line   9:      unnamed      "sysfs"  output  active-high [kernel]
	line  10:      unnamed      "sysfs"  output  active-high [kernel]
	line  11:      unnamed      "sysfs"  output  active-high [kernel]
	line  12:      unnamed      "sysfs"  output  active-high [kernel]
	line  13:      unnamed      "sysfs"  output  active-high [kernel]
	line  14:      unnamed       unused   input  active-high 
	line  15:      unnamed       unused   input  active-high 
root@iot2050-debian:~# 
``` 

I was not able to write a `1` to `"user-led0-green"`, see here:

```
$gpioset gpiochip1 15=1
gpioset: error setting the GPIO line values: Device or resource busy
```

Maybe worth looking into instead of relaying on mraa? Is the iot2050 still related to Intel?

jan-kiszka, 2020-08-24:
mraa give us two things:
 - abstraction of the underlying hardware (iot2000 vs. iot2050)
 - higher language bindings, most notably javascript

And mraa is not bound to Intel hardware at all. By now it's driven by Linaro mostly, IIRC.

That said, libgpiod should be usable as well if that lib has no issues by itself. But you may need to know which GPIO you need to flip. Keep in mind that GPIO numbers of the kernel are not stable, and they are not abstracted in any way. You first need to find out which output is on which kernel GPIO number, and then you can use it. Arduino IO pin 15 is not GPIO 15.

gramss, 2020-08-24:
Okay, got the mraa part. It's nice to know that things just work.
Firing up a Python3 shell with sort of auto-completing mraa to get D13 a digital output and write a 1 is quite straight forward.

Can you maybe point me to the device tree overlay of the mraa settings?

libgpiod is quite useful out of my perspective:
```
root@iot2050-debian:~# gpiodetect 
gpiochip6 [cp210x] (4 lines)
gpiochip5 [davinci_gpio.2] (90 lines)
gpiochip4 [davinci_gpio.1] (96 lines)
gpiochip3 [davinci_gpio.0] (56 lines)
gpiochip2 [pcal9535] (16 lines)
gpiochip1 [pcal9535] (16 lines)
gpiochip0 [pcal9535] (16 lines)
```

You get a nice overview about what chips are included. Only the naming in the fourth column would be great if they were all named. This way you could always get the right GPIO with using `gpiofind D13`. This would solve the problem of:

> Keep in mind that GPIO numbers of the kernel are not stable
 
I wanted to set line 15 of gpiochip1, which represents the GPIO labeled "user-led0-green" (probably from the device-tree). And not Arduino Pin D15 (which only goes to D13 - not sure why you extended the NodeRed Mraa References to D19 while we are at it..?). But I think I failed as the kernel owns this pin. (Other non-owned-by-kernel pins work fine with this tool).

My pitch here would be including this to the firmware of the IOT2050 as it is becoming the next sysfs controlling of GPIOs and also includes controlling Pull-Ups (in kernel 5.5). Would be cool to have a nice transition towards it. And a readable cli-interface to read and control GPIO pins.. :)

jan-kiszka, 2020-08-24:
> Okay, got the mraa part. It's nice to know that things just work.
> Firing up a Python3 shell with sort of auto-completing mraa to get D13 a digital output and write a 1 is quite straight forward.
> 
> Can you maybe point me to the device tree overlay of the mraa settings?
> 

There is no dto involved here. mraa has its configuration for board encoded into its code. Let's call that pragmatic.

> libgpiod is quite useful out of my perspective:
>
> ```
> root@iot2050-debian:~# gpiodetect
> gpiochip6 [cp210x] (4 lines)
> gpiochip5 [davinci_gpio.2] (90 lines)
> gpiochip4 [davinci_gpio.1] (96 lines) [...]

We still need to check if we can give names to lines of gpio controllers and if that will help here as well. See linux/Documentation/devicetree/bindings/gpio/gpio.txt, gpio-line-names.

> I wanted to set line 15 of gpiochip1, which represents the GPIO labeled "user-led0-green" (probably from the device-tree). And not Arduino Pin D15 (which only goes to D13 - not sure why you extended [...]

As long as the LED is assigned to the GPIO, you can't control the pin via the GPIO interface - the kernel's LED subsystem owns it. Use the LED interface then. That's also why this GPIO is not exposed via mraa.

> My pitch here would be including this to the firmware of the IOT2050 as it is becoming the next sysfs controlling of GPIOs and also includes controlling Pull-Ups (in kernel 5.5). Would be cool to have a nice transition towards it. And a readable cli-interface to read and control GPIO pins.. :)

I'm not sure how far we will get with DT tuning. But if you like to play with a recent mainline kernel already, build the image with `:kas/opt/upstream.yml` appended. That will give you a (widely) working 5.9-rc1 - with a broken mraa (because of "GPIO numbers of the kernel are not stable", #23).

jan-kiszka, 2021-01-18:
First experiment: https://github.com/siemens/linux/commit/3071cd632d80015f90868d2360d179b07aea955c. The external pins and also the user button are now listed in `gpioinfo`. If you need to switch the muxes, mraa is still needed, of course.

commit fce30a85 (2021-01-30): iot2050-image-example: Add gpiod

Fixes #40
closed 2021-01-31 by BaochengSu