siemens/meta-iot2050 #41: Overwrite with .bbappend recipe in new isar layer

applea9, 2020-08-26:
Hello,

I am trying to overwrite some existing recipes(customizations-example, e.g.) in my own layer with .bbappend recipe.

For example, I would like to replace the postinst file in customizations-example recipe.

Following is my steps:

1) Follow the yocto way to set BBFILE_PRIORITY to 7 in layer.conf
2) Create postinst file with some modification in files folder
3) Add it in SRC_URI in .bbappend file.

But the result is postinst file is not replaced.

Could you give some advice? thanks a lot.

jan-kiszka, 2020-08-26:
The key is something like `FILESEXTRAPATHS_prepend := "${THISDIR}/files:"` so that postinst is searched for in your own `files` folder.

applea9, 2020-08-26:
> The key is something like `FILESEXTRAPATHS_prepend := "${THISDIR}/files:"` so that postinst is searched for in your own `files` folder.

Thanks for quick reply. you are right, I have FILESEXTRAPATHS_prepend := "${THISDIR}/files:" in yocto recipe.  I will try it.

closed 2020-08-26 by applea9