siemens/meta-iot2050 #49: network not working on example-image

davimaf, 2020-09-29:
With the latest example-image build is not possibile to connect to IOT2050 to a network. All the network commands not working anymore, hanging up with blinking dash.
Tested commands:

- nmcli
- nmtui
- iot2050setup-->Networking
- ip a

I tried to restart service network-manager, but also this command hang up on nothing.

jan-kiszka, 2020-09-29:
The release image works? Do you see any suspicious messages on the serial console, specifically crash messages? What you describe sounds a lot like a mismatch when detecting the firmware of the device and selecting the right device tree from that.

davimaf, 2020-09-30:
> The release image works? Do you see any suspicious messages on the serial console, specifically crash messages? What you describe sounds a lot like a mismatch when detecting the firmware of the device and selecting the right device tree from that.

I only build up the `master `using example-image.yml without any modification (before I added some custom layers, thinking was my fault) and the system boot up without any warning. Then connecting ETH cable with DHCP and trying the commands above the terminal hangs on with blinking dash and after a timeout of 1-2 minutes exit without doing anything. 
Using iot2050setup, on network menu, the screen hang on blue screen for 1 minute then it exit.

I also tried to `systemctl network-manager status `and is in running state, but if i try to `restart `it, the command hang up and will not recover unless you close the terminal.

jan-kiszka, 2020-09-30:
This is an unusual behavior. Can you share the complete boot log (serial consol grab or dmesg output)? Also, are you on a basic or advanced device?

jan-kiszka, 2020-12-01:
Was this resolved, can we close it?

closed 2020-12-14 by jan-kiszka