siemens/meta-iot2050 #52: add IOT2050 to Blinka (CircuitPython) compatible board list

gramss, 2020-10-09:
Hey there,

we ( @TheresaHeim and I ) want to add the IOT2050 to the compatible list of boards of Blinka: https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/more-to-come

I²C and SPI should be functioning out of the box with the default system libraries for them.
Adding GPIO and PWM would be great to have it working on top of MRAA, which was already suggested here: 
https://github.com/adafruit/Adafruit_Python_GPIO/pull/46#issuecomment-428752678

And a PR here (was rejected as they were moving forward to Blinka):
https://github.com/adafruit/Adafruit_Python_GPIO/pull/68

This would enable a variety of code examples to use with common adafruit breakout boards!

Anybody willing to support us on our journey here? :)

Cheers

jan-kiszka, 2020-10-09:
Would surely be nice!
On first sight, this indeed looks like a generic task, mapping of mraa's python binding to blinka. Or do you already see something missing in meta-iot2050 and its mraa extension?

gramss, 2020-10-09:
Hey @jan-kiszka , thanks for your fast answer!

Yes, I fully agree on this. I do not see anything that would be dependent only on the IOT2050.
But I think that you cannot easily integrate all mraa compatible boards to Blinka, which reflects already in individual board_identifiers for the basic and advanced version of the IOT2050.
But this work should be as generic as possible! 
(IOT2040 could be added by this as well if desired)

We have to see what easy steps will look like to integrate it. The tutorial is already really user friendly, but @TheresaHeim is completely new to Linux and Python. So this might be a challenge. Would be cool if someone could support us with the nasty bits.. But with MRAA we might be able to skip working with linux internas at all! We'll see how it goes.. 😄

Cheers

jan-kiszka, 2020-10-10:
Yeah, current mraa support for iot2050 does not cover all bits of the hw, see e.g. #24. But that can be extended later, and what is there would already be useful to expose this way.

i'm not familiar with CircuitPython and Blinka yet, but I try to support from the platform perspective. Just drop your questions or issues.

TheresaHeim, 2020-11-05:
I have already changed and added the constants and detection codes. Now I am at the point to add the chip file. Which GPIOs are used in the IOT2050 and are they changing?

TheresaHeim, 2020-11-05:
I have already changed and added the constants and detection codes. Now I am at the point to add the chip file. Which GPIOs are used in the IOT2050 and are they changing? How are the names of the libgpio? Here is an example of another board of pin.py. 

[image]

jan-kiszka, 2020-11-05:
You are addressing the GPIO via libgpio, right? That one also incorrectly seems to rely on GPIO numbers. But, as written above, they are conceptually not stable. A mapping like above would break when we move from the TI kernel 4.19 to upstream kernel 5.10.

There is now https://github.com/siemens/meta-iot2050/pull/48 (which I still need to check out...) which addresses that for mraa. If you can use mraa instead of libgpio plus that MR, you would be on the safe side. mraa contains a mapping from I/O pin names to the currently valid GPIO number.

TheresaHeim, 2020-12-01:
The GPIOs are working now. 
But if I try to test an Adafruit Sensor (bme680) I get an Error in the file busio.py: ModuleNotFoundError: No module named "machine".  I have allready installed "machine" and updated blinka, like it was recommended in an other issue. (https://github.com/adafruit/Adafruit_Blinka/issues/263) It´s still not working. 
[image]

TheresaHeim, 2020-12-01:
Do you have an idea whats the problem could be?

jan-kiszka, 2020-12-01:
python2 / python3 confusion? I often install to python[2] by accident when I meant python3/pip3 etc.

TheresaHeim, 2020-12-01:
I used the right one: python3 -m pip install machine

jan-kiszka, 2020-12-01:
Check via `strace` which files are being opened and compare what has been installed. Maybe the local site is not in the python search path.

jan-kiszka, 2021-02-01:
FWIW, we have now gpiod support in master, and we have named GPIO lines. Maybe that helps also with Blinka.

jan-kiszka, 2023-07-19:
Still relevant?

jan-kiszka, 2023-10-23:
Seems this effort stalled.

closed 2023-10-23 by jan-kiszka