siemens/meta-iot2050 #54: The other ethernet port not works when one port is unpluged

applea9, 2020-10-16:
Hello,

I met a problem with IOT2050 ethernet ports. 

I set X1P1 in static IP mode and X1P2 in DHCP mode, both ports have ethernet cables pluged in and working well. When I unplug one ethernet cable from one port, the other port will also not work.

I see IOT2050 only has one NIC, is it normal case due to one NIC and two PHY?

jan-kiszka, 2020-10-16:
That is not a desired or expected behavior. We need to check this. Does it matter which cable you pull?

applea9, 2020-10-16:
> That is not a desired or expected behavior. We need to check this. Does it matter which cable you pull?

It only occurs when I unplug the DHCP port. And the static IP port's behavior is in the first several seconds it can't ping, afterwards it can be ping but I can't access IOT2050 with SSH or Web Application.

I do the ethernet configuration in /etc/network/interface, which will manage ethernet ports and network manager's ethernet configuration will not work.

jan-kiszka, 2020-10-16:
Did you disable the Network Manager then? If not, there might be two services working against each other. Default is Network Manager, not ifupdown.

applea9, 2020-10-16:
no, I'm using Network Manager for 4G and wifi configuration and only ethernet ports are managed by ifupdown. I read several documents said if configuration is done in ifupdown Network Manager will follow the configuration.

applea9, 2020-10-26:
I removed the configuration in /etc/network/interface and Network Manager is managing the ethernet port now. The problem still remains.

jan-kiszka, 2020-10-29:
Just tried that with the vanilla image built from 65b5aafeb8273fcee74054f8aad112287d46a872, and it works fine for me: X1P1 cross-linked to my host, X1P2 connected to router. When I unplug X1P2, I can still connect via ssh on the static link. Might be something specific to your local network setup?

applea9, 2020-11-04:
I rebuilt a new image based on [65b5aaf](https://github.com/siemens/meta-iot2050/commit/65b5aafeb8273fcee74054f8aad112287d46a872), it ok now. But I am still not clear why this problem occurs. 

Thanks for your help.

closed 2020-11-04 by applea9