siemens/meta-iot2050 #56: Unable to build anything

ServiceISRI, 2020-11-18:
I tried to build an image with the provided files:

Everytime the build fails and throws many error messages.

Host-System: 4.4.0-194-generic #226-Ubuntu SMP Wed Oct 21 10:19:36 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

Docker version 19.03.13, build 4484c46d9d

docker-compose version 1.27.4, build 40524192

command used: 

`kas-docker build kas-iot2050-example.yml:kas/opt/upstream.yml`
 Error: 

```
2020-11-18 11:57:44 - INFO     - kas 2.2 started
2020-11-18 11:57:44 - INFO     - /repo$ git rev-parse --show-toplevel
2020-11-18 11:57:44 - INFO     - /repo/kas/opt$ git rev-parse --show-toplevel
2020-11-18 11:57:44 - INFO     - /repo$ git rev-parse --show-toplevel
2020-11-18 11:57:44 - INFO     - Using /repo as root for repository meta-iot2050
2020-11-18 11:57:44 - INFO     - /work/isar$ git remote set-url origin https://github.com/ilbers/isar
2020-11-18 11:57:44 - INFO     - /work/isar$ git cat-file -t bf0aff2af6dfb8f6729ffced8c73d4a5ac73a871
2020-11-18 11:57:44 - INFO     - Repository isar already contains bf0aff2af6dfb8f6729ffced8c73d4a5ac73a871 as commit
2020-11-18 11:57:44 - INFO     - /repo$ git rev-parse --show-toplevel
2020-11-18 11:57:44 - INFO     - Using /repo as root for repository meta-iot2050
2020-11-18 11:57:44 - INFO     - /work/isar$ git status -s
2020-11-18 11:57:44 - INFO     - /work/isar$ git rev-parse --verify -q origin/bf0aff2af6dfb8f6729ffced8c73d4a5ac73a871
2020-11-18 11:57:44 - INFO     - /work/isar$ git checkout -q bf0aff2af6dfb8f6729ffced8c73d4a5ac73a871
2020-11-18 11:57:44 - INFO     - /repo$ git rev-parse --show-toplevel
2020-11-18 11:57:44 - INFO     - Using /repo as root for repository meta-iot2050
2020-11-18 11:57:44 - INFO     - /work/isar$ /tmp/tmpqzzl_k4u/get_bb_env /work/build
2020-11-18 11:57:44 - INFO     - /repo$ git rev-parse --show-toplevel
2020-11-18 11:57:44 - INFO     - Using /repo as root for repository meta-iot2050
2020-11-18 11:57:44 - INFO     - /work/isar$ git checkout -q -B patched-bf0aff2af6dfb8f6729ffced8c73d4a5ac73a871
2020-11-18 11:57:44 - INFO     - /work/isar$ git apply /repo/isar-patches/0001-dpkg-raw-Allow-patching-and-templating.patch
2020-11-18 11:57:44 - INFO     - Patch applied. (patch path: /repo/isar-patches/0001-dpkg-raw-Allow-patching-and-templating.patch, repo: isar, patch entry: 01-dpgk-raw-patch)
2020-11-18 11:57:44 - INFO     - /work/isar$ git add -A
2020-11-18 11:57:44 - INFO     - /work/isar$ git commit -a --author kas <kas@example.com> -m msg
2020-11-18 11:57:44 - INFO     - [patched-bf0aff2af6dfb8f6729ffced8c73d4a5ac73a871 74db95e] msg
2020-11-18 11:57:44 - INFO     - Author: kas <kas@example.com>
2020-11-18 11:57:44 - INFO     - 1 file changed, 1 insertion(+), 1 deletion(-)
2020-11-18 11:57:44 - INFO     - /work/isar$ git apply /repo/isar-patches/0002-wic-Ensure-that-sourceparams-values-can-contain.patch
2020-11-18 11:57:44 - INFO     - Patch applied. (patch path: /repo/isar-patches/0002-wic-Ensure-that-sourceparams-values-can-contain.patch, repo: isar, patch entry: 02-wks-parse-issue)
2020-11-18 11:57:44 - INFO     - /work/isar$ git add -A
2020-11-18 11:57:44 - INFO     - /work/isar$ git commit -a --author kas <kas@example.com> -m msg
2020-11-18 11:57:44 - INFO     - [patched-bf0aff2af6dfb8f6729ffced8c73d4a5ac73a871 0f3f02c] msg
2020-11-18 11:57:44 - INFO     - Author: kas <kas@example.com>
2020-11-18 11:57:44 - INFO     - 1 file changed, 1 insertion(+), 1 deletion(-)
2020-11-18 11:57:44 - INFO     - /work/isar$ git apply /repo/isar-patches/0005-sdkchroot-Add-support-for-adding-self-defined-sdk-pa.patch
2020-11-18 11:57:44 - INFO     - Patch applied. (patch path: /repo/isar-patches/0005-sdkchroot-Add-support-for-adding-self-defined-sdk-pa.patch, repo: isar, patch entry: 05-add-sdk-pack)
2020-11-18 11:57:44 - INFO     - /work/isar$ git add -A
2020-11-18 11:57:44 - INFO     - /work/isar$ git commit -a --author kas <kas@example.com> -m msg
2020-11-18 11:57:44 - INFO     - [patched-bf0aff2af6dfb8f6729ffced8c73d4a5ac73a871 b512fd3] msg
2020-11-18 11:57:44 - INFO     - Author: kas <kas@example.com>
2020-11-18 11:57:44 - INFO     - 2 files changed, 16 insertions(+), 10 deletions(-)
2020-11-18 11:57:44 - INFO     - /work/isar$ git apply /repo/isar-patches/0006-sdk-Make-all-links-in-the-SDK-chroot-relative.patch
2020-11-18 11:57:44 - INFO     - Patch applied. (patch path: /repo/isar-patches/0006-sdk-Make-all-links-in-the-SDK-chroot-relative.patch, repo: isar, patch entry: 06-relative-sdk-links)
2020-11-18 11:57:44 - INFO     - /work/isar$ git add -A
2020-11-18 11:57:44 - INFO     - /work/isar$ git commit -a --author kas <kas@example.com> -m msg
2020-11-18 11:57:44 - INFO     - [patched-bf0aff2af6dfb8f6729ffced8c73d4a5ac73a871 7b89c8c] msg
2020-11-18 11:57:44 - INFO     - Author: kas <kas@example.com>
2020-11-18 11:57:44 - INFO     - 1 file changed, 14 insertions(+)
2020-11-18 11:57:44 - INFO     - /work/isar$ git apply /repo/isar-patches/0007-sdk-Add-script-to-relocate-SDK.patch
2020-11-18 11:57:44 - INFO     - Patch applied. (patch path: /repo/isar-patches/0007-sdk-Add-script-to-relocate-SDK.patch, repo: isar, patch entry: 07-relocate-sdk)
2020-11-18 11:57:44 - INFO     - /work/isar$ git add -A
2020-11-18 11:57:44 - INFO     - /work/isar$ git commit -a --author kas <kas@example.com> -m msg
2020-11-18 11:57:44 - INFO     - [patched-bf0aff2af6dfb8f6729ffced8c73d4a5ac73a871 2b42c44] msg
2020-11-18 11:57:44 - INFO     - Author: kas <kas@example.com>
2020-11-18 11:57:44 - INFO     - 2 files changed, 43 insertions(+)
[... 102 lines cut ...]
File: '/usr/lib/python3.7/subprocess.py', lineno: 1522, function: _execute_child
1518:                    if errno_num != 0:
1519:                        err_msg = os.strerror(errno_num)
1520:                        if errno_num == errno.ENOENT:
1521:                            err_msg += ': ' + repr(err_filename)
*** 1522:                    raise child_exception_type(errno_num, err_msg, err_filename)
1523:                raise child_exception_type(err_msg)
1524:
1525:
1526:        def _handle_exitstatus(self, sts, _WIFSIGNALED=os.WIFSIGNALED,
Exception: FileNotFoundError: [Errno 2] No such file or directory: 'envsubst': 'envsubst'

ERROR: Logfile of failure stored in: /work/build/tmp/work/isar-arm64/linux-iot2050/5.x-upstream-r0/temp/log.do_transform_template.256
ERROR: Task (/repo/recipes-kernel/linux/linux-iot2050_5.x-upstream.bb:do_transform_template) failed with exit code '1'
NOTE: Tasks Summary: Attempted 181 tasks of which 172 didn't need to be rerun and 3 failed.

Summary: 3 tasks failed:
/work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-target.bb:do_bootstrap
/work/isar/meta/recipes-bsp/u-boot-script/u-boot-script_1.1.bb:do_install_builddeps
/repo/recipes-kernel/linux/linux-iot2050_5.x-upstream.bb:do_transform_template
Summary: There were 3 ERROR messages shown, returning a non-zero exit code.
2020-11-18 12:11:08 - ERROR    - Command returned non-zero exit status 1
```

 I tried building older versions, but they are failing too.

jan-kiszka, 2020-11-18:
`kas-docker --isar build kas-iot2050-example.yml:kas/opt/upstream.yml` - not the **--isar**. kas 2.3 will address this common mistake (which I recently made myself).

BTW, kas/opt/upstream.yml is in flux, currently broken because I updated the kernel today, not yet the firmware.

ServiceISRI, 2020-11-18:
Without --isar it still fails that same way:

```
2020-11-18 12:38:28 - INFO     - kas 2.2 started
2020-11-18 12:38:28 - INFO     - /repo$ git rev-parse --show-toplevel
2020-11-18 12:38:28 - INFO     - /repo/kas/opt$ git rev-parse --show-toplevel
2020-11-18 12:38:28 - INFO     - /repo$ git rev-parse --show-toplevel
2020-11-18 12:38:28 - INFO     - Using /repo as root for repository meta-iot2050
2020-11-18 12:38:28 - INFO     - /work/isar$ git remote set-url origin https://g                                                                                                                                                             ithub.com/ilbers/isar
2020-11-18 12:38:28 - INFO     - /work/isar$ git cat-file -t bf0aff2af6dfb8f6729                                                                                                                                                             ffced8c73d4a5ac73a871
2020-11-18 12:38:28 - INFO     - Repository isar already contains bf0aff2af6dfb8                                                                                                                                                             f6729ffced8c73d4a5ac73a871 as commit
2020-11-18 12:38:28 - INFO     - /repo$ git rev-parse --show-toplevel
2020-11-18 12:38:28 - INFO     - Using /repo as root for repository meta-iot2050
2020-11-18 12:38:28 - INFO     - /work/isar$ git status -s
2020-11-18 12:38:28 - INFO     - /work/isar$ git rev-parse --verify -q origin/bf                                                                                                                                                             0aff2af6dfb8f6729ffced8c73d4a5ac73a871
2020-11-18 12:38:28 - INFO     - /work/isar$ git checkout -q bf0aff2af6dfb8f6729                                                                                                                                                             ffced8c73d4a5ac73a871
2020-11-18 12:38:28 - INFO     - /repo$ git rev-parse --show-toplevel
2020-11-18 12:38:28 - INFO     - Using /repo as root for repository meta-iot2050
2020-11-18 12:38:28 - INFO     - /work/isar$ /tmp/tmpi96ft0_1/get_bb_env /work/b                                                                                                                                                             uild
2020-11-18 12:38:28 - INFO     - /repo$ git rev-parse --show-toplevel
2020-11-18 12:38:28 - INFO     - Using /repo as root for repository meta-iot2050
2020-11-18 12:38:28 - INFO     - /work/isar$ git checkout -q -B patched-bf0aff2a                                                                                                                                                             f6dfb8f6729ffced8c73d4a5ac73a871
2020-11-18 12:38:28 - INFO     - /work/isar$ git apply /repo/isar-patches/0001-d                                                                                                                                                             pkg-raw-Allow-patching-and-templating.patch
2020-11-18 12:38:28 - INFO     - Patch applied. (patch path: /repo/isar-patches/                                                                                                                                                             0001-dpkg-raw-Allow-patching-and-templating.patch, repo: isar, patch entry: 01-d                                                                                                                                                             pgk-raw-patch)
2020-11-18 12:38:28 - INFO     - /work/isar$ git add -A
2020-11-18 12:38:28 - INFO     - /work/isar$ git commit -a --author kas <kas@exa                                                                                                                                                             mple.com> -m msg
2020-11-18 12:38:28 - INFO     - [patched-bf0aff2af6dfb8f6729ffced8c73d4a5ac73a8                                                                                                                                                             71 7d20cee] msg
2020-11-18 12:38:28 - INFO     - Author: kas <kas@example.com>
2020-11-18 12:38:28 - INFO     - 1 file changed, 1 insertion(+), 1 deletion(-)
2020-11-18 12:38:28 - INFO     - /work/isar$ git apply /repo/isar-patches/0002-w                                                                                                                                                             ic-Ensure-that-sourceparams-values-can-contain.patch
2020-11-18 12:38:28 - INFO     - Patch applied. (patch path: /repo/isar-patches/                                                                                                                                                             0002-wic-Ensure-that-sourceparams-values-can-contain.patch, repo: isar, patch en                                                                                                                                                             try: 02-wks-parse-issue)
2020-11-18 12:38:28 - INFO     - /work/isar$ git add -A
2020-11-18 12:38:28 - INFO     - /work/isar$ git commit -a --author kas <kas@exa                                                                                                                                                             mple.com> -m msg
[... cut ...]                    rap
  /repo/recipes-kernel/linux/linux-iot2050_5.x-upstream.bb:do_transform_template
  /work/isar/meta/recipes-bsp/u-boot-script/u-boot-script_1.1.bb:do_install_buil                                                                                                                                                             ddeps
Summary: There were 3 ERROR messages shown, returning a non-zero exit code.
2020-11-18 12:38:31 - ERROR    - Command returned non-zero exit status 1

```

jan-kiszka, 2020-11-18:
I meant "note the --isar": you must call kas **with** that extra switch.

ServiceISRI, 2020-11-26:
Forgot to mention: even with --isar it failed.

jan-kiszka, 2020-11-26:
Please share the log of `kas-docker --isar build kas-iot2050-example.yml` - it will be different from the one above.

ServiceISRI, 2020-11-26:
Hi,
ich reloaded the latest version from github and ran the buildscript you provided:
`
ERROR: Task (/work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-target.bb:do_bootstrap) failed with exit code '1'`
[log_docker_build_with_isar.txt](https://github.com/siemens/meta-iot2050/files/5602457/log_docker_build_with_isar.txt)

bergmanu, 2020-11-26:
Hi, 
I also just cloned the meta-iot2050 layer today and executed the upstream build with
`./kas-docker --isar build kas-iot2050-example.yml:kas/opt/upstream.yml`
 and this worked fine. 
At least the build process, didn't check the display issue

jan-kiszka, 2020-11-26:
The new problem manifests here already:
```
update-binfmts: warning: unable to close /proc/sys/fs/binfmt_misc/register: Invalid argument
update-binfmts: warning: unable to enable binary format qemu-aarch64
update-binfmts: exiting due to previous errors
```

There might be a problem with the host Linux not supporting [binfmt-misc](https://en.wikipedia.org/wiki/Binfmt_misc). What distribution and version are you on? Kernel 4.4 indicates it's a rather old one.

@bergmanu, do we have list of known-to-work host distros already?

bergmanu, 2020-11-26:
Nope, there is no list. I am using Ubuntu 18.04 with kernel 5.4 right now, but never tested another distro. 
I am not sure which distro Felipe is using, but probably the same environment like me.

ServiceISRI, 2020-11-26:
On my first build i was using 
`ubuntu 16.04.7 LTS (Xenial Xerus)`
with Kernel `4.4.0-194-generic`

Now i am running a build on 
`Ubuntu 20.04.1 LTS Focal Fossa`
with Kernel `5.4.0-52-generic `

I will report soon if the build succeeds.

ServiceISRI, 2020-11-27:
I ran the build yesterday and i have had some errors at the end:
```
Exception ignored when trying to write to the signal wakeup fd:
BlockingIOError: [Errno 11] Resource temporarily unavailable

```
So I tried it again on some other server.

`./kas-docker --isar build kas-iot2050-example.yml`

The image-file was **successfully build both times**.

[log_docker_buildserver.txt](https://github.com/siemens/meta-iot2050/files/5607137/log_docker_buildserver.txt)

uname -a
`Linux DL360pG8 4.19.0-12-amd64 #1 SMP Debian 4.19.152-1 (2020-10-18) x86_64 GNU/Linux`
Docker -v
`Docker version 19.03.13, build 4484c46d9d`
docker-compose -v
`docker-compose version 1.21.0, build unknown`

jan-kiszka, 2020-11-27:
That BlockingIOErrors above are harmless and unfortunately hard to resolve (we are struggling with them for a long time). The image is built fine irrespective of them.

Good that it works now.

closed 2020-11-27 by jan-kiszka

Girish-75020413, 2022-07-20:
Hello Jan

Issue is resolved now. Thanks for your quick support.

Regards,
Girish M

jan-kiszka, 2022-07-20:
Didn't you want to comment on #337 instead?