siemens/meta-iot2050 #57: MRAA failed to open gpio

simonabler, 2020-11-24:
This error happens sometimes if you want to set a output value via node-red.

`Nov 24 15:21:39 iot2050-debian libmraa[302]: gpio48: Failed to open 'value': No such file or directory`

Nodered is starting with:

`Nov 19 16:19:03 iot2050-debian start_nodered.sh[292]: Welcome to Node-RED
Nov 19 16:19:03 iot2050-debian start_nodered.sh[292]: ===================
Nov 19 16:19:03 iot2050-debian start_nodered.sh[292]: 19 Nov 16:19:03 - [info] Node-RED version: v1.0.6
Nov 19 16:19:03 iot2050-debian start_nodered.sh[292]: 19 Nov 16:19:03 - [info] Node.js  version: v10.21.0
Nov 19 16:19:03 iot2050-debian start_nodered.sh[292]: 19 Nov 16:19:03 - [info] Linux 4.19.59 arm64 LE
Nov 19 16:19:10 iot2050-debian start_nodered.sh[292]: 19 Nov 16:19:10 - [info] Loading palette nodes
Nov 19 16:19:18 iot2050-debian libmraa[302]: libmraa version v2.0.0 initialised by user 'root' with EUID 0
Nov 19 16:19:18 iot2050-debian libmraa[302]: gpio: platform doesn't support chardev, falling back to sysfs
Nov 19 16:19:18 iot2050-debian libmraa[302]: libmraa initialised for platform 'SIMATIC IOT2050' of type 21
Nov 19 16:19:18 iot2050-debian libmraa[302]: led: init: current user doesn't have access rights for using LED
Nov 19 16:19:18 iot2050-debian libmraa[302]: led: init: current user doesn't have access rights for using LED
Nov 19 16:19:18 iot2050-debian libmraa[302]: led: init: current user doesn't have access rights for using LED
Nov 19 16:19:18 iot2050-debian libmraa[302]: led: init: current user doesn't have access rights for using LED
`

Most of the times the gpios are working.. until this error happens. "Failed to open 'value'"
After that I have to restart nodered to get it working again.

jan-kiszka, 2020-11-24:
That's with the last release or with current master? Did you change anything, or is it a vanilla image?

simonabler, 2020-11-24:
i build the image with ./kas-docker --isar build kas-iot2050-example.yml
from commit 65b5aafeb8273fcee74054f8aad112287d46a872 

i changed the systemd startup with my own start_nodered.sh
=>
node-red --settings /home/user/.node-red/settings.js

For example: The gpio worked the last to days without problems, but today it stopped with the error. Failed to open 'value':

jan-kiszka, 2020-11-24:
Don't recall right now if we are running node-red under a separate account or as root. In any case, I do not recall having seen such a pattern yet. It might be a regression over the development tree, it might be related to local changes.

Would be worth to check in your case if the user that starts node-red and, thus, does the mraa accesses has permissions to play with the LEDs as well.

jan-kiszka, 2020-11-24:
Ok, blind: It was 'root' starting things here. Then it's not a classic permission thing.

jan-kiszka, 2020-11-24:
Is there only one instance of node-red running? Maybe it's a race condition... Maybe with board-configuration.service. Try disabling the latter for testing purposes (systemctl).

simonabler, 2020-11-24:
top -c -p $(pgrep -d',' -f node-red) => only one instance running

 systemctl status board-configuration.service => dead last run 5 days ago

nothing special

jan-kiszka, 2020-11-25:
@simonabler, we will follow up directly with you later.

Meanwhile, @BaochengSu, @johnson325, any ideas? I would suspect right now that we are somehow losing the gpio export for 48. No idea, though, what could cause this.

jan-kiszka, 2020-11-27:
Please have a look at https://github.com/siemens/meta-iot2050/tree/jan/mraa-node-red-fixes, if that resolves the issues you see.

simonabler, 2020-11-28:
Patched the old system with the new generated packages

mraa_2.0.0_arm64.deb
node-red-gpio_0.0.6-IOT2050_arm64.deb

I will let you know if this is working properly.
Thank you for your really fast help!!

jan-kiszka, 2020-12-02:
@simonabler No news hopefully means good news: Does this work so far?

simonabler, 2020-12-02:
@jan-kiszka That's right :) .. It is working, so far.

closed 2020-12-14 by jan-kiszka