siemens/meta-iot2050 #58: Publish IOT2050 GPIO nodes

bergmanu, 2020-11-25:
Can we publish the IOT2050 node-red nodes to nodered.org? 
The they can also be installed to the Industrial OS.

jan-kiszka, 2020-11-25:
Valid request. Precondition is resolving #23, and that will likely imply device tree changes first (GPIO naming).

jan-kiszka, 2021-01-18:
First experiment: https://github.com/siemens/linux/commit/3071cd632d80015f90868d2360d179b07aea955c. The external pins and also the user button are now listed in `gpioinfo`. If you need to switch the muxes, mraa is still needed, of course.

Webbeh, 2021-04-06:
Bump.

#23 being solved, anything left to do for this ?

jan-kiszka, 2021-04-06:
The GPIO resolution has been improved in current development head. We still need to push the required changes to mraa. Those will have to be accepted, and we will need a locally built version (as it comes with meta-iot2050) for a while. But once they are accepted, we could also publish our Node-RED nodes via npmjs.com.

Webbeh, 2021-07-23:
Any news on this ?

jan-kiszka, 2021-07-23:
Slowly progressing on the mraa side but still no break-through, sorry.

Webbeh, 2021-07-23:
Anything we can do in the meantime?

jan-kiszka, 2021-07-23:
...to achieve what exactly?

Webbeh, 2021-07-23:
Sorry. To either install them manually if they're not installed or... well, to help you ? :)

jan-kiszka, 2021-07-23:
You can build the node-red-gpio package via meta-iot2050 and install it on a device that lacks it. IIRC, the IndOS has our mraa package already, right? Then this should be sufficient (in addition to adding node-red itself). Maybe you will have to move things around, put node-red-node-intel-gpio in the right node_modules folder that your own node-red installation finds.

Regarding the to-do list: The major topics are
- refactor and clean up mraa board enabling patches for IOT2050 (most fixes are upstream now)
- sort out whether the "drive" extension of the GPIO sysfs interface we need can be upstreamed to Linux

The latter may not be a blocker, though we also have #135.

The former is a rather fuzzy task and also involves internal clarifications. I started that but then got more urgent tasks last week.

bergmanu, 2023-01-12:
Nodes are available now: 
https://flows.nodered.org/node/node-red-node-intel-gpio

closed 2023-01-12 by bergmanu