siemens/meta-iot2050 #62: libmraa doesnt find libjson-c.so.3

seclari, 2020-12-08:
This error happens, if you want to include mraa in Eclipse when using the SDK build with "./kas-docker --isar build kas-iot2050-example.yml:kas/opt/sdk.yml" on 07.12.2020:

warning: libjson-c.so.3, needed by /home/edge/IOT2050_Images/sdk-isar-arm64/lib/aarch64-linux-gnu/libmraa.so, not found (try using -rpath or -rpath-link)
/home/edge/IOT2050_Images/sdk-isar-arm64/lib/aarch64-linux-gnu/libmraa.so: undefined reference to `json_tokener_parse'

It seams that libjson-c.so.3 does not exist. Searching the folder "sdk-isar-arm64" the libjson is also not found.

Total output in the console in Eclipse:

make all 
Building target: Test
Invoking: Cross G++ Linker
aarch64-linux-gnu-g++ -lmraa --sysroot=/home/edge/IOT2050_Images/sdk-isar-arm64 -o "Test"  ./src/Test.o   
/home/edge/IOT2050_Images/sdk-isar-arm64/usr/bin/../lib/gcc-cross/aarch64-linux-gnu/8/../../../../aarch64-linux-gnu/bin/../../bin/aarch64-linux-gnu-ld.bfd.bin: warning: libjson-c.so.3, needed by /home/edge/IOT2050_Images/sdk-isar-arm64/lib/aarch64-linux-gnu/libmraa.so, not found (try using -rpath or -rpath-link)
/home/edge/IOT2050_Images/sdk-isar-arm64/usr/bin/../lib/gcc-cross/aarch64-linux-gnu/8/../../../../aarch64-linux-gnu/bin/../../bin/aarch64-linux-gnu-ld.bfd.bin: /home/edge/IOT2050_Images/sdk-isar-arm64/lib/aarch64-linux-gnu/libmraa.so: undefined reference to `json_tokener_parse'
/home/edge/IOT2050_Images/sdk-isar-arm64/usr/bin/../lib/gcc-cross/aarch64-linux-gnu/8/../../../../aarch64-linux-gnu/bin/../../bin/aarch64-linux-gnu-ld.bfd.bin: /home/edge/IOT2050_Images/sdk-isar-arm64/lib/aarch64-linux-gnu/libmraa.so: undefined reference to `json_object_array_get_idx'
/home/edge/IOT2050_Images/sdk-isar-arm64/usr/bin/../lib/gcc-cross/aarch64-linux-gnu/8/../../../../aarch64-linux-gnu/bin/../../bin/aarch64-linux-gnu-ld.bfd.bin: /home/edge/IOT2050_Images/sdk-isar-arm64/lib/aarch64-linux-gnu/libmraa.so: undefined reference to `json_object_get_boolean'
/home/edge/IOT2050_Images/sdk-isar-arm64/usr/bin/../lib/gcc-cross/aarch64-linux-gnu/8/../../../../aarch64-linux-gnu/bin/../../bin/aarch64-linux-gnu-ld.bfd.bin: /home/edge/IOT2050_Images/sdk-isar-arm64/lib/aarch64-linux-gnu/libmraa.so: undefined reference to `json_object_put'
/home/edge/IOT2050_Images/sdk-isar-arm64/usr/bin/../lib/gcc-cross/aarch64-linux-gnu/8/../../../../aarch64-linux-gnu/bin/../../bin/aarch64-linux-gnu-ld.bfd.bin: /home/edge/IOT2050_Images/sdk-isar-arm64/lib/aarch64-linux-gnu/libmraa.so: undefined reference to `json_object_is_type'
/home/edge/IOT2050_Images/sdk-isar-arm64/usr/bin/../lib/gcc-cross/aarch64-linux-gnu/8/../../../../aarch64-linux-gnu/bin/../../bin/aarch64-linux-gnu-ld.bfd.bin: /home/edge/IOT2050_Images/sdk-isar-arm64/lib/aarch64-linux-gnu/libmraa.so: undefined reference to `json_object_get_int'
/home/edge/IOT2050_Images/sdk-isar-arm64/usr/bin/../lib/gcc-cross/aarch64-linux-gnu/8/../../../../aarch64-linux-gnu/bin/../../bin/aarch64-linux-gnu-ld.bfd.bin: /home/edge/IOT2050_Images/sdk-isar-arm64/lib/aarch64-linux-gnu/libmraa.so: undefined reference to `json_object_get_string'
/home/edge/IOT2050_Images/sdk-isar-arm64/usr/bin/../lib/gcc-cross/aarch64-linux-gnu/8/../../../../aarch64-linux-gnu/bin/../../bin/aarch64-linux-gnu-ld.bfd.bin: /home/edge/IOT2050_Images/sdk-isar-arm64/lib/aarch64-linux-gnu/libmraa.so: undefined reference to `json_object_array_length'
/home/edge/IOT2050_Images/sdk-isar-arm64/usr/bin/../lib/gcc-cross/aarch64-linux-gnu/8/../../../../aarch64-linux-gnu/bin/../../bin/aarch64-linux-gnu-ld.bfd.bin: /home/edge/IOT2050_Images/sdk-isar-arm64/lib/aarch64-linux-gnu/libmraa.so: undefined reference to `json_object_object_get_ex'
collect2: error: ld returned 1 exit status
make: *** [makefile:47: Test] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

jan-kiszka, 2020-12-08:
I suppose this is an SDK built with latest master, right? We might be missing further dependencies for it. Please retry with

```
diff --git a/conf/distro/isar.conf b/conf/distro/isar.conf
index 2f09eaf..5d23882 100644
--- a/conf/distro/isar.conf
+++ b/conf/distro/isar.conf
@@ -51,4 +51,4 @@ HOST_DISTRO_APT_SOURCES_remove_packages-snapshot = "${DISTRO_APT_SOURCES_MAINLIN
 HOST_DISTRO_APT_PREFERENCES += "conf/distro/preferences.upstream-libssl"
 
 SDK_INSTALL += "linux-headers-${KERNEL_NAME} mraa"
-SDK_PREINSTALL += "zlib1g-dev"
+SDK_PREINSTALL += "zlib1g-dev libjson-c-dev"
```

There might be more packages missing - same pattern. Once things work as desired, please share what you tuned, then we will update the recipes (or send pull requests directly).

seclari, 2020-12-09:
Yes, the SDK was built with the latest master.
I've tried your proposed changes. 
I still keep the same mistake. libjson-c.so.3 could not be found. 
I searched for libjson and only found it in sdk-isar-arm43/lib/x86_64-linux-gnu.  I find the libmraa.so in sdk-isar-arm43/lib/aarch64-linux-gnu. So they are not in the same folder. I tried to incldue the path of libjson for linking but it didnt work. 
Could it be that there is only a 32-bit-version of libjson and that I need a 64-bit-version? How would i get it?

jan-kiszka, 2020-12-14:
Tested now: we were missing the explicit specification of the architecture.

commit 726c37a5 (2020-12-14): sdk: Add libjson-c, fix zlib arch

libjson-c is needed for linking against mraa. And zlib1g-dev should come
for the target architecture.

Closes: #62
PR #64 (2020-12-15): sdk: Add libjson-c, fix zlib arch
libjson-c is needed for linking against mraa. And zlib1g-dev should come
for the target architecture.

Closes: #62
commit 503d7104 (2020-12-15): sdk: Add libjson-c, fix zlib arch

libjson-c is needed for linking against mraa. And zlib1g-dev should come
for the target architecture.

Closes: #62
closed 2020-12-15 by BaochengSu