siemens/meta-iot2050 #72: Build image native on arm64 (M1)

gramss, 2020-12-22:
Hey,

I tried to build this on a M1-based Mac, native with the latest Docker-Preview and brew.

Therefore I first had to build siemens/kas as an arm64 image (including small naming changes in the pkg-names installed via `apt` in the Dockerfile).

These where the changes I had to do for now:

```
diff --git a/kas-container b/kas-container
index 232461c..5ae59ab 100755
--- a/kas-container
+++ b/kas-container
@@ -81,8 +81,10 @@ enable_isar_mode() {
        esac
 }
 
-KAS_IMAGE_VERSION_DEFAULT="2.3.3"
-KAS_CONTAINER_IMAGE_PATH_DEFAULT="ghcr.io/siemens/kas"
+# KAS_IMAGE_VERSION_DEFAULT="2.3.3"
+KAS_IMAGE_VERSION_DEFAULT="latest"
+# KAS_CONTAINER_IMAGE_PATH_DEFAULT="ghcr.io/siemens/kas"
+KAS_CONTAINER_IMAGE_PATH_DEFAULT="iotbot"
 KAS_CONTAINER_IMAGE_NAME_DEFAULT="kas"
 
 set_container_image_var() {
diff --git a/kas/iot2050.yml b/kas/iot2050.yml
index 40b8fc4..a6edfe9 100644
--- a/kas/iot2050.yml
+++ b/kas/iot2050.yml
@@ -41,4 +41,4 @@ local_conf_header:
   standard: |
     CONF_VERSION = "1"
   crossbuild: |
-    ISAR_CROSS_COMPILE = "1"
+    ISAR_CROSS_COMPILE = "0"
diff --git a/recipes-core/images/iot2050-image-example.bb b/recipes-core/images/iot2050-image-example.bb
index 6c440c8..1024a93 100644
--- a/recipes-core/images/iot2050-image-example.bb
+++ b/recipes-core/images/iot2050-image-example.bb
@@ -112,3 +112,7 @@ IMAGE_INSTALL += " \
     node-red-node-serialport \
     node-red-node-sqlite \
     "
+
+IMAGE_INSTALL += "linux-headers-${KERNEL_NAME}"
+
+ISAR_CROSS_COMPILE = "0"
```

And I also had to alter this part for Isar `sdkchroot` - the if condition: (probably should have stoped here, but the rest of the .bb looked like it can handle other arch's like RISC-V)
```
python() {
    if d.getVar("HOST_ARCH") not in ['i386', 'amd64'] and d.getVar('ISAR_CROSS_COMPILE', True) == '1':
        raise bb.parse.SkipRecipe("SDK doesn't support {} as host".format(
            d.getVar("ROOTFS_ARCH")))
}
```
In file: isar/meta/recipes-devtools/sdkchroot/sdkchroot.bb

But this is only how far I got. I can install `reprepro` via brew, but I guess that `isar-apt` expects ubuntu/debian and not MacOs..? The `tcf-agent` part just happened as I wrote this issue.. I saw it now for the first time after running it a couple times.. Might not be the biggest problem here.
```
2020-12-22 03:57:34 - INFO     - kas 2.3.3 started
2020-12-22 03:57:34 - INFO     - /repo$ git rev-parse --show-toplevel
2020-12-22 03:57:34 - INFO     - /repo$ git rev-parse --show-toplevel
2020-12-22 03:57:34 - INFO     - Using /repo as root for repository meta-iot2050
2020-12-22 03:57:34 - INFO     - /work/isar$ git remote set-url origin https://github.com/ilbers/isar
2020-12-22 03:57:34 - INFO     - /work/isar$ git cat-file -t 76eac233f1b1d0d117b3f05b3dec8a6da1cdf4c7
2020-12-22 03:57:34 - INFO     - Repository isar already contains 76eac233f1b1d0d117b3f05b3dec8a6da1cdf4c7 as commit
2020-12-22 03:57:34 - INFO     - /repo$ git rev-parse --show-toplevel
2020-12-22 03:57:34 - INFO     - Using /repo as root for repository meta-iot2050
2020-12-22 03:57:34 - INFO     - /work/isar$ git status -s
2020-12-22 03:57:35 - INFO     - M meta/recipes-devtools/sdkchroot/sdkchroot.bb
2020-12-22 03:57:35 - WARNING  - Repo isar is dirty - no checkout
2020-12-22 03:57:35 - INFO     - /repo$ git rev-parse --show-toplevel
2020-12-22 03:57:35 - INFO     - Using /repo as root for repository meta-iot2050
2020-12-22 03:57:35 - INFO     - /work/isar$ /tmp/tmp3zn2p2ba/get_bb_env /work/build
2020-12-22 03:57:35 - INFO     - /repo$ git rev-parse --show-toplevel
2020-12-22 03:57:35 - INFO     - Using /repo as root for repository meta-iot2050
2020-12-22 03:57:35 - INFO     - /repo$ git rev-parse --show-toplevel
2020-12-22 03:57:35 - INFO     - Using /repo as root for repository meta-iot2050
2020-12-22 03:57:35 - INFO     - /work/build$ /work/isar/bitbake/bin/bitbake -k -c build iot2050-image-example
Parsing recipes: 100% |#####################################################################################| Time: 0:00:01
Parsing of 49 .bb files complete (0 cached, 49 parsed). 49 targets, 0 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
Initialising tasks: 100% |##################################################################################| Time: 0:00:00
NOTE: No setscene tasks
NOTE: Executing Tasks
ERROR: isar-apt-1.0-r0 do_cache_config: Execution of '/work/build/tmp/work/isar-arm64/isar-apt/1.0-r0/temp/run.do_cache_config.147' failed with exit code 127:
/work/build/tmp/work/isar-arm64/isar-apt/1.0-r0/temp/run.do_cache_config.147: 55: /work/build/tmp/work/isar-arm64/isar-apt/1.0-r0/temp/run.do_cache_config.147: reprepro: not found
WARNING: exit code 127 from a shell command.

ERROR: Logfile of failure stored in: /work/build/tmp/work/isar-arm64/isar-apt/1.0-r0/temp/log.do_cache_config.147
Log data follows:
| DEBUG: Executing shell function do_cache_config
| /work/build/tmp/work/isar-arm64/isar-apt/1.0-r0/temp/run.do_cache_config.147: 55: /work/build/tmp/work/isar-arm64/isar-apt/1.0-r0/temp/run.do_cache_config.147: reprepro: not found
| WARNING: exit code 127 from a shell command.
| ERROR: Execution of '/work/build/tmp/work/isar-arm64/isar-apt/1.0-r0/temp/run.do_cache_config.147' failed with exit code 127:
| /work/build/tmp/work/isar-arm64/isar-apt/1.0-r0/temp/run.do_cache_config.147: 55: /work/build/tmp/work/isar-arm64/isar-apt/1.0-r0/temp/run.do_cache_config.147: reprepro: not found
| WARNING: exit code 127 from a shell command.
| 
ERROR: Task (/work/isar/meta/recipes-devtools/isar-apt/isar-apt.bb:do_cache_config) failed with exit code '1'
WARNING: tcf-agent-1.7.0-dc373fa9-r0 do_fetch: Failed to fetch URL git://github.com/eclipse/tcf.agent.git;protocol=https, attempting MIRRORS if available
ERROR: tcf-agent-1.7.0-dc373fa9-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; unset _PYTHON_SYSCONFIGDATA_NAME; export GIT_PROXY_COMMAND="oe-git-proxy"; export NO_PROXY="*"; export PATH="/work/isar/scripts:/work/isar/bitbake/bin:/usr/sbin:/usr/bin:/sbin:/bin"; export HOME="/tmp/tmp60jjto7s"; LANG=C git -c core.fsyncobjectfiles=0 clone --bare --mirror https://github.com/eclipse/tcf.agent.git /work/build/downloads/git/github.com.eclipse.tcf.agent.git --progress failed with exit code 128, output:
Cloning into bare repository '/work/build/downloads/git/github.com.eclipse.tcf.agent.git'...
remote: Enumerating objects: 2393, done.        
remote: Counting objects: 100% (2393/2393), done.        
remote: Compressing objects: 100% (2125/2125), done.        
error: RPC failed; curl 56 GnuTLS recv error (-54): Error in the pull function.
fatal: the remote end hung up unexpectedly
fatal: early EOFs:  75% (19966/26621), 8.00 MiB | 1.73 MiB/s   
fatal: index-pack failed

ERROR: tcf-agent-1.7.0-dc373fa9-r0 do_fetch: Fetcher failure for URL: 'git://github.com/eclipse/tcf.agent.git;protocol=https'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /work/build/tmp/work/isar-arm64/tcf-agent/1.7.0-dc373fa9-r0/temp/log.do_fetch.256
ERROR: Task (/repo/recipes-app/tcf-agent/tcf-agent.bb:do_fetch) failed with exit code '1'
Currently  1 running tasks (68 of 288)  23% |#################                                                            |
0: linux-iot2050-4.19.59+-r0 do_fetch (pid 258)   0% |                                                                    |
```

Would be cool to see if isar also works without cross-compiling. Could also help to see if #13 could be more achieve-able by that..

jan-kiszka, 2020-12-23:
I'm planning to test native ARM builds eventually, maybe on an RPi4 or in a QEMU VM. Most of the changes you did  should be needed, except fo the sdk-related ones. But there might be bugs as this path wasn't tested yet.

Mac is not native, though. You need a Debian host - or a Debian container. But the latter also seems to have limitations on Mac, see https://github.com/siemens/kas/issues/38. So better start simple, using Linux as host or VM.

jan-kiszka, 2021-01-01:
Not fully tested yet, but at least one piece of the puzzle is resolved: https://groups.google.com/g/isar-users/c/ZsjeCrxEk6Y

I'm currently building inside a QEMU arm64 Debian VM (my RPi4 does not have enough RAM). That's so far without kas-container, just via `kas build /meta-iot2050/kas-iot2050-example.yml'. To enable also the container build, we need to provide an arm64 instance of kas-isar. I have no clue yet, though, how to generate and ship such a (multiarch?) container.

Note that I did not touch `ISAR_CROSS_COMPILE` - it's a nop by default if host==target.

jan-kiszka, 2021-01-02:
Container based build seems to work now, just set `KAS_IMAGE_VERSION=next` prior to calling `kas-container`.

jan-kiszka, 2021-01-02:
...oh, and you need https://github.com/siemens/meta-iot2050/commits/jan/patches.

BaochengSu, 2021-01-18:
@jan-kiszka Does #81 is targeting on this issue?

jan-kiszka, 2021-01-18:
Yes, the first two patches of that MR.

jan-kiszka, 2021-03-02:
Master now comes with kas 2.4 which supports arm64-hosted builds.

closed 2021-03-02 by jan-kiszka

CanNuhlar, 2023-02-24:
Hi Jan,

I'm also trying to achieve the same thing. Current kas seems to be version 3.2.2 which should work via `KAS_IMAGE_VERSION=next` although that's not the case. I'm also hitting the reprepro issue. 

Am I missing something or the instructions are outdated?

Thanks!

jan-kiszka, 2023-02-25:
On some RPi 4, 64-bit Debian bullseye base system:
```
sudo apt install git docker.io
sudo usermod -aG docker <user>
<logout, login>
git clone https://github.com/siemens/meta-iot2050
cd meta-iot2050
./kas-container build kas-iot2050-boot-pg2.yml
```
That still works. It is kas 3.1 so far, and I only partially tested a image building (slow SD card on the RPi...). If you see problems, please open up a new issue.