siemens/meta-iot2050 #86: Containerized SDK

Silvanoc, 2021-01-18:
The goal of this issue is gathering **some feedback from your side** WRT a new functionality in ISAR: _creation of **container images providing the SDK** built with ISAR and distributing them using a container registry_.

I got from @jan-kiszka the hint that I could demonstrate the functionality based in this project. I see the need of this functionality on ISAR users (like you). Therefore I thought that rather than convincing the ISAR developers/maintainers of the advantages of this functionality, I'll use your feedback to confirm that this is a meaningful functionality and feed this information back to the ISAR community.

NOTE 1: This is currently only a **PoC** (proof-of-concept) of a functionality that will hopefully get integrated into ISAR!

NOTE 2: The code that I'm linking to in this issue is not ready yet for integration!

### How to use the containerized SDK

1. Go the directory containing the source code that you want to build
2. Run a container based on the SDK container image with `docker run --rm -ti --volume $(pwd):/build ghcr.io/silvanoc/meta-iot2050/isar-sdk-arm64:latest`.
3. From the container you can start using the SDK.

### Building the containerized SDK

I've documented how to build the SDK container image in the document [BUILDING-SDK-CONTAINER.md](https://github.com/Silvanoc/meta-iot2050/blob/master/kas/BUILDING-SDK-CONTAINER.md). The images are being [automatically built using CI (GitHub Actions)](https://github.com/Silvanoc/meta-iot2050/runs/1716315673?check_suite_focus=true)!

The whole building process is based on still ongoing contribution efforts from my side to ISAR and KAS, therefore patching ISAR and pulling a KAS image from my personal area. Once the contributions are through, they should be come part of upstream ISAR and KAS.

If you like this functionality as an ISAR user, please don't hesitate showing your interest in the [corresponding mailing list discussion](https://groups.google.com/g/isar-users/c/l8oqKpQQ0j0).

jan-kiszka, 2021-01-18:
Very nice demo! I guess I need to give that a try soon.

Some remarks:
 - you likely miss some packages in the SDK because it builds way too fast (but would otherwise be terminated by github limits) and is supposed to have also linux-headers-iot2050 included
 - SDK_FORMATS could join the other SDK_* in conf/distro/isar.conf (yeah, isar is a bad name here, see also my Ubuntu queue)
 - once upstream Isar indicates acceptance of the approach, the patch to add the additional packages to kas-isar could be submitted for kas upstream

Silvanoc, 2021-01-18:
> Very nice demo! I guess I need to give that a try soon.
Thanks!
> 
> Some remarks:
> 
>     * you likely miss some packages in the SDK because it builds way too fast (but would otherwise be terminated by github limits) and is supposed to have also linux-headers-iot2050 included
I didn't know which was the right configuration and have taken `kas/iot2050.yml`. That's the issue.

**Which one should I take?** 
> 
>     * SDK_FORMATS could join the other SDK_* in conf/distro/isar.conf (yeah, isar is a bad name here, see also my Ubuntu queue)
I've used a KAS configuration file because it was the easiest for me without thinking about the "right place", what's fine for a PoC.

Your remark very valuable to place it at the right place, when I massage if for contribution.
> 
>     * once upstream Isar indicates acceptance of the approach, the patch to add the additional packages to kas-isar could be submitted for kas upstream

Of course, it's in the pipeline!

I just didn't want to trigger a parallel integration process in KAS until the ISAR integration is done. I find it's fine for this demo to use the image from my personal GitHub space.

jan-kiszka, 2021-01-18:
> 
> I didn't know which was the right configuration and have taken `kas/iot2050.yml`. That's the issue.
> 
> **Which one should I take?**

See README :smile:: kas-iot2050-example.yml.

Silvanoc, 2021-01-18:
> > I didn't know which was the right configuration and have taken `kas/iot2050.yml`. That's the issue.
> > **Which one should I take?**
> 
> See README smile: kas-iot2050-example.yml.

Right, sorry. My fingers were faster than my brain :grimacing: It's building right now.

BTW running for 1h is not an issue, I've in fact built `kas-iot2050-example.yml` a couple of times using GitHub Actions :wink:

jan-kiszka, 2021-01-18:
Really? That would be very valuable news. I once tried that, see https://github.com/siemens/meta-iot2050/commits/jan/workflow, but it ran into timeouts, already for the first image.

Silvanoc, 2021-01-18:
> Really? That would be very valuable news. I once tried that, see https://github.com/siemens/meta-iot2050/commits/jan/workflow, but it ran into timeouts, already for the first image.

Follow [this workflow](https://github.com/Silvanoc/meta-iot2050/runs/1722933611) and you'll get your answer :wink:

Silvanoc, 2021-01-18:
A new image providing the SDK container built from the right configuration is available now. It's taken 1h 25min to build, but it's done :tada:

@jan-kiszka you can give it a try now following [this](https://github.com/Silvanoc/meta-iot2050/blob/master/kas/BUILDING-SDK-CONTAINER.md#running-the-sdk) and whatever else you want to test.

Silvanoc, 2021-01-25:
Any has been able to test it? Some feedback would be valuable.

jan-kiszka, 2021-01-25:
Not yet, currently working around npm packaging issues, once again. Will try once that is done.

jan-kiszka, 2021-01-28:
Just rebased over master - works great!

Would you move forward and submit a v2 of the Isar patch(es)?

Silvanoc, 2021-01-28:
> Would you move forward and submit a v2 of the Isar patch(es)?

Sure

jan-kiszka, 2021-06-01:
Merged into master, thanks for enabling this!

closed 2021-06-01 by jan-kiszka