siemens/meta-iot2050 #90: Error when building

AshWhitear, 2021-01-22:
hey there, any chance you guys know how to resolve this: 

`ERROR: Task (/repo/recipes-app/npm/npm-buildchroot_6.10.3+ds.bb:do_install_builddeps) failed with exit code '1'
`
getting it when running:

`./kas-container build kas-iot2050-example.yml`

Thanks, Ash.

BaochengSu, 2021-01-22:
Could you share the log file: `build/tmp/work/isar-arm64/npm-buildchroot/6.10.3+ds-r0/temp/log.do_install_builddeps`?

AshWhitear, 2021-01-22:
Absolutely, thank you for such a quick response: 

[log.txt](https://github.com/siemens/meta-iot2050/files/5853460/log.txt)

BaochengSu, 2021-01-22:
Looks like some node-package in backports repo breaks the nodes dependency.

For a quick workaround, try to add `node-tar` to `conf/distro/preferences.npm-backport`

see [this commit](https://github.com/siemens/meta-iot2050/commit/babdf89d6fed40182fcc1589140054ae4f2b9d76) on [my branch](https://github.com/siemens/meta-iot2050/tree/su/npm-fix). At least the npm build is passed, still watching other builds.

BaochengSu, 2021-01-22:
Oh, forget to mention, you might need perform a clean build.

AshWhitear, 2021-01-22:
all clean, just building now..

BaochengSu, 2021-01-22:
Seems we need fix more, build failed at image rootfs install stage. Would take more time than I thought to debug this.

May I ask if you can use the package lock build? The latest tested snapshot build is 2 weeks ago, see [this commit](https://github.com/siemens/meta-iot2050/commit/e0ba820101b6d9364d92f43214ce0532def03ff3) for reference. Then use `./kas-container build kas-iot2050-example.yml:kas/opt/package-lock.yml` to build the package lock version. It uses debian packages from 2 weeks ago for building, if that suits your requirement.

AshWhitear, 2021-01-22:
just hit exactly the same error, thank you so much for trying though.

no worries at all, ill kick off a package lock build now - can i still get the kernal-headers included with this build appending: `linux-headers-${KERNEL_NAME} \` to `IMAGE_INSTALL` in the `recipes-core/images/iot2050-image-example.bb` file?

BaochengSu, 2021-01-22:
@jan-kiszka The npm package in buster-backports has been upgraded to 7.4.0, see [1] & [2]. So maybe we can move forward for a4c9ca8fda7c6cb4aa743cbfa5962f8e978ac1dc ?

[1] https://metadata.ftp-master.debian.org/changelogs//main/n/npm/npm_7.4.0+ds-1~bpo10+2_changelog
[2] https://packages.debian.org/buster-backports/npm

jan-kiszka, 2021-01-22:
Would be worth a try, yes.

This npm parenting is continuous "fun". Waiting for the day it gets better.

AshWhitear, 2021-01-22:
just got so close to a successful build then: `ERROR: Task (/repo/recipes-core/images/iot2050-image-example.bb:do_rootfs_install) failed with exit code '1'`

any ideas?

AshWhitear, 2021-01-22:
the repos that cant be found can be found if they have a slash on the end, not sure if that helps?

jan-kiszka, 2021-01-22:
"Could not connect to snapshot.debian.org:80 (185.17.185.185), connection timed out Could not connect to snapshot.debian.org:80 (193.62.202.27), connection timed out [IP: 185.17.185.185 80]" - that's the problem with building agains snapshot.debian.org, it's rather unreliable (or does throttling). Retry the build, specifically as we are caching already fetched packages under build/downloads. It should succeed. Eventually...

AshWhitear, 2021-01-22:
I switched up my connection and seems to have gotten further, any thoughts on this one?

`Get:236 http://snapshot.debian.org/archive/debian-security/20200117T184524Z buster/updates/main arm64 openjdk-11-jdk-headless arm64 11.0.5+10-1~deb10u1 [205 MB]
Err:236 http://snapshot.debian.org/archive/debian-security/20200117T184524Z buster/updates/main arm64 openjdk-11-jdk-headless arm64 11.0.5+10-1~deb10u1
  Undetermined Error [IP: 193.62.202.27 80]
Fetched 151 MB in 21min 48s (115 kB/s)
E: Failed to fetch http://snapshot.debian.org/archive/debian-security/20200117T184524Z/pool/updates/main/o/openjdk-11/openjdk-11-jdk-headless_11.0.5+10-1~deb10u1_arm64.deb  Undetermined Error [IP: 193.62.202.27 80]
E: Some files failed to download
WARNING: exit code 100 from a shell command.`

jan-kiszka, 2021-01-22:
openjdk is rather large package. So the risk is higher to hit a failure of snapshot.debian.org while fetching it. Retry.

AshWhitear, 2021-01-22:
okay will keep at it, thank you guys again for all the help - its really appreciated.

AshWhitear, 2021-01-25:
quick update, after many tries is build all good - for any one else, **tl;dr**

To build your image until this issue is closed, use:
`./kas-container build kas-iot2050-example.yml:kas/opt/package-lock.yml`

Be persistent, any snapshot.debian errors are temporary, but linux is forever.

jan-kiszka, 2021-01-25:
Good to know.

I made no progress yet to gett latest build again. At least I filed https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=980928.

BaochengSu, 2021-01-25:
Besides our cached packages inside build/downloads, you can also use the `apt-cacher-ng` or similar tools to setup a cache proxy so that you won't have to fetch the snapshot.debian.org every time, even if you delete the build/downloads.

Details plz refer to [here](https://hub.docker.com/r/sameersbn/apt-cacher-ng). 

You may need to change the url from `http://snapshot.debian.org` to `http://<your-cacher-server-ip>:<port>/snapshot.debian.org` in `conf/distro/debian-snapshot.list`.

jan-kiszka, 2021-01-28:
Forgot to annotate: Fixed by https://github.com/siemens/meta-iot2050/pull/92 - hopefully for a while.

closed 2021-01-28 by jan-kiszka