siemens/meta-iot2050 #93: Error while building

hannes56a, 2021-01-28:
Hi,

I tried to build the image, but it fails very early with this error:

 

    2021-01-28 14:24:56 - INFO     - kas 2.3.3 started
    2021-01-28 14:24:56 - INFO     - /repo$ git rev-parse --show-toplevel
    2021-01-28 14:24:56 - INFO     - /repo/kas/opt$ git rev-parse --show-toplevel
    2021-01-28 14:24:56 - ERROR    - [Errno 13] Permission denied: '/work/build'
    Traceback (most recent call last):
    File "/usr/local/lib/python3.7/dist-packages/kas/kas.py", line 182, in main
    sys.exit(kas(sys.argv[1:]))
    File "/usr/local/lib/python3.7/dist-packages/kas/kas.py", line 171, in kas
    plugin.run(args)
    File "/usr/local/lib/python3.7/dist-packages/kas/plugins/build.py", line 83, in run
    macro.run(ctx, args.skip)
    File "/usr/local/lib/python3.7/dist-packages/kas/libcmds.py", line 96, in run
    command.execute(ctx)
    File "/usr/local/lib/python3.7/dist-packages/kas/libcmds.py", line 191, in execute
    os.mkdir(ctx.build_dir)
    PermissionError: [Errno 13] Permission denied: '/work/build'

 So I tried with "sudo", but it fails with

    Error: Running as root - may break certain recipes.
    Better give a regular user docker access. Set KAS_ALLOW_ROOT=yes to override.

 

I tried "export KAS_ALLOW_ROOT=yes" but i doesnt work :(

"echo $KAS_ALLOW_ROOT" says "yes"

 unknown user

    $ echo $KAS_ALLOW_ROOT
    yes

 

Host system: Ubuntu 18.04 LTS

Any hint for me?

Thanks a lot for reading

jan-kiszka, 2021-01-28:
What was the exact command you used to start the build?

hannes56a, 2021-01-28:
I tried these three, always the same error:

./kas-container build kas-iot2050-example.yml:kas/opt/package-lock.yml

./kas-container build kas-iot2050-example.yml

./kas-docker build kas-iot2050-example.yml

jan-kiszka, 2021-01-28:
OK, then it must be something specific to the host configuration. Did you follow the setup instructions according to https://support.industry.siemens.com/tf/ww/en/posts/how-to-build-github-example-image/237116/?page=0&pageSize=10 - except for using 18.04, rather than 20.04? Could you also try with 20.04?

hannes56a, 2021-01-28:
Hi, many thanks for the quick answers.
I used my normal Ubuntu (not VM) and started the instructions starting at point 3.

I will try it with 20.04.

hannes56a, 2021-01-28:
Hi again, looks like i found **### my** failure.
Not sure, but i think i used sudo in point 7 and/or 9 of the instructions...
So these directories had the wrong permissions i think.
I did again now in a "fresh" directory and it works for now (not ready yet)

jan-kiszka, 2021-01-28:
Good to know! Building will take some time. Please close once that succeeded.

hannes56a, 2021-01-29:
Hi,
at see end i´ve this error message (about 20 times):

[image]

Is that normal. If i repeat builing (its very fast than) the error messages are gone.

jan-kiszka, 2021-01-29:
It was "normal" and harmless: https://github.com/siemens/kas/commit/87af46ff8cb62e246ef20a4faa00a4180af58e49. Will be fixed with the next kas release.

hannes56a, 2021-01-29:
Thanks a lot!

closed 2021-01-29 by hannes56a