siemens/meta-iot2050 #97: ERROR libusb_control_transfer timeout

AshWhitear, 2021-02-02:
Hey everyone, 

Getting this when trying to adjust the serial to 232 using the iot2050setup page. I build an image off the example config in this repo only adding kernel headers, not sure if device names have changed or something?

Any help would be amazing, thank you in advance!

Ash.

jan-kiszka, 2021-02-02:
Did you try that with current master (31af2feec7b9cc3e35db3642e3be2bed86f84f79)? I just changed a lot in this area, see 564aa4f2a1a095f8693515744e7b96c2e68bef16, but I was not able to test all details. Can you explain what you tested exactly, the steps?

jan-kiszka, 2021-02-02:
...and the board type (advanced/basic).

I was playing with iot2050setup again on the basic, and changing the config there has some impact. We have an issue with reading that setting back via `switchserialmode -d`, though.

AshWhitear, 2021-02-02:
> Did you try that with current master ([31af2fe](https://github.com/siemens/meta-iot2050/commit/31af2feec7b9cc3e35db3642e3be2bed86f84f79))? I just changed a lot in this area, see [564aa4f](https://github.com/siemens/meta-iot2050/commit/564aa4f2a1a095f8693515744e7b96c2e68bef16), but I was not able to test all details. Can you explain what you tested exactly, the steps?

Haven't tried the latest commits out, still working on master from about 10 or so days ago. I can see what you mean though, lots of changes in this area. The steps to reproduce: after logging into SSH, type `iot2050setup` and hit enter, in the menu `Peripherals` hit enter, `Configure External COM Ports` hit enter then finally select `RS232` hit enter. For me this goes blank for a while, then an error pops up below the "you should power cycle" modal.

Currently we're using an advanced box, but will be using a mixture of both simple and advanced boxes (hopefully) with the same image.

AshWhitear, 2021-02-03:
Today I rebuilt a new image based off of the latest master and not getting the ERROR any more which is great news, probably resolves this particular ticket although I can't seem to read anything off of the DB9 serial port, using 232 (ttyUSB0 on the downloadable example image I believe). Do you know if these mappings have changed at all? 

Thanks, Ash.

AshWhitear, 2021-02-03:
the serial port thats on the front just to clarify - thanks!

AshWhitear, 2021-02-04:
hey @jan-kiszka - im now using a basic box with a newly build release image and having trouble with the serial port still. Do you know which /dev/tty* its available under and how I can switch to RS232 manually? We have a tight deadline on this, sorry for keep pinging.

AsuraZeng, 2021-02-04:
HI AshWhitear
basic box serial port is /dev/ttyS2
and you can use iot2050setup to switch to rs232 mode.

AshWhitear, 2021-02-04:
hey @AsuraZeng thank you for such a quick response, I have changed to RS232 and rebooted just in case watching `/dev/ttyS2` using command `cat /dev/ttyS2` im not seeing any data from a proven source

AsuraZeng, 2021-02-04:
Can you check that the Baudrate is consistent?
`stty -F /dev/ttyS2`

jan-kiszka, 2021-02-04:
Also note that all ports are in RS232 mode by default.

I will try to validate the port today again, if it works in all modes.

AshWhitear, 2021-02-04:
Thanks for getting back to me both, just grabbed some food.

@AsuraZeng the response from that command has the following consistent response:

```
speed 9600 baud; line = 0;
-brkint -imaxbel
```

@jan-kiszka thank you, our project is using the basic variety boxes with the current master release. Just that front DB9 port doesnt seem to be giving out any data.

If I can give any more details / run any more commands etc, please let me know and I have a dev box ready to go and can get back to you asap!

jan-kiszka, 2021-02-04:
Looks like I flipped some control lines in master: I'm getting rs232 by selecting 485. Should be fixed soon, but I need to double-check this.

AshWhitear, 2021-02-04:
Got you, no worries - any command I can run manually to test this / fix it for my local copy?

jan-kiszka, 2021-02-04:
Select RS485 in iot2050setup, no termination. That worked here.

jan-kiszka, 2021-02-04:
The flip was in switchserialmode itself: The original code claimed to switch mode0 while it was toggling the mode1 line. Make the line selection symbolic revealed that. PR to follow in some minutes.

AshWhitear, 2021-02-04:
Gave it a go, it's definitely doing more than it was but I'm just getting gibberish now and not consistently, just once:

```
cat /dev/ttyS2
▒LSjZ▒
```
What are your thoughts?

jan-kiszka, 2021-02-04:
Baudrate? I tested with 115200 set on both ends.

AshWhitear, 2021-02-04:
The device I'm using is set at 9600

jan-kiszka, 2021-02-04:
Check both ends. It works fine here, at least with my fixed version of switchserialmode.

AshWhitear, 2021-02-04:
Hmm stty seems to report 9600 too, is there a anyway I can apply your fix to my current image without recompiling the whole image?

jan-kiszka, 2021-02-04:
Yes, just apply the patch, start the re-build and pick up the binary after that package was in do_deploy_deb:

```
scp build/tmp/work/iot2050-debian-arm64/switchserialmode/0.1-r0/src/obj-aarch64-linux-gnu/switchserialmode root@192.168.200.1:/usr/bin/switchserialmode
```

AshWhitear, 2021-02-04:
just gave this a try but still no luck, can I run the command it ultimately runs manually?

AshWhitear, 2021-02-04:
Ive just tried: `switchserialmode ttyuart -D /dev/ttyS2 -m rs232` but just get a few gibberish characters and then nothing

jan-kiszka, 2021-02-04:
`switchserialmode -m rs232` is already sufficient here.

You may have some other issue then. Test the cable and peer side on a different device than the 2050 first if you didn't do that yet.

AshWhitear, 2021-02-04:
I have that a go too but still no luck, I have tried a different device but just nothing seems to come out. Plugging into the GPIO I can see data coming out, although not particularly friendly data. 

Really stuck here, not sure what else I can try.

AshWhitear, 2021-02-05:
Used prebuilt image from Siemens support site, made iot02050setup changes for RS232 for both advanced and basic boxes - these two modals differ upon selection, one asks if you'd like persist through a reboot the other asks for a reboot before it'll work. Got our electrical engineer to build us a new cable, works on advanced box not basic, tried another of our 20 newly arrived Siemens basic boxes with the same SD - no luck. 

Something doesn't seem right, the iot2050setup commands don't seem to be working - unless there is more I'm meant to do? I can see the advanced box has a battery and the basic ones do not, other than that - jumpers etc. are all identical which should rule out hardware.

Is there a set of lower level commands you guys use to test that I can use here? Getting rather desperate as these boxes are sold with DB9 serial with RS232, having seen it work on the advanced box we have planned on using that capability when we ordered the basic boxes seeing no differences in the documentation re: this area.

Thanks, Ash.

AshWhitear, 2021-02-06:
Removed my previous reply as its not helpful and I think I've got it..

Just for anyone else with a serial port that doesn't appear to be working but wants rs232 (note `/dev/ttyUSB0` for advanced, `/dev/ttyS2` for basic):

1. after applying @jan-kiszkas patch from above, go to `iot2050setup` and enable `RS232`, may be best to `reboot` here
2. check if you get any response from command `dd if={devicename}`
3. if not, run `systemctl enable serial-getty@{devicename}.service` (feel rather silly but it appears the port isn't enabled from warehouse nor by the iot2050setup command)

Hope this helps someone else, can't thank every one involved enough for their help.

Ash.

applea9, 2021-02-07:
is reboot necessary when apply change to serial port?

AshWhitear, 2021-02-07:
I think it may depend on the type of board you have, but from what I've seen the setup modal recommends advanced boards be rebooted after changing between modes but basic boards don't seem to suggest this - haven't got time to test but may be best to just follow those guides as the underlying commands each run (off the top of my head) are different.

jan-kiszka, 2021-02-07:
Yes, it's a hardware limitation that is supposed to be resolved with the next board revision.

AshWhitear, 2021-02-09:
Tell a lie, those instructions get the port working briefly but the port seems to keep disconnecting when pulling the stream in 😔

jan-kiszka, 2021-02-10:
OK, I don't think we clarified that yet: Did things work with an older version, specifically our current release?

commit 21646bd5 (2021-02-18): switchserialmode: Fix flipped mode0/1 control

The original code claimed to flip mode0 while it was toggling mode1, and
vice versa. While translating those comments into line names, that
became broken. Fix it, also by using better argument names for
gpio_set_mode.

Closes #97
closed 2021-02-18 by BaochengSu