siemens/meta-iot2050 #99: Ubuntu image generation - Node-red issues

Aeroten, 2021-02-11:
Hello Jan,

Tried today out of curiosity to generate the Ubuntu image. I get some issues with node-red. Tried to remove it from the recipe but it seems there are some dependencies I'm missing and it doesn't solve the problem.

Here is the full terminal output. Any idea ?

```
$ ./kas-container build kas-iot2050-example.yml:kas/opt/ubuntu.yml
2021-02-11 10:50:52 - INFO     - kas 2.3.3 started
2021-02-11 10:50:52 - INFO     - /repo$ git rev-parse --show-toplevel
2021-02-11 10:50:52 - INFO     - /repo/kas/opt$ git rev-parse --show-toplevel
2021-02-11 10:50:52 - INFO     - /repo$ git rev-parse --show-toplevel
2021-02-11 10:50:52 - INFO     - Using /repo as root for repository meta-iot2050
2021-02-11 10:50:52 - INFO     - /work/isar$ git remote set-url origin https://github.com/ilbers/isar
2021-02-11 10:50:52 - INFO     - /work/isar$ git cat-file -t f5a6fdfd9aad49202093aab2158f625429eaf7c5
2021-02-11 10:50:52 - INFO     - Repository isar already contains f5a6fdfd9aad49202093aab2158f625429eaf7c5 as commit
2021-02-11 10:50:52 - INFO     - /repo$ git rev-parse --show-toplevel
2021-02-11 10:50:52 - INFO     - Using /repo as root for repository meta-iot2050
2021-02-11 10:50:52 - INFO     - /work/isar$ git status -s
2021-02-11 10:50:52 - INFO     - /work/isar$ git rev-parse --verify -q origin/f5a6fdfd9aad49202093aab2158f625429eaf7c5
2021-02-11 10:50:52 - INFO     - /work/isar$ git checkout -q f5a6fdfd9aad49202093aab2158f625429eaf7c5
2021-02-11 10:50:52 - INFO     - /repo$ git rev-parse --show-toplevel
2021-02-11 10:50:52 - INFO     - Using /repo as root for repository meta-iot2050
2021-02-11 10:50:52 - INFO     - /work/isar$ /tmp/tmp1467qoei/get_bb_env /work/build
2021-02-11 10:50:52 - INFO     - /repo$ git rev-parse --show-toplevel
2021-02-11 10:50:52 - INFO     - Using /repo as root for repository meta-iot2050
2021-02-11 10:50:52 - INFO     - /repo$ git rev-parse --show-toplevel
2021-02-11 10:50:52 - INFO     - Using /repo as root for repository meta-iot2050
2021-02-11 10:50:52 - INFO     - /work/build$ /work/isar/bitbake/bin/bitbake -k -c build iot2050-image-example
Loading cache: 100% |####################################################################################| Time: 0:00:00
Loaded 50 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies
Initialising tasks: 100% |###############################################################################| Time: 0:00:00
NOTE: No setscene tasks
NOTE: Executing Tasks
ERROR: mindconnect-node-red-contrib-mindconnect-3.7.0-r0 do_install_builddeps: Execution of '/work/build/tmp/work/iot2050-ubuntu-arm64/mindconnect-node-red-contrib-mindconnect/3.7.0-r0/temp/run.do_install_builddeps.75' failed with exit code 1:
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
hostname: Temporary failure in name resolution
semop(1): encountered an error: Function not implemented
Error in the build process: exit status 1
dpkg: error: cannot access archive 'mindconnect-node-red-contrib-mindconnect-build-deps_3.7.0_arm64.deb': No such file or directory
mk-build-deps: dpkg --unpack failed
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /work/build/tmp/work/iot2050-ubuntu-arm64/mindconnect-node-red-contrib-mindconnect/3.7.0-r0/temp/log.do_install_builddeps.75
Log data follows:
| DEBUG: Executing shell function do_install_builddeps
| /bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
| hostname: Temporary failure in name resolution
| semop(1): encountered an error: Function not implemented
| Error in the build process: exit status 1
| dpkg: error: cannot access archive 'mindconnect-node-red-contrib-mindconnect-build-deps_3.7.0_arm64.deb': No such file or directory
| mk-build-deps: dpkg --unpack failed
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/work/build/tmp/work/iot2050-ubuntu-arm64/mindconnect-node-red-contrib-mindconnect/3.7.0-r0/temp/run.do_install_builddeps.75' failed with exit code 1:
| /bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
| hostname: Temporary failure in name resolution
| semop(1): encountered an error: Function not implemented
| Error in the build process: exit status 1
| dpkg: error: cannot access archive 'mindconnect-node-red-contrib-mindconnect-build-deps_3.7.0_arm64.deb': No such file or directory
| mk-build-deps: dpkg --unpack failed
| WARNING: exit code 1 from a shell command.
|
ERROR: Task (/repo/recipes-app/mindconnect-node-red-contrib-mindconnect/mindconnect-node-red-contrib-mindconnect_3.7.0.bb:do_install_builddeps) failed with exit code '1'
ERROR: node-red-dashboard-2.22.1-r0 do_install_builddeps: Execution of '/work/build/tmp/work/iot2050-ubuntu-arm64/node-red-dashboard/2.22.1-r0/temp/run.do_install_builddeps.77' failed with exit code 1:
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
hostname: Temporary failure in name resolution
semop(1): encountered an error: Function not implemented
Error in the build process: exit status 1
dpkg: error: cannot access archive 'node-red-dashboard-build-deps_2.22.1_arm64.deb': No such file or directory
[... 186 lines cut ...]
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/work/build/tmp/work/iot2050-ubuntu-arm64/node-red-contrib-modbus/5.13.2-r0/temp/run.do_install_builddeps.74' failed with exit code 1:
| /bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
| hostname: Name or service not known
| semop(1): encountered an error: Function not implemented
| Error in the build process: exit status 1
| dpkg: error: cannot access archive 'node-red-contrib-modbus-build-deps_5.13.2_arm64.deb': No such file or directory
| mk-build-deps: dpkg --unpack failed
| WARNING: exit code 1 from a shell command.
|
ERROR: Task (/repo/recipes-app/node-red-contrib-modbus/node-red-contrib-modbus_5.13.2.bb:do_install_builddeps) failed with exit code '1'
NOTE: Tasks Summary: Attempted 263 tasks of which 255 didn't need to be rerun and 8 failed.

Summary: 8 tasks failed:
  /repo/recipes-app/mindconnect-node-red-contrib-mindconnect/mindconnect-node-red-contrib-mindconnect_3.7.0.bb:do_install_builddeps
  /repo/recipes-app/node-red-dashboard/node-red-dashboard_2.22.1.bb:do_install_builddeps
  /repo/recipes-app/node-red-node-serialport/node-red-node-serialport_0.10.2.bb:do_install_builddeps
  /repo/recipes-app/node-red-contrib-s7/node-red-contrib-s7_2.2.1.bb:do_install_builddeps
  /repo/recipes-app/node-red-node-sqlite/node-red-node-sqlite_0.4.3.bb:do_install_builddeps
  /repo/recipes-app/node-red/node-red_1.0.6.bb:do_install_builddeps
  /repo/recipes-app/node-red-contrib-opcua/node-red-contrib-opcua_0.2.62.bb:do_install_builddeps
  /repo/recipes-app/node-red-contrib-modbus/node-red-contrib-modbus_5.13.2.bb:do_install_builddeps
Summary: There were 8 ERROR messages shown, returning a non-zero exit code.
2021-02-11 10:52:19 - ERROR    - Command returned non-zero exit status 1
```

Aeroten, 2021-02-11:
Got also some errors like this in the first build try:

```
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/IBM865.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/IBM1167.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/IBM4899.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/IBM1371.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/IBM1137.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/libISOIR165.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/IBM4971.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/GBK.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/IBM921.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/IBM437.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/IBM256.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/ISO-2022-JP.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/CP1258.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/IBM500.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/CP771.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/ISO_6937-2.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/ISO-2022-CN.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/IBM423.so': Permission denied
rm: cannot remove 'meta-iot2050/build/tmp/work/iot2050-ubuntu-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/gconv/UHC.so': Permission denied
```

jan-kiszka, 2021-02-11:
`semop(1): encountered an error: Function not implemented` - this indicated that the binfmt-misc configuration that kas-container sets up, also for your host, is not picking the qemu-user-static version of the container but sticks with an incompatible one of the host.

What is your host system? Does it have binfmt-misc installed and configured by itself?

And to clean up artifacts of a build, use `./kas-container --isar clean`.

Aeroten, 2021-02-11:
Hi Jan

There might be many problems at the same time.

qemu-user-static was not installed, indeed. Now I've installed the package, should solve some errors.

I started again with a clean version.
During the work I still have the such errors (only with node-red packages):

Host system is Debian running in WSL 2 on windows and Docker Desktop. I could already generate images for IOT2050 using WSL2 some month ago.

```
ERROR: Task (/repo/recipes-app/mindconnect-node-red-contrib-mindconnect/mindconnect-node-red-contrib-mindconnect_3.7.0.bb:do_install_builddeps) failed with exit code '1'
ERROR: node-red-contrib-s7-2.2.1-r0 do_install_builddeps: Execution of '/work/build/tmp/work/iot2050-ubuntu-arm64/node-red-contrib-s7/2.2.1-r0/temp/run.do_install_builddeps.29794' failed with exit code 1:
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
hostname: Temporary failure in name resolution
semop(1): encountered an error: Function not implemented
Error in the build process: exit status 1
dpkg: error: cannot access archive 'node-red-contrib-s7-build-deps_2.2.1_arm64.deb': No such file or directory
mk-build-deps: dpkg --unpack failed
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /work/build/tmp/work/iot2050-ubuntu-arm64/node-red-contrib-s7/2.2.1-r0/temp/log.do_install_builddeps.29794
Log data follows:
| DEBUG: Executing shell function do_install_builddeps
| /bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
| hostname: Temporary failure in name resolution
| semop(1): encountered an error: Function not implemented
| Error in the build process: exit status 1
| dpkg: error: cannot access archive 'node-red-contrib-s7-build-deps_2.2.1_arm64.deb': No such file or directory
| mk-build-deps: dpkg --unpack failed
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/work/build/tmp/work/iot2050-ubuntu-arm64/node-red-contrib-s7/2.2.1-r0/temp/run.do_install_builddeps.29794' failed with exit code 1:
| /bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
| hostname: Temporary failure in name resolution
| semop(1): encountered an error: Function not implemented
| Error in the build process: exit status 1
| dpkg: error: cannot access archive 'node-red-contrib-s7-build-deps_2.2.1_arm64.deb': No such file or directory
| mk-build-deps: dpkg --unpack failed
| WARNING: exit code 1 from a shell command.
```

Aeroten, 2021-02-11:
I make some extra checks and I come back.
Did you every try to build through WSL ?

Aeroten, 2021-02-11:
Also one thing I wonder... Why only by Node-Red packages and not the rest ?

jan-kiszka, 2021-02-11:
qemu-user-static ist not needed on the host. It comes with the kas-isar image which registers the right version with the host kernel via binfmt-misc. The problem is "the right version", see also https://github.com/siemens/kas/commit/c7e15bdbd65b91aed0ea2b80ba26aa6e0c7f506f. That was about Debian 11. With Ubuntu as targets, different versions will come, and those may trigger unimplemented parts of qemu-user-static earlier / at different spots.

Now the particular problem with binfmt-misc is that you cannot configure it per container (see also https://lkml.org/lkml/2021/1/18/1267). We currently use a hack that sets it up from the container on the host. In addition, we have seen caching effects: An earlier configuration and usage of binfmt-misc may cause later usages to not pick up the desired qemu-* binary, rather an older one. I suspect we see this scenario here.

I'm not using Windows, so I can't help with WSL in particular. But I can retry building the Ubuntu variant to exclude the issue was not imported by some upstream change, e.g. a newer Ubuntu package version.

Aeroten, 2021-02-11:
Ok I get it more or less.
Anything I can do on my side ?

jan-kiszka, 2021-02-11:
I'v  re-build locally, but it showed no regressions. I've also rebased at that chance, you will find the new version in the same branch, public CI currently running (https://github.com/siemens/meta-iot2050/runs/1881860607?check_suite_focus=true).

Did you already try restarting the WSL VM (not sure how that works, when in doubt via rebooting the host)?

Aeroten, 2021-02-11:
I'm going to give a try.
Just to check this afternoon I installed Ubuntu 20.04 LTS, same story, same errors

```
ERROR: Task (/repo/recipes-app/node-red-contrib-modbus/node-red-contrib-modbus_5.13.2.bb:do_install_builddeps) failed with exit code '1'
ERROR: node-red-node-sqlite-0.4.3-r0 do_install_builddeps: Execution of '/work/build/tmp/work/iot2050-ubuntu-arm64/node-red-node-sqlite/0.4.3-r0/temp/run.do_install_builddeps.15838' failed with exit code 1:
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
hostname: Name or service not known
semop(1): encountered an error: Function not implemented
Error in the build process: exit status 1
dpkg: error: cannot access archive 'node-red-node-sqlite-build-deps_0.4.3_arm64.deb': No such file or directory
mk-build-deps: dpkg --unpack failed
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /work/build/tmp/work/iot2050-ubuntu-arm64/node-red-node-sqlite/0.4.3-r0/temp/log.do_install_builddeps.15838
Log data follows:
| DEBUG: Executing shell function do_install_builddeps
| /bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
| hostname: Name or service not known
| semop(1): encountered an error: Function not implemented
| Error in the build process: exit status 1
| dpkg: error: cannot access archive 'node-red-node-sqlite-build-deps_0.4.3_arm64.deb': No such file or directory
| mk-build-deps: dpkg --unpack failed
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/work/build/tmp/work/iot2050-ubuntu-arm64/node-red-node-sqlite/0.4.3-r0/temp/run.do_install_builddeps.15838' failed with exit code 1:
| /bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
| hostname: Name or service not known
| semop(1): encountered an error: Function not implemented
| Error in the build process: exit status 1
| dpkg: error: cannot access archive 'node-red-node-sqlite-build-deps_0.4.3_arm64.deb': No such file or directory
| mk-build-deps: dpkg --unpack failed
| WARNING: exit code 1 from a shell command.
|
ERROR: Task (/repo/recipes-app/node-red-node-sqlite/node-red-node-sqlite_0.4.3.bb:do_install_builddeps) failed with exit code '1'
NOTE: Tasks Summary: Attempted 263 tasks of which 0 didn't need to be rerun and 8 failed.

Summary: 8 tasks failed:
  /repo/recipes-app/node-red-contrib-s7/node-red-contrib-s7_2.2.1.bb:do_install_builddeps
  /repo/recipes-app/node-red/node-red_1.0.6.bb:do_install_builddeps
  /repo/recipes-app/mindconnect-node-red-contrib-mindconnect/mindconnect-node-red-contrib-mindconnect_3.7.0.bb:do_install_builddeps
  /repo/recipes-app/node-red-dashboard/node-red-dashboard_2.22.1.bb:do_install_builddeps
  /repo/recipes-app/node-red-node-serialport/node-red-node-serialport_0.10.2.bb:do_install_builddeps
  /repo/recipes-app/node-red-contrib-opcua/node-red-contrib-opcua_0.2.62.bb:do_install_builddeps
  /repo/recipes-app/node-red-contrib-modbus/node-red-contrib-modbus_5.13.2.bb:do_install_builddeps
  /repo/recipes-app/node-red-node-sqlite/node-red-node-sqlite_0.4.3.bb:do_install_builddeps
Summary: There were 8 ERROR messages shown, returning a non-zero exit code.
2021-02-11 18:28:33 - ERROR    - Command returned non-zero exit status 1
```

Aeroten, 2021-02-11:
By the way, here is how to restart the WSL machines on Windows:
in Powershell

`wsl.exe --shutdown `

jan-kiszka, 2021-02-11:
Github CI still runs on 18.04.5 (in a VM, obviously), and that works. But 20.04 must be fine as well.

Things rather point to WSL, somehow, though also my understanding is that it's "just" a Hyper-V guest, no "special" kernel emulation anymore.

Aeroten, 2021-02-12:
OK I'll go on investigate and keep you in touch. Thank you.
I'm still surprised that it only makes issues with node-red and not the other package.

jan-kiszka, 2021-02-12:
It stumbles over the execution of certain tasks for the target arm64 architecture. The node packages are built this way while others are cross-compiled over x86, not requiring qemu-user. But I bet that final installation of the target image (which also happens over qemu-user) would fail as well.

jan-kiszka, 2021-03-02:
Could you resolve the issue?

Aeroten, 2021-03-18:
Hi Jan,

Yes i could. I used an bootable USB stick with Debian and ran the script.
I compiled.

I don't know if it works, but at least I got an image. :-)

Thanks for everything !

closed 2021-03-18 by Aeroten