siemens/meta-iot2050 #108: Ethernet is not working if DHCP configured port is unplugged

mwanders98, 2021-03-05:
Hello, 

we are using the IOT2050 Advanced and encounterd the same problem of issue #54. I want to bring up the problem again, cause this issue persists for us even using the mentioned workaround of using a newer image (newest Image built by Github Actions).

We are configuring the interfaces on boot time by a script, which is running as a systemd service unit:
```
[Unit]
Description=Network Configurator
After=network.target

[Service]
Type=oneshot
ExecStart=/bin/bash /opt/configurator.sh
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target
```

The script reads the config from an USB stick, writes the config to `/etc/network/interfaces` and restarts the network with `systemctl restart networking`.

Our config is like so:
* eth1 / X1P1: dhcp
* eth0 / X1P2: static

The problem is, that the `systemctl restart networking` command does not work, when the cable of the DHCP port is unplugged. The result is a device which cannot be communicated to, the LEDs on the NICs are blinking but nothing happens. 

The network restart fails with the error:
```
systemd[1]: Starting Raise network interfaces
ifup[289]: RTNETLINK answers: File exists
ifup[289]: ifup: failed to bring up eth0
systemd[1]: Failed to start Raise network interfaces
```

We then tried to force up the interfaces, but this also didn't help reliable:
```
systemctl restart networking
ifup --force --ignore-error -v eth0
ifup --force --ignore-error -v eth1
systemctl restart networking
```

To tackle down the problem further, we found the following in `/var/log/syslog`:
`Feb 25 10:33:07 iot2050 kernel: [  185.914984] icssg-prueth pruss0_eth eth0: xmit timeout`

Also we observed, that the dhclient seems to DISCOVER on the unplugged port (eth1), which obviously takes an eternity and does not work cause its unplugged.

Further we found the following resources, which could help to resolve the issue (but I don't know too much about it)
* https://e2e.ti.com/support/processors/f/791/t/938412?AM6548-6xPRU-ICSSG-Ethernet-driver-issue-on-custom-board
* https://e2e.ti.com/cfs-file/__key/communityserver-discussions-components-files/791/prueth.c
In `prueth.c`we found an open todo `can we recover or need to reboot firmware? ` on line 4092 (thats where the xmit timeout error comes from. Additionally we found something in u-boot, but I don't have any references anymore. 

What does help is to assign the unplugged port a static nonsense ip. Then the device boots up and everything works as expected. But this cannot be a solution in an industrial production environment.

I hope we can resolve this issue or you guys can give us advise how to fix it.

jan-kiszka, 2021-03-08:
@BaochengSu, @AsuraZeng, have you seen this before?

@mwanders98, there is no difference in using the original release (with an older 4.19 kernel) and the current one in master?
Next I would have asked you to try the upstream queue - but this unfortunately requires an update of the firmware which is not recommended without a programmer at hand to restore in case of problems.

BaochengSu, 2021-03-08:
No, haven't seen such issue before.

We will try to reproduce it.

@mwanders98, could you please share us the /opt/configurator.sh or its equivalent so that we can reproduce it from our side?

mwanders98, 2021-03-09:
I put the `/opt/configurator.sh` and the `/etc/systemd/system/configurator.service` into a [gist](https://gist.github.com/mwanders98/56b24396f0b33f2835190c60ed0ed361).

I just tested this scripts with a fresh copy of IOT2050_Example_Image_V1.0.2.img. 
I put both files on the system and did a `systemctl daemon reload` and `systemctl enable configurator.service`.

So then I have different configurations, whether the USB stick is inserted or not and whether the DHCP port is connected or not. 
After each configuration change I reboot the device and test if the static configured ethernet port is "pingable" with the configured ip and I test dhcp too, if its connected.

The resulting behaviour is pretty confusing in my opinion. Sometimes it takes an eternity till the ports are accessible. Sometimes nothing works at all. 

If you want to test it yourself, I would suggest you to open two terminals with a ping to the wan and lan ip's and to play a bit with the different configurations and to reboot each time.

Tomorrow, I will try this setup again with the newest image built by Github Actions from the main branch.

jan-kiszka, 2021-03-09:
Do your issues always correlate with that "xmit timeout" errors on the kernel console?

mwanders98, 2021-03-09:
Yes. I can see them in the syslog always.

BaochengSu, 2021-03-10:
@mwanders98 I saw from the scripts that the networking.service is managing the network, has the NetworkManager.service been disabled? Otherwise these two would conflict with each other.

The default network config tool for headless Debian buster is ifupdown, i.e. the networking.service belongs to. For our IOT2050 example image, to simplify the wireless network configuration, we use the Network Manager as the default configuration tool.

So if you are not using any wireless network technologies, or even don't mind the complexities of the wireless configuration, feel free to just disable the NetworkManager service.

mwanders98, 2021-03-23:
Sorry for my late response. I deactivated the NetworkManager but the error still exists. 
I tested this on the newest Debian image from [Github Actions](https://github.com/siemens/meta-iot2050/actions/runs/622988791).

Do you have an alternative idea how I can configure the network configuration dynamically at boot time? For example I could run the script before any network related systemd services are starting up. My question here is which systemd target would be correct? The network-pre.target does not exist, if I remember correctly.

jan-kiszka, 2021-03-23:
If you install ifupdown, you can configure networking the classic way. And even if NetworkManager is enabled, it will stay away from interfaces described that way.

But are you sure you could address/work around this issue at all via scripts? The timeout error of the kernel rather points to the driver or even lower, not the way how and by whom networking is configured.

mwanders98, 2021-03-23:
The configuration itself works fine via /etc/network/interfaces.

No, I am not sure at all, just a naive idea from my side for a temporary workaround. I will try the ubuntu branch tomorrow or do you have another idea how to fix this? 
You mentioned a firmware update, which firmware should I update to? And how to do it? I've seen an iot2050-firmware-update script..

jan-kiszka, 2021-03-23:
Ubuntu will mean same kernel, possibly just a different NetworkManager version, but same (default) way of configuration.

We have no new firmware image to share that you can safely install. You can build own images from source, and those may work, but if they don't, you need a programmer to reflash a working image.

mwanders98, 2021-03-24:
I downloaded the newest images from Github Actions artifacts, but it seems that they are shipping with kernel 4.19 not 5.x, is this correct?
So it looks like I didn't ever had a "new" image. 

Therefore I tried to build the images myself but I ran into following issues:

````

| I: Extracting fdisk...
........
| I: Extracting liblzma5...
| I: Extracting zlib1g...
| I: Running command: chroot /work/build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs /debootstrap/debootstrap --second-stage
| W: Failure trying to run:  mount -t proc proc /proc
| W: See //debootstrap/debootstrap.log for details
| W: Failure trying to run:  /sbin/ldconfig
| W: See //debootstrap/debootstrap.log for details
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/work/build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/temp/run.do_bootstrap.71' failed with exit code 1:
| I: Running command: debootstrap --arch arm64 --foreign --verbose --variant=minbase --include=locales,gnupg --components=main,contrib,non-free buster /work/build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs http://deb.debian.org/debian
| I: Retrieving InRelease
| I: Checking Release signature
| I: Valid Release signature (key id 6D33866EDD8FFA41C0143AEDDCC9EFBF77E11517)
| I: Retrieving Packages
......
| I: Validating Packages
| I: Resolving dependencies of required packages...
| I: Resolving dependencies of base packages...
| I: Checking component main on http://deb.debian.org/debian...
| I: Retrieving libacl1 2.2.53-4
| I: Validating libacl1 2.2.53-4
........
| I: Extracting liblzma5...
| I: Extracting zlib1g...
| I: Running command: chroot /work/build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs /debootstrap/debootstrap --second-stage
| W: Failure trying to run:  mount -t proc proc /proc
| W: See //debootstrap/debootstrap.log for details
| W: Failure trying to run:  /sbin/ldconfig
| W: See //debootstrap/debootstrap.log for details
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-target.bb:do_bootstrap) failed with exit code '1'
NOTE: Tasks Summary: Attempted 154 tasks of which 80 didn't need to be rerun and 1 failed.

````

I tried the different build commands mentioned in the README, but neither works. 
My build host is a Ubuntu vmware vm which is completely empty but was upgraded from 18.04 to 20.04.2 today.

`Linux build 5.4.0-70-generic #78-Ubuntu SMP Fri Mar 19 13:29:52 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

What could fix this build error?

jan-kiszka, 2021-04-29:
When sharing build issues, please make sure to also share the exact command line used to start the build.

Regarding 5.x kernels (upstream): You can build that already by appending `:kas/opt/upstream.yml` to the main yaml. However, it takes a newer firmware as well, and installing that my leave your device broken (see https://github.com/siemens/meta-iot2050/blob/master/recipes-bsp/u-boot/README.md).

serafimdasaudade, 2021-05-06:
Hi, I'm also have the same issue with IOT2050 Basic. 
If I connect the ethernet cable after the equipment boots. The ethernet port (configure as dhcp) don't get IP address.

jan-kiszka, 2021-05-06:
But that works if the cable was plugged before booting?

Did you try with the latest image from CI here? Or with the released one? Any differences in behavior?

serafimdasaudade, 2021-05-06:
It works well if the cable is plugged before booting.
The image was build from the github commit from the date 2021-03-05.
I did not try with other image.

jan-kiszka, 2021-09-13:
Problem still applies to V01.01.01 release?

serafimdasaudade, 2022-01-26:
Sorry for the late reply.
I download the V01.01.01 release from the siemens forum and I got the same issue.

Vishal8523, 2022-08-08:
I also have the same issue even with the recent release. Is there any resolution to this issue ?

BaochengSu, 2022-08-09:
@Vishal8523 Do you have the `dmesg | grep pru` log?

Vishal8523, 2022-08-23:
[image]

jan-kiszka, 2023-09-12:
I've integrated the newer TI firmware that may fix this but not yet tested the outcome. Any feedback highly welcome!

jan-kiszka, 2023-10-23:
According to our tests, the issue is resolved with the latest master version. Please reopen if that should not be the case.

closed 2023-10-23 by jan-kiszka