siemens/meta-iot2050 #109: Sierra Modem Not Compatible

adriantaw, 2021-03-12:
Hi, we built an image based on the latest branch available hoping that the latest modem driver is installed.

Unfortunately after installing the latest image the driver isn't recognized.
We have this Sierra Modem Sierra "MC7455 LTE Module" (https://support.industry.siemens.com/tf//ww/en/posts/245626).

I also recognized that the P1 static port (192.168.200.1) is deactivated by default after the image is installed. We could only connect by using the DHCP on the P2. I have the advanced version of the IoT.

jan-kiszka, 2021-03-12:
What was the image version you tried? Current master branch? Because f7e41b5878e1ccb21144aede0336e7354e67ef13 is included there.

For that version, I can also not confirm that the static port wouldn't be on by default.

adriantaw, 2021-03-12:
Yes I am using the latest one. I did double check with fe_lipe and the driver is indeed included, but somehow it still doesn't recognized my modem. Any clue?

jan-kiszka, 2021-03-12:
Could you provide `lspci -nnvv` and `lsusb` when the modem is plugged?

adriantaw, 2021-03-16:
Hereby the requested info
[image]
[image]
[image]

jan-kiszka, 2021-03-16:
OK, it's not being detected, neither by the USB host controllers nor the PCIe bridge.

@BaochengSu, would we support USB-attached modems in the Mini PCIe slot at all? If not, that pattern would make sense if the Sierra Modem is USB-only.

@adriantaw, can you verify in some other device with Mini PCIe slot whether the card is present there? Maybe some older notebook with such an interface.

jan-kiszka, 2021-03-18:
@adriantaw, please make sure that the device came out of a cold boot (power cycle e.g.) when listing the PCI device. We might have an issue with warm reboots and PCIe reset.

adriantaw, 2021-03-23:
@BaochengSu @jan-kiszka I have tried to power cycle the device and done the command again, but still getting the same message. Any clue when we can fix this compatibility? We have customer ordering multiple number of IoT and these modem based on our recommendation in the forum/

BaochengSu, 2021-03-23:
@jan-kiszka We do support USB modems but this MC7455 is not tested by us. The sierra card we've tested is MC7304

@adriantaw MC7455 is using usb3.0 and there is no USB3.0 signals in the miniPCIE slots of IOT2050. And from [Sierra's datasheet](https://www.tme.eu/Document/93515952d31c35657372371d1abc8897/4116238%20AirPrime%20MC7455.pdf), page 22:

```
Note: On any given interface (USB, SIM, etc.), leave unused inputs and outputs as noconnects.
```
Maybe a quick try could be perform on the card by cover pins 23,25,31,33 with taps. Also maybe sierra's tech support could be another resource to seek help.

Meanwhile we are trying to acquire this card to reproduce the bug.

BaochengSu, 2021-03-23:
@adriantaw Do you have this card on hand now? If so could you ship a sample to us so that we can accelerate the debugging? My email address: baocheng.su@siemens.com

BaochengSu, 2021-03-24:
Talked with @adriantaw, actually it's a MC7403 card, however, that makes no big difference with MC7455, so this issue should also exist on MC7455 card.

After covering the pins 23, 25, 31, 33 with taps, plug it into the miniPCIE slot of IOT2050, the card could be detected. Then use iot2050setuptool to activate the celullar 4g, at least the Internet is accessable with my China Telecom sim card.

As for the root cause, [this page](https://techship.com/faq/minipcie-based-cellular-modules-supporting-usb3-are-not-detected-in-the-host-system/) explains very clearly. TL;DR, MC74xx use PCIE signals as USB3.0 signals, but IOT2050 doesn't support the USB3.0 signals via PCIE pins.

The above page also gives a solution to address this problem, by forcing the card to use usb 2.0 model with AT-commands. However, currently the sierrawireless.com is under maintanance hence I am not able to get the AT-commands reference from this site, I am not able to verify that solution atm.

BaochengSu, 2021-04-09:
Looks recently the sierrawireless.com was under attack.

Anyway, today I made some work and finally git it work on IOT2050 without taping the USB3.0 signals.

Here is the detail:

1. Either using  a USB-miniPCIE adaptor or taping the USB3.0 signals, connect the modem to PC or IOT2050. I was connecting it to a Ubuntu 20 PC, however it should also work on latest IOT2050 example image.
2. Download the latest firmware of the modem from [here](https://source.sierrawireless.com/resources/airprime/minicard/74xx/airprime-em_mc74xx-approved-fw-packages/#sthash.6S3Ew43G.dpbs), make sure to download the correct binary for your modem. I am using a `Sierra MC7430` so the one I chose is `Generic | SWI9X30C_02.36.00.00`, if you are using the same one as mine, download [here](https://source.sierrawireless.com/-/media/support_downloads/airprime/74xx/fw/7430/swi9x30c_02,-d-,36,-d-,00,-d-,00_generic_002,-d-,077_000.ashx).
3. Unzip the downloaded file, you will get 2 files, a `.cwe` file and a `.nvu`.
4. Install tools:
```bash
sudo add-apt-repository universe
sudo apt update
sudo apt install libqmi-glib5 libqmi-proxy libqmi-utils -y
```
5. Stop the ModemManager service
```bash
sudo systemctl stop ModemManager
```
6. Update the firmware, replace the file names if you are not using the same modem as mine.
```bash
deviceid=`lsusb | grep -i -E '1199:9071|1199:9079|413C:81B6' | awk '{print $6}'`
sudo qmi-firmware-update --update -d "$deviceid" SWI9X30C_02.36.00.00.cwe SWI9X30C_02.36.00.00_GENERIC_002.077_000.nvu
```
7. Then the modem will reset.
8. Use `lsusb -t` command to check if the modem is online. You would see 3 `qcserial` devices.
9. Use serail tool to connect the the modem. The modem has 3 ttyUSBx, we need connect to the last one.
```bash
picocom -b 9600 /dev/ttyUSBx
```
10. Issue below AT-commands:
```
AT!ENTERCND="A710"
AT!USBSPEED=0
AT!RESET
``` 

The modem will reset, and after the resetting, you can use it on the IOT2050 directly w/o taping the USB3.0 signals.

Ref:
- https://ltehacks.com/viewtopic.php?t=429
- https://github.com/danielewood/sierra-wireless-modems