siemens/meta-iot2050 #113: cdc_mbim module missing

Webbeh, 2021-04-06:
Trying to use a Sierra-branded Dell DW5811E lte modem, I'm encountering the issue that the cdc_mbim kernel module is missing, and ModemManager hence can't see the device.

Any easy fix for this issue without recompiling the kernel with the module ?

Thank you !

Webbeh, 2021-04-06:
Further details : 
uname -a : 
`Linux iot2050-debian 4.19.94-rt39 #1 SMP PREEMPT RT Tue Mar 23 07:56:25 UTC 2021 aarch64 GNU/Linux`

lsusb : 
`Bus 001 Device 004: ID 413c:81c2 Dell Computer Corp.`

usb-devices : 
```T:  Bus=01 Lev=02 Prnt=02 Port=01 Cnt=01 Dev#=  4 Spd=480 MxCh= 0
D:  Ver= 2.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=413c ProdID=81c2 Rev=00.06
S:  Manufacturer=Sierra Wireless, Incorporated
S:  Product=DW5811e Snapdragon™ X7 LTE
S:  SerialNumber=LF92059268021029
C:  #Ifs= 5 Cfg#= 1 Atr=a0 MxPwr=500mA
I:  If#=0x0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
I:  If#=0xc Alt= 0 #EPs= 1 Cls=02(commc) Sub=0e Prot=00 Driver=(none)
I:  If#=0xd Alt= 0 #EPs= 0 Cls=0a(data ) Sub=00 Prot=02 Driver=(none)
I:  If#=0x2 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)
I:  If#=0x3 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)
```
(Note the absence of drivers)

ModemManager's output : 
```Apr 06 07:27:21 iot2050-debian ModemManager[2599]: <info>  Couldn't check support for device '/sys/devices/platform/interconnect@100000/4000000.dwc3/4010000.usb/xhci-hcd.0.auto/usb1/1-1/1-1.4': not supported by any plugin
Apr 06 07:27:21 iot2050-debian ModemManager[2599]: <info>  Couldn't check support for device '/sys/devices/platform/pruss0_eth': not supported by any plugin
```

jan-kiszka, 2021-04-06:
If the build passes, you should find a testing image under artifacts at https://github.com/siemens/meta-iot2050/actions/runs/721725051. Rebuilding, at least of the kernel, is unfortunately the only solution of some kernel driver is missing.

Webbeh, 2021-04-06:
Will try to install it as soon as it's built, and will let you know soon after.

Thank you !

Webbeh, 2021-04-06:
[image]

[image]

The SIM is present and no PIN is enabled. Do you think it could be due to the mPCIe - M.2 adapter not allowing the device to communicate with the SIM tray?

jan-kiszka, 2021-04-06:
This should work, at least it was tested to work fine with other modems. Already tried to search the web for this error pattern?

Webbeh, 2021-04-06:
Nothing much is showing up with google searches, sadly. Also, nothing seems to happen (either in dmesg or in syslog) whenever I try hot swapping the SIM.
Tried with two different, known working, 4G-enabled SIM cards, without and with reboot, and nothing much seems to happen. I wonder if I may have missed a step.

Perhaps another missing driver ?

[image]

jan-kiszka, 2021-04-06:
@BaochengSu, any ideas? Could that adapter be the issue?

Webbeh, 2021-04-06:
Alright, so after trying with another SIM, I seem to have a link (very low but that's due to the missing 2nd antenna).

So either my adapter is of very poor quality, or the sim tray is damaged somehow. In any case, it seems to not be an IOT or driver issue anymore, and we can add the device to the compabitle list !

[image]

jan-kiszka, 2021-04-06:
Ah, good. Do you have an antenna attached, and is there the option on that adapter to attach one? That may explain poor quality.

PR for the config switch is open https://github.com/siemens/meta-iot2050/pull/114.

Webbeh, 2021-04-06:
I have one antenna only as I only found one MHF4 - RPSMA adapter cable, hence the low quality, but yes I do have two connectors.

...However, 4G connectivity was lost again after a reboot. Sim missing again... Guess I'll try another IOT2050 as soon as I receive one.

commit 45ecd8f9 (2021-04-09): linux-iot2050: Enable further USB-based modems

Closes #113
commit 3bdaf222 (2021-04-15): linux-iot2050: Enable further USB-based modems

Closes #113
closed 2021-04-08 by Webbeh