siemens/meta-iot2050 #117: IOT2050 random blocks

serafimdasaudade, 2021-04-17:
Hi, 
I'm doing some tests with iot2050 with the objective of replace the io2040 for new  projects.
I'm running the software develop for io2040 (this software is running in some installations about 2 years we no problems) but on the iot2050, I'm facing some random locks.
The iot2050 stop blik the STAT led, It don't respond to ping or accept ssh connection. 
The SO logs don't gave the information about the lock.

Can someone give me some tip to record the system log's. Maybe using the uart debug port?

Best Regards,

jan-kiszka, 2021-04-17:
Yes, a good starting point would be attaching a UART-USB adapter (same as used for Raspberry Pi) to the debug port and record the kernel output with a second machine. Maybe you are "only" losing the network link. Then you can also log into the IOT2050 via debug port and examine the reason.

Is your box operated under normal environmental conditions  (e.g. in a lab, free air circulation)? Just to rule out temperature related issues.

serafimdasaudade, 2021-04-17:
the equipment is operating on normal environmental (23Cº),
what log file you suggest?  /var/log/messages ?

Best Regards,

jan-kiszka, 2021-04-19:
Yes, /var/log/message may also indicate if you see a network loss or a real crash/lock-up. In the latter case, the log may stop unexpectedly.

serafimdasaudade, 2021-06-04:
I got some update. 
The power supply been used in the test was not reliable.  
After change the powersupply the equipment is running for 28days. 

I will close the issue.

Best Regards,

closed 2021-06-04 by serafimdasaudade