siemens/meta-iot2050 #120: Atheros AR9271 WiFi USB dongle doesn't work.

Davide3i, 2021-04-24:
Hello.
I'm trying to use a WiFi USB dongle (ALFA AWUS036NHA), which uses chipset Atheros AR9271. I've tested and used it with many other Linux distros without any kind of issue.
I've installed _[firmware-atheros](https://packages.debian.org/buster/firmware-atheros)_, _lsusb_ correctly detects the WiFi USB dongle, but I cannot see it with _ifconfig_.

[image]
[image]

jan-kiszka, 2021-04-25:
This requires CONFIG_ATH9K_HTC which is actually set (to 'm') since day 1. What does the kernel console report (dmesg) when you plug in the stick?

Davide3i, 2021-04-25:
I can read this. Looks like it gets disconnected as soon as I connect it.

```
[  238.260414] usb 3-1: USB disconnect, device number 3
[  255.947201] usb 3-1: new high-speed USB device number 4 using xhci-hcd
[  256.111663] usb 3-1: New USB device found, idVendor=0cf3, idProduct=9271, bcdDevice= 1.08
[  256.111678] usb 3-1: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[  256.111683] usb 3-1: Product: UB91C
[  256.111688] usb 3-1: Manufacturer: ATHEROS
[  256.111693] usb 3-1: SerialNumber: 12345
```

jan-kiszka, 2021-04-25:
...but then it is correctly recognized by the USB core. No further printing after that? Does lsmod list ath9k_htc & Co.?

Davide3i, 2021-04-25:
No other messages after that.

Moreover...

`
:~$ lsmod | grep ath
{no output}`

Thank you for your help.

jan-kiszka, 2021-04-25:
Does `modprobe ath9k_htc` work?
What image version are you using?

Davide3i, 2021-04-25:
`sudo modprobe ath9k_htc
modprobe: FATAL: Module ath9k_htc not found in directory /lib/modules/4.19.94-rt39
`

I suppose this is the image version, right?
`v2.1(release):v2.1-421-g996d37930-dirty
`

jan-kiszka, 2021-04-25:
Ok, Industrial OS. Then you are actually on the wrong issue tracker because the kernel config of that image is not managed here. Please file a request via the support forum.

closed 2021-04-25 by jan-kiszka

Davide3i, 2021-04-25:
Oh, I'm sorry. Thank you.