siemens/meta-iot2050 #123: Default available Node-RED modules get uninstalled

kumajaya, 2021-04-26:
After installing node-red-contrib-bit from Node-RED user directory /root/.node-red using "npm install node-red-contrib-bit", all default available modules:

- @mindconnect/node-red-contrib-mindconnect
- node-red-contrib-modbus
- node-red-contrib-opcua
- node-red-contrib-s7
- node-red-dashboard
- node-red-node-serialport
- node-red-node-sqlite

disappear or maybe get uninstalled.

I managed to reinstall all modules without any difficulty execpt node-red-node-sqlite with "python: not found" error.

kumajaya, 2021-04-27:
I found it's a "normal" behavior because even "npm install" without any arguments would remove preinstalled module. I think this is related to npm dependency hell which I don't fully understand.  At least reinstall uninstalled preinstalled modules worked for me.

jan-kiszka, 2021-04-28:
Managing nodes via the node red pallette is supposed to work, non-descructively (and was working a while back when I last tried). And that also only issues npm commands (see logs).

Please share exact steps to reproduce, including image version used - and check again if management via node red isn't working.

kumajaya, 2021-04-29:
Just received the fourth device today. Like the previous three, I will install Ubuntu image on it and try to add a node from Node-RED pallete.

kumajaya, 2021-05-28:
I lost all preinstalled modules after updating node-red-node-sqlite from Node-RED pallette with log:
```-----------------------------------------------------------
2021-05-28T10:30:42.317Z Install : node-red-node-sqlite 0.6.0

2021-05-28T10:30:50.580Z npm install --no-audit --no-update-notifier --no-fund --save --save-prefix=~ --production node-red-node-sqlite@0.6.0
2021-05-28T10:31:21.232Z [err] npm
2021-05-28T10:31:21.233Z [err]  WARN deprecated node-pre-gyp@0.11.0: Please upgrade to @mapbox/node-pre-gyp: the non-scoped node-pre-gyp package is deprecated and only the @mapbox scoped package will recieve updates in the future
2021-05-28T10:51:01.523Z [out] 
2021-05-28T10:51:01.523Z [out] > sqlite3@4.2.0 install /root/.node-red/node_modules/sqlite3
2021-05-28T10:51:01.523Z [out] > node-pre-gyp install --fallback-to-build
2021-05-28T10:51:01.523Z [out] 
2021-05-28T10:51:06.439Z [err] node-pre-gyp
2021-05-28T10:51:06.442Z [err]  
2021-05-28T10:51:06.442Z [err] WARN
2021-05-28T10:51:06.442Z [err]  
2021-05-28T10:51:06.442Z [err] Using request for node-pre-gyp https download
2021-05-28T10:51:06.442Z [err]  
2021-05-28T10:51:06.481Z [out] [sqlite3] Success: "/root/.node-red/node_modules/sqlite3/lib/binding/node-v64-linux-arm64/node_sqlite3.node" already installed
2021-05-28T10:51:06.482Z [out] Pass --update-binary to reinstall or --build-from-source to recompile
2021-05-28T10:51:06.871Z [err] npm
2021-05-28T10:51:06.872Z [err]  notice created a lockfile as package-lock.json. You should commit this file.
2021-05-28T10:51:06.881Z [err] npm
2021-05-28T10:51:06.881Z [err]  WARN optional
2021-05-28T10:51:06.881Z [err]  SKIPPING OPTIONAL DEPENDENCY: fsevents@~2.3.1 (node_modules/node-red-contrib-opcua/node_modules/chokidar/node_modules/fsevents):
2021-05-28T10:51:06.881Z [err] npm
2021-05-28T10:51:06.881Z [err]  WARN
2021-05-28T10:51:06.881Z [err]  notsup
2021-05-28T10:51:06.881Z [err]  SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm64"})
2021-05-28T10:51:06.882Z [err] 
2021-05-28T10:51:06.889Z [out] + node-red-node-sqlite@0.6.0
2021-05-28T10:51:06.889Z [out] added 4 packages from 2 contributors, removed 775 packages, updated 1 package and moved 64 packages in 1210.049s
2021-05-28T10:51:06.979Z rc=0```

jan-kiszka, 2021-05-28:
Which image revision did you use?

jan-kiszka, 2021-05-28:
Sigh, confirmed - I suppose this slipped through as I didn't restart node red after the update. Need to understand why this wipes the existing nodes. Possibly an issue that came in with the last node red update.

zhqingyong, 2021-06-09:
The bug still  alive  --！

commit 285ff80d (2021-06-10): node-red: Fix update via palette of preinstalled nodes

Just like in meta-iot2000, we need to create a package.json in the
.node-red folder that contains our preinstalled nodes. Otherwise,
Node-RED removes them on any update of individual nodes.

Closes #123
closed 2021-06-10 by BaochengSu