siemens/meta-iot2050 #124: Modem Manager could not reconnect after connection failure of bad signal

applea9, 2021-04-27:
Hi,

I met a problem with 4G connection in IOT2050, the 4G module used is quectel EC20. In my test, when the 4G signal becomes bad,  the 4G connection will fail. But when the signal quality becomes good, the 4G connection could not reestablish.

the syslog is as following:

[image]

Any ideas about this problem, thanks for your help!

AsuraZeng, 2021-04-27:
Hi
you can try to modify the file under `/etc/NetworkManager/system-connections/cellular-4g`
autoconnect from false to true

applea9, 2021-04-27:
> Hi
> you can try to modify the file under `/etc/NetworkManager/system-connections/cellular-4g`
> `autoconnect` from false to true

I have set autoconnect to true. I can see modem manager is continuously retrying.

AsuraZeng, 2021-04-27:
do you have connected antenna on the 4G module?

applea9, 2021-04-27:
> do you have connected antenna on the 4G module?

Yes I have. But in the test, in order to simulate the bad signal, I uninstalled the 4G antenna. And after the connection is failed, I install the antenna back to the 4G module.

AsuraZeng, 2021-04-29:
we are now looking into it, will respond shortly

applea9, 2021-04-29:
In the test only IOT2050 advanced has this issue.

AsuraZeng, 2021-04-30:
Could you pls contact me through email:  chao.zeng@siemens.com

AsuraZeng, 2021-05-21:
Hi
pls have a try on branch #136

applea9, 2021-06-02:
It works well now, thanks.

closed 2021-06-02 by applea9