siemens/meta-iot2050 #128: Read board unique serial number

applea9, 2021-04-29:
Hi,

In the back of IOT2050, we can find serial number like sV-M4B2V0834. Can we read this serial number in the OS, and use it as unique machian ID?

Thanks for your reply.

jan-kiszka, 2021-04-29:
Try if `fw_printenv board_serial` provides what you are looking for.

applea9, 2021-04-30:
Thanks, it works. By the way, how to read the serial number in IOT2040?

jan-kiszka, 2021-04-30:
IIRC, it is part of the DMI information (check dmidecode).

closed 2021-04-30 by applea9

bramroelandts, 2021-08-16:
@jan-kiszka Is there a way to get the unique serial number as non-root user? `fw_printenv board_serial` complains about permission denied on /dev/mtd3 and /dev/mtd4. Even if I `chown` these two to admin, after a while their permissions reset back to root.
We'd like to read the serial number from NodeJS, but running the whole script as root does not seem like such a good idea.

jan-kiszka, 2021-08-16:
The trick will likely be some udev rule that grants read-permissions on those mtd devices to a specific group or the world. There should be plenty of examples on that in the net (though possibly not for mtd itself).

bramroelandts, 2021-08-16:
Thanks for the quick reply and helpful answer. In case anyone is interested, add a udev rule as file `/etc/udev/rules.d/99-serial.rules`:

`SUBSYSTEM=="mtd", ATTRS{name}=="ospi.env", ACTION=="add", RUN+="/bin/chmod 444 /dev/$name"
 SUBSYSTEM=="mtd", ATTRS{name}=="ospi.env.backup", ACTION=="add", RUN+="/bin/chmod 444 /dev/$name"`