siemens/meta-iot2050 #134: Booting a bare metal solution

Koala28, 2021-05-18:
Hello,

is it possible to boot a bare metal solution instead of booting the Linux on the IOT2050 GP?

Where the entry point of this solution has to be located?

jan-kiszka, 2021-05-18:
You can easily boot everything that U-Boot is capable of booting. U-Boot is set up to use distro boot by default, but you can reconfigure its environment to load you own binaries according to their needs.

Note, though, that baremetal will mean you have first of all zero support for the SOC - unless you find a runtime or RTOS which is enabled for it. And then comes the specifics of the IOT2050, basically what we encoded in the device trees.

What is your use case?

Koala28, 2021-05-18:
We have all the stuff we need running well in a bare metal approach on the TMDX654IDKEVM (booting directly from the SD-Card). Only the internal memory of the AM65xx is used, not the DDR.
The next step is to get it run on the IOT2050. So if we rename the binary as requested in the IOT2050 namespace this should be enough? Are there any restrictions regarding locating the entry point or the used memory space beside the common restrictions of the AM65xx?

jan-kiszka, 2021-05-18:
How is the boot process on the EVM? Is your binary acting as first-stage bootloader as well, or is there also U-Boot, TF-A etc. in the loop?

The former case would not be officially supported. It would not be possible on the advanced due to secure boot. It would be theoretically possible on the basic but it might be an enormous effort to get things working, given potential differences between the EVM and our hardware.

Koala28, 2021-05-18:
No, the binary is not acting as a first stage bootloader. For this we are using the tiboot3.bin and sysfw.bin coming with the SDK. The binary I talk about is "only" the application including a bare metal runtime environment using a minimum of the hardware (Ethernet MACs, UART, digital IO). Secure Boot should be no issue because we're using the IOT2050 GP.

jan-kiszka, 2021-05-18:
tiboot3.bin should be loading tispl.bin for the next stage. On the IOT2050, this is located on the SPI flash, see also https://github.com/siemens/u-boot/blob/jan/iot2050/arch/arm/dts/k3-am65-iot2050-boot-image.dtsi. You could try to replace that, but that only makes sense if you have a flash programmer as well, so that you can recover the likely initially bricked device.

Maybe you can also load your binary from U-Boot. That would allow using an easier reprogrammable medium.Try digging into that first.

Koala28, 2021-05-20:
Is it possible to get the *.pinmux file of the IOT2050 GP?

jan-kiszka, 2021-05-20:
We do not have such a file. Pin-muxing information is in the device tree, in the latest version also more annotated /wrt pin-muxing inputs for the Arduino connector.