siemens/meta-iot2050 #137: Add RTL8188EU Kernel Driver

bergmanu, 2021-05-25:
I got some USB-WiFi sticks using the rtl8188eu kernel driver, this seems not to be activated. 
I've read about some more dongles using this driver.

jan-kiszka, 2021-05-25:
This is a staging driver, that is likely the reason. Staging sometimes contains funny stuff, both regarding stability as well as sometimes also due to unclean licensing. We would have to do a careful evaluation first.

jan-kiszka, 2021-05-25:
I've scanned the code and checked the history - seems it's uncontroversial progressing well towards being moved out of staging. Two questions, though:
- did anyone test it's stability over our 4.19 kernel already (via a custom image)?
- where do we get any potentially needed firmware from?

bergmanu, 2021-05-25:
I did a simple test by just adding _CONFIG_RTL8188EU=m_ to iot2050_defconfig_extra.cfg. This was not solving the problem.  Don't know if this step is enough?

There is also a [description ](https://askubuntu.com/questions/1027191/ubuntu-16-04-d-link-n150-pico-wi-fi-adapter-doesnt-work) how to install the firmware for this dongle, but therefore the kernel headers are needed.

jan-kiszka, 2021-05-25:
I suspect you are missing a firmware for the device. Check the kernel logs.

That description is rather old. That git repos was a work of a repo of the kernel developer who is working on the staging driver. So at least with 5.10, we should have good chances that things will work this way.

bergmanu, 2021-05-25:
May 25 13:41:39 iot2050-debian kernel: [ 4554.048938] usb 1-1.1: new high-speed USB device number 7 using xhci-hcd
May 25 13:41:39 iot2050-debian kernel: [ 4554.153215] usb 1-1.1: New USB device found, idVendor=2001, idProduct=331b, bcdDevice= 0.00
May 25 13:41:39 iot2050-debian kernel: [ 4554.161578] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
May 25 13:41:39 iot2050-debian kernel: [ 4554.168903] usb 1-1.1: Manufacturer: Realtek
May 25 13:41:39 iot2050-debian kernel: [ 4554.173276] usb 1-1.1: SerialNumber: A0A3F014A874

That's all from the kernel logs.

jan-kiszka, 2021-05-25:
Yeah, you are missing https://github.com/torvalds/linux/commit/5f74a8cbb38d10615ed46bc3e37d9a4c9af8045a - added only to 5.0.

bergmanu, 2021-05-25:
Can I integrate that somehow in my build? 
Or do I need to wait until we have kernel 5.x?

jan-kiszka, 2021-05-25:
I don't think we will easily be able to identify and backport needed patches for this device to the 4.19 kernel anymore.

So, for now you could
- build latest firmware and image with :kas/opt/upstream.yml
- install that new firmware (keep a programmer in reach in case this goes wrong)
- boot the new image

jan-kiszka, 2021-09-13:
Latest master (only works with rebuilt/reflashed firmware) should allow this now. But we still won't set CONFIG_STAGING by default.

Upstream is now working on R8188EU, so maybe a kernel release somewhere next year will have this fully integrated.

closed 2021-09-13 by jan-kiszka