siemens/meta-iot2050 #143: Add node-red-contrib-ibm-watson-iot  please

zhqingyong, 2021-06-09:
nice node！

jan-kiszka, 2021-06-09:
Mind to tell us in which ways it is nice? :wink: 

Seriously, some reasoning would be needed. Preinstalling nodes has some pros (e.g. showing users interesting things, not requiring them to be online with the device) and cons (longer list of dependencies, license checks, longer build times).

zhqingyong, 2021-06-09:
> Mind to tell us in which ways it is nice? 😉
> 
> Seriously, some reasoning would be needed. Preinstalling nodes has some pros (e.g. showing users interesting things, not requiring them to be online with the device) and cons (longer list of dependencies, license checks, longer build times).

yeah,I want to test this platform.But I cannot build the example image ,and I cannot install it in node-red.So, I had to ask for it to be built in, and it does a pretty good job.
best respect！

jan-kiszka, 2021-06-09:
OK, multiple issues:
- Rebuilding on own image is supposed to work, according to README. Report issues if not. Meanwhile, you can pick up images from the github Actions (artifacts).
- Installing additional nodes is also supposed to work, except when not (#123...). Will try to understand that soon.
- Preinstalling makes sense when the pros outweighs the cons for a specific component. Not yet clear to me if that is the case here.

zhqingyong, 2021-06-09:
Thanks.
Siemens also has official documentation on this node.

"Application example  02/2017
SIMATIC IOT2000 Connection
to IBM Watson IoT Platform
SIMATIC IOT2040"

best respect！

zhqingyong, 2021-06-10:
> OK, multiple issues:
>
> * Rebuilding on own image is supposed to work, according to README. Report issues if not. Meanwhile, you can pick up images from the github Actions (artifacts).
> * Installing [...]

I've solved this problem~^_^ 
I can install it on the node-red control panel.
Now,we can close it.

Best respect！

closed 2021-06-10 by jan-kiszka