siemens/meta-iot2050 #145: These problems occur when I use TTL to write firmware

zhqingyong, 2021-06-10:
When I start on SD，and run mkfs.ext4 /dev/mmcblk1 after.
And dd the image from my USB storage to emmc.
Then the problem arises
↓↓↓
2236976128 bytes (2.2 GB, 2.1 GiB) copied, 209.316 s, 10.7 MB/s
[ 595.589478] GPT:Primary header thinks Alt. header is not at the end of the di sk.
[ 595.600029] GPT:4369093 != 31080447
[ 595.603538] GPT:Alternate GPT header not at the end of the disk.
[ 595.609569] GPT:4369093 != 31080447
[ 595.613070] GPT: Use GNU Parted to correct GPT errors.
[ 595.618247] mmcblk1: p1

jan-kiszka, 2021-06-10:
You need to boot it once so that `expand-on-first-boot` could run. Or you need to update the partition table manually.

zhqingyong, 2021-06-10:
> You need to boot it once so that `expand-on-first-boot` could run. Or you need to update the partition table manually.

I reboot the device ，but the question still alive.
I can use the RJ45 port to write firmware.
I don‘t know what happened.

jan-kiszka, 2021-06-10:
There is new feature in the latest image builds which allow to install from an external medium onto the eMMC on first boot from that external device, see https://github.com/siemens/meta-iot2050#installing-the-image-on-the-emmc-iot2050-advanced-only. Once installed and booted from eMMC once, you will not see those GPT warnings anymore.

jan-kiszka, 2021-09-13:
New image version now available.

closed 2021-09-13 by jan-kiszka

jan-kiszka, 2022-03-04:
Which problem exactly?

kevin-G, 2022-03-04:
I'm using the new image, when I use dd to write the image to emmc, I got the below error:

2236976128 bytes (2.2 GB, 2.1 GiB) copied, 209.316 s, 10.7 MB/s
[ 595.589478] GPT:Primary header thinks Alt. header is not at the end of the di sk.
[ 595.600029] GPT:4369093 != 31080447
[ 595.603538] GPT:Alternate GPT header not at the end of the disk.
[ 595.609569] GPT:4369093 != 31080447
[ 595.613070] GPT: Use GNU Parted to correct GPT errors.
[ 595.618247] mmcblk1: p1

thanks

jan-kiszka, 2022-03-04:
What was your input to dd? Do you also get real error messages and boot failures?

kevin-G, 2022-03-04:
thanks for your reply! I was using below:

dd if=/home/IOT2050_Example_Image_V1.0.2.img of=/dev/mmcblk1 bs=100M conv=fsync status=progress 
or
dd if=/home/IOT2050_Example_Image_V1.0.2.img of=/dev/mmcblk1 bs=4M oflag=sync

no error message, just stop at below message, and after reboot can't enter the system. I was using two different images, one from Siemens download page, one was built from meta-iot2050, always the same:

2236976128 bytes (2.2 GB, 2.1 GiB) copied, 209.316 s, 10.7 MB/s
[ 595.589478] GPT:Primary header thinks Alt. header is not at the end of the disk.
[ 595.600029] GPT:4369093 != 31080447
[ 595.603538] GPT:Alternate GPT header not at the end of the disk.
[ 595.609569] GPT:4369093 != 31080447
[ 595.613070] GPT: Use GNU Parted to correct GPT errors.
[ 595.618247] mmcblk1: p1

BTW, with the built image, when power on, I was trying to press user button to flash to emmc, both USB and SD didn't work, I monitored by serial port, I can see it always boot from itself. is it only work for brand new IoT?

thanks again for your help!

bergmanu, 2022-03-07:
Hi, 
not sure whether this helps, but I once had the same issue and was able to solve it by running the command `echo 'w' | fdisk /dev/mmcblk1` from Example Image running on SD card. 
Can't really remember whether it immediately worked after this command or if I had to re-write the image to eMMc again.

kevin-G, 2022-03-07:
> 

Hi,
I just try your solution and it works, thanks a lot! 
now I can start to work on eMMC. 
thanks again for everyone's help!