siemens/meta-iot2050 #148: How to compile the debian bullseye image？

lxmuyu, 2021-06-23:
I compile the image and modify the iot2050-debian.conf
```
...
require  conf/distro/debian-bullseye.conf
...
```
and fail
```
ERROR: buildchroot-host-1.0-r0 do_rootfs_install: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_rootfs_install(d)
     0003:
File: '/work/isar/meta/classes/rootfs.bbclass', lineno: 182, function: do_rootfs_install
     0178:                                     shared=True)
     0179:
     0180:        bb.build.exec_func(cmd, d)
     0181:
 *** 0182:        if (d.getVarFlag(cmd, 'isar-apt-lock') or "") == "release-after":
     0183:            bb.utils.unlockfile(lock)
     0184:    progress_reporter.finish()
     0185:}
     0186:addtask rootfs_install before do_rootfs_postprocess after do_unpack
File: '/work/isar/bitbake/lib/bb/build.py', lineno: 251, function: exec_func
     0247:    with bb.utils.fileslocked(lockfiles):
     0248:        if ispython:
     0249:            exec_func_python(func, d, runfile, cwd=adir)
     0250:        else:
 *** 0251:            exec_func_shell(func, d, runfile, cwd=adir)
     0252:
     0253:    try:
     0254:        curcwd = os.getcwd()
     0255:    except:
File: '/work/isar/bitbake/lib/bb/build.py', lineno: 452, function: exec_func_shell
     0448:    with open(fifopath, 'r+b', buffering=0) as fifo:
     0449:        try:
     0450:            bb.debug(2, "Executing shell function %s" % func)
     0451:            with open(os.devnull, 'r+') as stdin, logfile:
 *** 0452:                bb.process.run(cmd, shell=False, stdin=stdin, log=logfile, extrafiles=[(fifo,readfifo)])
     0453:        finally:
     0454:            os.unlink(fifopath)
     0455:
     0456:    bb.debug(2, "Shell function %s finished" % func)
File: '/work/isar/bitbake/lib/bb/process.py', lineno: 182, function: run
     0178:        if not stderr is None:
     0179:            stderr = stderr.decode("utf-8")
     0180:
     0181:    if pipe.returncode != 0:
 *** 0182:        raise ExecutionError(cmd, pipe.returncode, stdout, stderr)
     0183:    return stdout, stderr
Exception: bb.process.ExecutionError: Execution of '/build/tmp/work/iot2050-debian-arm64/buildchroot-host/1.0-r0/temp/run.rootfs_install_pkgs_download.152' failed with exit code 100:
Reading package lists...
Building dependency tree...
Package debhelper is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libdebhelper-perl

Package make is a virtual package provided by:
  make-guile 4.3-4.1 [Not candidate version]

Package automake is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package devscripts is a virtual package provided by:
  devscripts:arm64 2.21.2 [Not candidate version]
[... 116 lines cut ...]
Package equivs is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package docbook-to-man is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'make' has no installation candidate
E: Package 'debhelper' has no installation candidate
E: Package 'autotools-dev' has no installation candidate
E: Package 'docbook-to-man' has no installation candidate
E: Package 'automake' has no installation candidate
E: Package 'devscripts' has no installation candidate
E: Package 'equivs' has no installation candidate
E: Package 'gcc' has no installation candidate
E: Package 'build-essential' has no installation candidate
WARNING: exit code 100 from a shell command.

ERROR: Logfile of failure stored in: /build/tmp/work/iot2050-debian-arm64/buildchroot-target/1.0-r0/temp/log.do_rootfs_install.151
ERROR: Task (/work/isar/meta/recipes-devtools/buildchroot/buildchroot-target.bb:do_rootfs_install) failed with exit code '1'
NOTE: Tasks Summary: Attempted 78 tasks of which 76 didn't need to be rerun and 2 failed.
```

how to fix it ?

jan-kiszka, 2021-06-23:
Good point, we should become ready for that. I've just hacked up this change:

```
diff --git a/conf/distro/iot2050-debian.conf b/conf/distro/iot2050-debian.conf
index 7b2be88..04ae721 100644
--- a/conf/distro/iot2050-debian.conf
+++ b/conf/distro/iot2050-debian.conf
@@ -1,14 +1,15 @@
 #
-# Copyright (c) Siemens AG, 2019
+# Copyright (c) Siemens AG, 2019-2021
 #
 # Authors:
 #  Le Jin <le.jin@siemens.com>
+#  Jan Kiszka <jan.kiszka@siemens.com>
 #
 # This file is subject to the terms and conditions of the MIT License.  See
 # COPYING.MIT file in the top-level directory.
 #
 
-require conf/distro/debian-buster.conf
+require conf/distro/debian-bullseye.conf
 
 DISTRO_NAME = "IOT2050 Debian System"
 
@@ -29,18 +30,6 @@ OVERRIDES .= ':${PACKAGES_SELECTION}'
 # packages-snapshot: Use the snapshot debian packages
 PACKAGES_SELECTION ?= "packages-mainline"
 
-DISTRO_APT_SOURCES = " \
-    conf/distro/debian-buster.list \
-    conf/distro/debian-buster-backports.list \
-    conf/distro/debian-bullseye.list \
-    "
-
-DISTRO_APT_PREFERENCES = "conf/distro/preferences.npm-backport"
-DISTRO_APT_PREFERENCES += "conf/distro/preferences.modemmanager-backport"
-
-HOST_DISTRO_APT_SOURCES = "${DISTRO_APT_SOURCES}"
-HOST_DISTRO_APT_PREFERENCES = "${DISTRO_APT_PREFERENCES}"
-
 DISTRO_APT_SOURCES_MAINLINE_LIST := "${DISTRO_APT_SOURCES}"
 DISTRO_APT_SOURCES_SNAPSHOT_LIST ?= "conf/distro/debian-snapshot.list"
 
@@ -53,9 +42,6 @@ HOST_DISTRO_APT_SOURCES_remove_packages-snapshot = "${DISTRO_APT_SOURCES_MAINLIN
 # Required as long as openssl is not rebuilt for the host arch
 HOST_DISTRO_APT_PREFERENCES += "conf/distro/preferences.upstream-libssl"
 
-# ModemManager buster-backports
-HOST_DISTRO_APT_PREFERENCES += "conf/distro/preferences.modemmanager-backport"
-
 SDK_INSTALL += "linux-headers-${KERNEL_NAME} mraa"
 SDK_PREINSTALL += "zlib1g-dev:${DISTRO_ARCH} libjson-c-dev:${DISTRO_ARCH}"
 
diff --git a/conf/machine/iot2050.conf b/conf/machine/iot2050.conf
index 0aed652..34ea821 100644
--- a/conf/machine/iot2050.conf
+++ b/conf/machine/iot2050.conf
@@ -10,8 +10,6 @@
 
 DISTRO_ARCH ?= "arm64"
 
-HOST_DISTRO ?= "debian-buster"
-
 PREFERRED_PROVIDER_u-boot-${MACHINE} = "u-boot-iot2050"
 
 KERNEL_NAME ?= "iot2050"
```

Unfortunately, its newer compiler breaks the kernel build for 4.19. Possibly, `:kas/opt/upstream.yml` will build. But that kernel will need a firmware update first which was not yet officially released. You could build and flash what we are currently testing (via `kas-iot2050-boot.yml`, but you should be ready for https://github.com/siemens/meta-iot2050/tree/master/recipes-bsp/u-boot#recovering-a-bricked-device in case something goes wrong.

Update: There are also changes to the mraa recipe needed, possibly to more. Will take some more effort that I can invest right now. Maybe that kernel build issue can also be fixed later. Will keep you posted on updates.

lxmuyu, 2021-06-23:
@jan-kiszka 
I just try the bullseye version to update the version of nodejs. Now I get a new method to update nodejs - apt perference. I'll try first. Thank you for reply.

jan-kiszka, 2021-06-27:
May not work if that means recompiling mraa against it. I've resolved related issues in https://github.com/siemens/meta-iot2050/commits/jan/bullseye.

lxmuyu, 2021-06-29:
@jan-kiszka 
we checkout the jan/bullseye branch, and get the image. Thank you for your hard work!

Webbeh, 2021-07-23:
[image]

When trying to compile this image. Seems like the eclipse/tcf.agent repo doesn't exist anymore.

jan-kiszka, 2021-07-23:
Just rebased that branch over master, should resolve it.

BaochengSu, 2021-09-03:
master now has been updated to bullseye, so close this issue.

If anything new found, please file an new issue.

closed 2021-09-03 by BaochengSu