siemens/meta-iot2050 #151: Node-Red 2.0 loses the IOT2050 GPIO nodes (+ other)

Webbeh, 2021-07-23:
Hi,

I was trying to make use of the latest node-red version : 2.0.2.

I first updated node to v14.17.2, which went without any issue.
After that, I tried to update node-red : `npm i -g --unsafe-perm node-red`. It gave a few node-gyp errors (weirdly) but did the installation nonetheless. 

Node-Red v2.0.2 is now running but I lost the IOT-specific MRAA nodes (they just won't appear in the palette). It seems `mraa doesn't even want to compile with `npm i -g mraa`, which I reckon may be why the nodes disappeared.

Furthermore, upon starting the node-red service, the other small errors happen (see at the bottom of this post).

This is on a new device with the RT image from [CI #159](https://github.com/siemens/meta-iot2050/actions/runs/1043064266). Nothing else has been done on the device.

```
Jul 23 06:23:14 iot2050-debian node-red[7687]: Welcome to Node-RED
Jul 23 06:23:14 iot2050-debian node-red[7687]: ===================
Jul 23 06:23:14 iot2050-debian node-red[7687]: 23 Jul 06:23:14 - [info] Node-RED version: v2.0.2
Jul 23 06:23:14 iot2050-debian node-red[7687]: 23 Jul 06:23:14 - [info] Node.js  version: v14.17.3
Jul 23 06:23:14 iot2050-debian node-red[7687]: 23 Jul 06:23:14 - [info] Linux 4.19.94-rt39 arm64 LE
Jul 23 06:23:16 iot2050-debian node-red[7687]: 23 Jul 06:23:16 - [info] Loading palette nodes
Jul 23 06:23:25 iot2050-debian node-red[7687]: 23 Jul 06:23:25 - [info] Dashboard version 2.27.0 started at /ui
Jul 23 06:23:26 iot2050-debian node-red[7687]: 23 Jul 06:23:26 - [warn] ------------------------------------------------------
Jul 23 06:23:26 iot2050-debian node-red[7687]: 23 Jul 06:23:26 - [warn] [node-red-node-serialport/serialport] Error: libnode.so.64: cannot open shared object file: No such file or directory
Jul 23 06:23:26 iot2050-debian node-red[7687]: 23 Jul 06:23:26 - [warn] [node-red-node-sqlite/sqlite] Error: Cannot find module '/root/.node-red/node_modules/node-red-node-sqlite/node_modules/sqlite3/lib/binding/node-v83-linux-arm64/node_sqlite3.node'
Jul 23 06:23:26 iot2050-debian node-red[7687]: Require stack:
Jul 23 06:23:26 iot2050-debian node-red[7687]: - /root/.node-red/node_modules/node-red-node-sqlite/node_modules/sqlite3/lib/sqlite3-binding.js
Jul 23 06:23:26 iot2050-debian node-red[7687]: - /root/.node-red/node_modules/node-red-node-sqlite/node_modules/sqlite3/lib/sqlite3.js
Jul 23 06:23:26 iot2050-debian node-red[7687]: - /root/.node-red/node_modules/node-red-node-sqlite/sqlite.js
Jul 23 06:23:26 iot2050-debian node-red[7687]: - /usr/lib/node_modules/node-red/node_modules/@node-red/registry/lib/loader.js
Jul 23 06:23:26 iot2050-debian node-red[7687]: - /usr/lib/node_modules/node-red/node_modules/@node-red/registry/lib/index.js
Jul 23 06:23:26 iot2050-debian node-red[7687]: - /usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/index.js
Jul 23 06:23:26 iot2050-debian node-red[7687]: - /usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/index.js
Jul 23 06:23:26 iot2050-debian node-red[7687]: - /usr/lib/node_modules/node-red/lib/red.js
Jul 23 06:23:26 iot2050-debian node-red[7687]: - /usr/lib/node_modules/node-red/red.js
Jul 23 06:23:26 iot2050-debian node-red[7687]: 23 Jul 06:23:26 - [warn] ------------------------------------------------------
Jul 23 06:23:26 iot2050-debian node-red[7687]: 23 Jul 06:23:26 - [info] Settings file  : /root/.node-red/settings.js
Jul 23 06:23:26 iot2050-debian node-red[7687]: 23 Jul 06:23:26 - [info] Context store  : 'default' [module=memory]
Jul 23 06:23:26 iot2050-debian node-red[7687]: 23 Jul 06:23:26 - [info] User directory : /root/.node-red
Jul 23 06:23:26 iot2050-debian node-red[7687]: 23 Jul 06:23:26 - [warn] Projects disabled : editorTheme.projects.enabled=false
Jul 23 06:23:26 iot2050-debian node-red[7687]: 23 Jul 06:23:26 - [warn] Flows file name not set. Generating name using hostname.
Jul 23 06:23:26 iot2050-debian node-red[7687]: 23 Jul 06:23:26 - [info] Flows file     : /root/.node-red/flows_iot2050-debian.json
Jul 23 06:23:26 iot2050-debian node-red[7687]: 23 Jul 06:23:26 - [info] Creating new flow file
Jul 23 06:23:26 iot2050-debian node-red[7687]: 23 Jul 06:23:26 - [warn]
Jul 23 06:23:26 iot2050-debian node-red[7687]: ---------------------------------------------------------------------
Jul 23 06:23:26 iot2050-debian node-red[7687]: Your flow credentials file is encrypted using a system-generated key.
Jul 23 06:23:26 iot2050-debian node-red[7687]: If the system-generated key is lost for any reason, your credentials
Jul 23 06:23:26 iot2050-debian node-red[7687]: file will not be recoverable, you will have to delete it and re-enter
Jul 23 06:23:26 iot2050-debian node-red[7687]: your credentials.
Jul 23 06:23:26 iot2050-debian node-red[7687]: You should set your own key using the 'credentialSecret' option in
Jul 23 06:23:26 iot2050-debian node-red[7687]: your settings file. Node-RED will then re-encrypt your credentials
Jul 23 06:23:26 iot2050-debian node-red[7687]: file using your chosen key the next time you deploy a change.
Jul 23 06:23:26 iot2050-debian node-red[7687]: ---------------------------------------------------------------------
Jul 23 06:23:26 iot2050-debian node-red[7687]: 23 Jul 06:23:26 - [info] Server now running at http://127.0.0.1:1880/
Jul 23 06:23:26 iot2050-debian node-red[7687]: 23 Jul 06:23:26 - [info] Starting flows
Jul 23 06:23:26 iot2050-debian node-red[7687]: 23 Jul 06:23:26 - [info] Started flows
```

jan-kiszka, 2021-07-23:
The manual update is working against the Debian "packaging" (npm stuff is not truly properly packagable with finite effort). In this case, the update overwrote the Debian-only installation of node-red-gpio, thus you lost those features.

However, I cannot tell what also corrupted `/root/.node-red/node_modules` here. That is unexpected. Maybe it is related to the manual update of NodeJS. You are completely off-track /wrt updates from the distribution in that case anyway. And also off-track /wrt what we tested (latest one was NodeJS 12 from bullseye).

Webbeh, 2021-07-23:
Which debian repo are you using to get the NodeJS 12 package from bullseye ? http://security.debian.org/bullseye/updates ?

I'll try another clean install again and see if it helps.

jan-kiszka, 2021-07-23:
https://github.com/siemens/meta-iot2050/commits/jan/bullseye, and that implies bullseye updates as well. But it also meant recompiling the nodes with native parts against that nodejs version - something you likely didn't do when just updating nodejs.

Webbeh, 2021-07-23:
Ohh literally just saw that there was a bullseye branch, must've skipped it somehow. Will give this a go.

bramroelandts, 2021-08-10:
@jan-kiszka I would also like to use NodeRED 2.0, requiring a newer node version. Installing a new node version with `n install lts` works, but results in segfaults from NodeRED, probably because of reasons you mentioned in this thread.

What's the expected timeline on the bullseye branch? Is it currently stable and ready for us to compile and use? Are there incomplete parts? When will bullseye get merged into the master branch? Are there any plans to upgrade to NodeRED 2.0 right from the compilation environment, or should we stick with updating it on-target?

jan-kiszka, 2021-08-10:
Final tests for the next release are running. This one will be on buster again. For the next-but-one, bullseye is a likely target. I can't tell for sure when that will happen, maybe around autumn or winter (this year). And that will surely be a chance to update Node RED to its latest version as well.

For the time being, I would recommend trying an integration via meta-iot2050 (or an own layer on top), pulling in the bullseye branch and deviating in recipe versions where needed. That way you will continue to get the GPIO node preinstallation.

Webbeh, 2022-01-13:
Alright so I got everything working with node-red 2.1.4, still on nodeJS 12.

The intel-gpio seem to function properly, set aside the fact that the digital input don't publish their state upon start of a flow.

I think this can be closed. However, it'd be great if we could do something about the npm mraa node not being able to compile with node > 12...

jan-kiszka, 2022-01-13:
Please open a new issue regarding NodeJS > 12 support. However, we may not have the time to look into that soon, specifically as not even Debian bookworm has moved beyond 12 so far, and that would complicate already the development setup.

closed 2022-01-13 by jan-kiszka