siemens/meta-iot2050 #156: Query last DI status

Webbeh, 2021-08-02:
We have an usecase where we need to know the status of a Digital Input at any given time.

For example, we can't get the status of the digital input after a restart of Node-Red / reboot of the device.

It would be good if we could poll it every now and then or, at least, have it push its status upon restart of node-red.

Thank you very much !

jan-kiszka, 2021-08-02:
IIRC, there is a way in Node RED to generate an even on startup. This can then trigger the initial reading of the inputs.

Webbeh, 2021-08-02:
Do you have any idea how to implement that for our usecase ? :)

jan-kiszka, 2021-08-02:
I was confused: We are actually missing https://github.com/siemens/meta-iot2000/blob/master/meta-iot2000-example/recipes-node/node-red-node-intel-gpio/files/0003-intel-gpio-Add-support-for-initial-message-from-digi.patch in this layer. Happy to take a PR that pulls this over.

PR #212 (2021-11-12): node-red-gpio: port DIN's initial message from iot2000
Port a part of siemens/meta-iot2000@5fc2bbed commit related to DIN's
initial message support.

Fixes #156
commit f374dfba (2021-11-12): node-red-gpio: port DIN's initial message from iot2000

Port a part of siemens/meta-iot2000@5fc2bbed commit related to DIN's
initial message support.

Fixes #156
closed 2021-11-12 by BaochengSu