siemens/meta-iot2050 #161: After building example Ubuntu wic.img it appears that it has no kernel image in it

omersert, 2021-08-25:
Hello, I checked out to jan/ubuntu branch and tried to create the Example Ubuntu image with the command below
 `./kas-docker build kas-iot2050-example.yml:kas/opt/ubuntu.yml`  
In the end I got a wic.img. After opening that archive the root filesystem was okay but couldn't see the kernel image(vmlinuz) and initrd. Here is what the archive looks like :

[image]

I cannot boot this image for IOT2050 Advanced I changed the boot order of uboot and gave priority to usb.  What might cause that?

Best regards.
Ömer

jan-kiszka, 2021-08-25:
The kernel is under /boot and will be picked up by the boot loader (U-Boot) from there.

I didn't try to boot the generated image myself recently, maybe something broke. Do you have a boot log taken from the debug UART?

omersert, 2021-08-25:
Since I don't have uart connection cables yet I dont't have the logs. But  as soon as I acquire the required equipments I will share here.

Thanks for the quick response.