siemens/meta-iot2050 #166: Defined GPIO behavior during boot

BrotKolben, 2021-09-01:
While booting up the IOT2050, configured digital outputs might "flicker" between LOW and HIGH, and not show a defined state. Configuration in iot2050setup does not prevent this from happening: 
[image]
The flickering occurs up-until the device is fully booted up. 
Having a default LOW state during boot would be a desired requirement from safety perspective.

jan-kiszka, 2021-09-01:
I think we discussed this somewhere already: Without consistent changes across the device trees (firmware and kernel), the safe state of GPIOs is "input" (high impedance). I think this is what we have today.

jan-kiszka, 2021-09-13:
In fact, that may even be unsolvable via firmware patches, because before the firmware comes power-on, and that defines the GPIO mode in hardware, not by any software bit. @BaochengSu, am I right?

BaochengSu, 2021-09-13:
@jan-kiszka I think so, the GPIO pin has a initail mode before change it in the spl, that initail mode could not be configure to output low or high, even could not be a GPIO at all.

Anyway, I am contacting with TI regarding this topic.

BaochengSu, 2021-09-13:
Checked w/ TI, confirmed that our understand is correct.

jan-kiszka, 2021-10-18:
Let's close this as "not possible", unfortunately.

closed 2021-10-18 by jan-kiszka