siemens/meta-iot2050 #180: Yocto u-boot

MickaelDelanoe, 2021-09-09:
Hi,
I know this meta is not for Yocto Project, but may be you can help me anyway.

I try to build a personal OS image for IOT2050 with Yocto Project.

I want to modify my personal existing meta developed for an IoT ARM V7.
So, I try to set it for IOT2050 hardware : ARM TI AM6548 HS

So far, I added a new conf file "iot2050.conf" as follow and a `u-boot-iot2050.bb` file.
My `u-boot-iot2050.bb` is base on `u-boot-ti-mainline_git.bb` (http://git.yoctoproject.org/cgit/cgit.cgi/meta-ti/tree/recipes-bsp/u-boot/u-boot-ti-mainline_git.bb)

__iot2050.conf__
```
include conf/machine/include/tune-cortexa5.inc
include conf/machine/include/am65xx.inc
include conf/machine/opwee-common.inc

SOC_FAMILY = "am65xx:iot2050"

DISTRO_ARCH ?= "arm64"
TARGET_ARCH="arm"

PREFERRED_PROVIDER_virtual/bootloader = "u-boot-iot2050"
PREFERRED_PROVIDER_u-boot = "u-boot-iot2050"

EXTRA_IMAGEDEPENDS="u-boot dosfstools-native mtools-native"
# UBOOT_MAKE_TARGET = "u-boot-ti.inc"
UBOOT_CONFIG ??= "spi"
UBOOT_CONFIG[spi] = "am65x_evm_a53_defconfig"

KERNEL_NAME ?= "iot2050"

IMAGE_TYPE ?= "wic-img"
WKS_FILE ?= "sdimage-iot2050"
```

But I would like to know, which value can be set for `UBOOT_MAKE_TARGET` ?

Can you help me in this issue ?

Best regards,

jan-kiszka, 2021-09-09:
That should be empty (if that works) or `flash.bin`.

jan-kiszka, 2021-09-09:
At least with the U-Boot version just merged into master today...

MickaelDelanoe, 2021-09-10:
Yes, sorry I mix up myself
When I left it empty, I got an error for `checkgcc6` during the makefile. and I thought it was related to UBOOT_MAKE_TARGET according to old research. Anyway, I got lost in my issues.
Now I'm stuck because I'm using OpenSUSE 42.3, for cross-compilation, and it doesn't support gcc6

jan-kiszka, 2021-09-10:
Why gcc6? We are building with gcc10 by now, without issues.

MickaelDelanoe, 2021-09-10:
Well, I think is because I use diectly https://source.denx.de/u-boot/u-boot/ and note `meta-iot2050/recipes-bsp/u-boot/u-boot-iot2050_2021.04.bb`

jan-kiszka, 2021-09-10:
We are working on upstreaming everything but we are not yet there. The staging branch for U-Boot master is https://github.com/siemens/u-boot/commits/jan/iot2050. Following https://github.com/siemens/u-boot/blob/jan/iot2050/doc/board/siemens/iot2050.rst, you should be able to rebuild things independently of Yocto or our recipes, and then migrating those steps over might be simpler.

Before trying anything out on the device, make sure to have a flash programmer with you! Too many pieces, and one is always wrong on first try - that's at least my experience.

BaochengSu, 2021-09-13:
TI has an [`arago project`](http://arago-project.org/wiki/index.php/Main_Page) which is a yocto based project and used by TI to build their own Linux SDK, that could be a useful reference for you I think.

closed 2021-11-23 by MickaelDelanoe