siemens/meta-iot2050 #215: Add new layer and recipes

MickaelDelanoe, 2021-11-16:
Hello,

I would like to add a new Layer (and recipes) to the existing project.
I would like add `meta-iot-cloud` layer (https://layers.openembedded.org/layerindex/branch/krogoth/layer/meta-iot-cloud/) and a personnel recipes `python-aws-iot-devices-sdk_%.bbappend`

So, I tried to modify `kas/iot2050.yml` to add layer : 

```yml
  isar:
    url: https://github.com/ilbers/isar
    refspec: 03124cca669f50b682336a0bdf4ede5a4238e144
    layers:
      meta: meta-iot-cloud
    patches:
```

And, I updated the cloned project (v01.01.01) by:
 1. Adding a recipes `recipes-aws` in the clonned project.
 2. Adding `python-aws-iot-devices-sdk_%.bbappend` to `IMAGE_INSTALL`

But I got the error : 
```
```

Do you know, how can I add `meta-iot-cloud` layer ?

Best Regards,

jan-kiszka, 2021-11-16:
Note that meta-iot2050 is not Yocto/OE layer.

I'm not aware of prebuilt debian packages for the AWS IoT Device SDK, so you likely have to remodel the required Yocto recipes into Debian package rules and/or Isar recipes. That can be simple for simple builds (demo/example: https://github.com/ilbers/isar/blob/master/meta-isar/recipes-app/samefile/samefile_2.14.bb).

MickaelDelanoe, 2021-11-23:
According previous exemples, I tried the simple following recipe who seem works...

```
SRC_URI = " \
    git://github.com/aws/aws-iot-device-sdk-python.git;protocol=https \
    "

SRCREV = "c92908a125bf3ed3b28b33b80c9b10b65fa6d34c"

S = "${WORKDIR}/git"

DEBIAN_BUILD_DEPENDS += "python-dev, python-distutils-extra, openssl"
DEBIAN_DEPENDS = "python3"

inherit dpkg

do_prepare_build[cleandirs] += "${S}/debian"

do_prepare_build() {
    deb_debianize
}
```

`AWSIoTPythonSDK` seem well installed on the image after build, burn, boot and import in a debug script on the IOT2050 ...

I will see the real implementation when I would have succeded to add my personnal Python development in the image.

closed 2021-11-23 by MickaelDelanoe