siemens/meta-iot2050 #219: Build issues using Ubuntu 18.04

LongHairedHacker, 2021-11-19:
Hi,

I'm encountered an issue to build an image using Ubuntu 18.04 (`Linux PL911 5.4.0-90-generic #101~18.04.1-Ubuntu SMP Fri Oct 22 09:25:04 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`) as the host system.
I've already verified that it works under Ubuntu 20.04, but for unrelated reasons I'll have to stick with 18.04 for my workstation for the short term future.

So I decided to investigate the issue, since there is no obvious reason it shouldn't work. 
Ubuntu 18.04 has support for binfmt and most of the build process happens inside a container anyway.

Here's the full output of running `./kas-container build kas-iot2050-example.yml`:
[build.log](https://github.com/siemens/meta-iot2050/files/7569108/build.log)

It ends with:
```
W: Failure trying to run: chroot "/build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs" /bin/true
W: See /build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs/debootstrap/debootstrap.log for details
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/temp/log.do_bootstrap.302
ERROR: Task (/work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-target.bb:do_bootstrap) failed with exit code '1'
```
[log.do_bootstrap.302](https://github.com/siemens/meta-iot2050/files/7569210/log.do_bootstrap.txt) does not have any additional information.

[debootstrap.log](https://github.com/siemens/meta-iot2050/files/7569366/debootstrap.log) is a little more useful:
```
chroot: failed to run command ‘/bin/true’: No such file or directory
```

If I try to run the same chroot directly I get a similar error:
```
sudo chroot "./build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs" /bin/true
chroot: failed to run command ‘/bin/true’: No such file or directory
```

However the `true` binary clearly exists at that location:
```
ls -alh ./build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs/bin/true
-rwxr-xr-x 1 root root 31K Sep 22  2020 ./build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs/bin/true
```
After some poking around with strace I noticed that `No such file or directory` is a little misleading.
The file that's missing is `/usr/bin/qemu-aarch64-static`, which of course does not exist in the chroot environment.

```
$ update-binfmts --display
....
qemu-aarch64 (enabled):
     package = qemu-user-static
        type = magic
      offset = 0
       magic = \x7f\x45\x4c\x46\x02\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\xb7\x00
        mask = \xff\xff\xff\xff\xff\xff\xff\x00\xff\xff\xff\xff\xff\xff\xff\xff\xfe\xff\xff\xff
 interpreter = /usr/bin/qemu-aarch64-static
    detector =
...
$ sudo cp /usr/bin/qemu-aarch64-static ./build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/bin/
$ sudo chroot "./build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs" /bin/true
$ sudo chroot "./build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs" /bin/true && echo it works
it works
$ sudo chroot "./build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs" /bin/ls
bin  boot  debootstrap	dev  etc  home	lib  proc  root  run  sbin  sys  tmp  usr  var
```

Ist that a bug, a known issue with my older Ubuntu or something that's broken on my machine specifically?

jan-kiszka, 2021-11-19:
`qemu-aarch64-static` comes from the kas-isar container that is abstracting away the host OS (as far as possible).

The question is why that is not used / correctly configured when `kas-container` starts the kas-isar container. That might be due to limitation of the docker version / configuration in Ubuntu 18.04.

What is the binfmts configuration after a fresh Ubuntu host boot inside the kas-isar container? Run `./kas-container shell kas-iot2050-example.yml` to get there.

LongHairedHacker, 2021-11-19:
Thanks for the quick response.

```
$ ./kas-container shell kas-iot2050-example.yml
2021-11-19 13:05:06 - INFO     - kas 2.5 started
2021-11-19 13:05:06 - INFO     - /repo$ git rev-parse --show-toplevel
2021-11-19 13:05:06 - INFO     - /repo$ git rev-parse --show-toplevel
2021-11-19 13:05:06 - INFO     - Using /repo as root for repository meta-iot2050
2021-11-19 13:05:06 - INFO     - /work/isar$ git remote set-url origin https://github.com/ilbers/isar
2021-11-19 13:05:06 - INFO     - /work/isar$ git cat-file -t e274130b870b7e31532fe191b9932cde1d819b4b
2021-11-19 13:05:06 - INFO     - Repository isar already contains e274130b870b7e31532fe191b9932cde1d819b4b as commit
2021-11-19 13:05:06 - INFO     - /repo$ git rev-parse --show-toplevel
2021-11-19 13:05:06 - INFO     - Using /repo as root for repository meta-iot2050
2021-11-19 13:05:06 - INFO     - /work/isar$ git status -s
2021-11-19 13:05:06 - INFO     - /work/isar$ git rev-parse --verify -q origin/e274130b870b7e31532fe191b9932cde1d819b4b
2021-11-19 13:05:06 - INFO     - /work/isar$ git checkout -q e274130b870b7e31532fe191b9932cde1d819b4b
2021-11-19 13:05:06 - INFO     - /repo$ git rev-parse --show-toplevel
2021-11-19 13:05:06 - INFO     - Using /repo as root for repository meta-iot2050
2021-11-19 13:05:06 - INFO     - /work/isar$ git checkout -q -B patched-e274130b870b7e31532fe191b9932cde1d819b4b
2021-11-19 13:05:06 - INFO     - /work/isar$ git apply /repo/isar-patches/0001-u-boot-Exclude-libubootenv.a-from-dh_strip.patch
2021-11-19 13:05:06 - INFO     - Patch applied. (patch path: /repo/isar-patches/0001-u-boot-Exclude-libubootenv.a-from-dh_strip.patch, repo: isar, patch entry: u-boot-fix)
2021-11-19 13:05:06 - INFO     - /work/isar$ git add -A
2021-11-19 13:05:06 - INFO     - /work/isar$ git commit -a --author kas <kas@example.com> -m msg
2021-11-19 13:05:06 - INFO     - [patched-e274130b870b7e31532fe191b9932cde1d819b4b d44f024] msg
2021-11-19 13:05:06 - INFO     - Author: kas <kas@example.com>
2021-11-19 13:05:06 - INFO     - 1 file changed, 3 insertions(+)
2021-11-19 13:05:06 - INFO     - /repo$ git rev-parse --show-toplevel
2021-11-19 13:05:06 - INFO     - Using /repo as root for repository meta-iot2050
2021-11-19 13:05:06 - INFO     - /work/isar$ /tmp/tmpd4d0d7s9/get_bb_env /build
2021-11-19 13:05:06 - INFO     - /repo$ git rev-parse --show-toplevel
2021-11-19 13:05:06 - INFO     - Using /repo as root for repository meta-iot2050
builder@0969009e3296:/build$ update-binfmts --display
python2.7 (enabled):
     package = python2.7
        type = magic
      offset = 0
       magic = \x03\xf3\x0d\x0a
        mask =
 interpreter = /usr/bin/python2.7
    detector =
python3.7 (enabled):
     package = python3.7
        type = magic
      offset = 0
       magic = \x42\x0d\x0d\x0a
        mask =
 interpreter = /usr/bin/python3.7
    detector =
qemu-aarch64 (enabled):
     package = qemu-user-static
        type = magic
      offset = 0
       magic = \x7f\x45\x4c\x46\x02\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\xb7\x00
        mask = \xff\xff\xff\xff\xff\xff\xff\x00\xff\xff\xff\xff\xff\xff\xff\xff\xfe\xff\xff\xff
 interpreter = /usr/libexec/qemu-binfmt/aarch64-binfmt-P
    detector =
qemu-alpha (enabled):
     package = qemu-user-static
        type = magic
      offset = 0
       magic = \x7f\x45\x4c\x46\x02\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x26\x90
        mask = \xff\xff\xff\xff\xff\xff\xff\x00\xff\xff\xff\xff\xff\xff\xff\xff\xfe\xff\xff\xff
 interpreter = /usr/libexec/qemu-binfmt/alpha-binfmt-P
    detector =
qemu-arm (enabled):
     package = qemu-user-static
        type = magic
      offset = 0
       magic = \x7f\x45\x4c\x46\x01\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x28\x00
        mask = \xff\xff\xff\xff\xff\xff\xff\x00\xff\xff\xff\xff\xff\xff\xff\xff\xfe\xff\xff\xff
[... 178 lines cut ...]
qemu-xtensa (enabled):
     package = qemu-user-static
        type = magic
      offset = 0
       magic = \x7f\x45\x4c\x46\x01\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x5e\x00
        mask = \xff\xff\xff\xff\xff\xff\xff\x00\xff\xff\xff\xff\xff\xff\xff\xff\xfe\xff\xff\xff
 interpreter = /usr/libexec/qemu-binfmt/xtensa-binfmt-P
    detector =
qemu-xtensaeb (enabled):
     package = qemu-user-static
        type = magic
      offset = 0
       magic = \x7f\x45\x4c\x46\x01\x02\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x5e
        mask = \xff\xff\xff\xff\xff\xff\xff\x00\xff\xff\xff\xff\xff\xff\xff\xff\xff\xfe\xff\xff
 interpreter = /usr/libexec/qemu-binfmt/xtensaeb-binfmt-P
    detector =
 ```
 
 That looks like it is indeed using the qemu binaries from inside the container.

In the mean time I found another Ubuntu 18.04 host to test it on. The regular build command works there.
The other machine does not have `qemu-user-static` installed.
So I uninstalled the `qemu-user-static`package my original machine and and now it works there on as well.
To avoid confusion: The ` update-binfmts --display` was done with the package still installed. So that's still the broken state.
I'll do one after uninstalling for comparison later.

jan-kiszka, 2021-11-19:
OK, that makes "sense": The problem with current binfmt_misc support in the kernel is that it does not support containers. We hack around that by giving the container the task to program what it and also the host should use from then on. However, what you program first, becomes sticky. And now, when the host programs something before kas-container was run, you will get qemu-user-static configuration and versions from the host. That is usually incompatible with host being older than container.

There is hope (kernel patches to make binfmt_misc container-compatible), but those will take a couple of years to hit all mainline distributions.

LongHairedHacker, 2021-11-19:
Ah okay. That makes sense.

I also found a less invasive workaround:
If I do `sudo update-binfmts --disable qemu-aarch64` on the host, before running the container, everything works, even with the package installed.
Seems like the container can overwrite disable settings.

jan-kiszka, 2022-01-07:
Closing as we have a workaround.

closed 2022-01-07 by jan-kiszka