siemens/meta-iot2050 #220: UART login trouble

MickaelDelanoe, 2021-11-22:
Hello,

After build and burned my image from v01.01.01, i can't login to the IoT2050.
After well boot, when I got the login instruction, nothing happened when I dial root, and a "new boot line" appear after few seconds

```
[   38.408332] IPv6: ADDRCONF(NETDEV_UP): eno2: link is not ready

Debian GNU/Linux 10 iot2050-debian ttyS3

iot2050-debian login: [   68.130407] random: crng init done
[   68.133827] random: 6 urandom warning(s) missed due to ratelimiting

```

Do you know what can be the trouble ?

Best Regards,

jan-kiszka, 2021-11-22:
No issues known so far. Can you log in via SSH? Does the official binary image show the same problem?

MickaelDelanoe, 2021-11-22:
Yes, when I built v01.01.01 exemple image I got the same error (I will try with the download image from siemens website)

For the SSH conexion I got `Permission denied (publickey,password).` when I try `ssh -o strictHostKeyChecking=no root@192.168.200.1`

MickaelDelanoe, 2021-11-22:
Ok, It seem that the trouble come from my VirtualBox (Debian) on my first Mac.

- Login works fine with VM on an other Mac
- Login works fine with `screen` utility on first Mac

closed 2021-11-22 by MickaelDelanoe