siemens/meta-iot2050 #221: Cloning : Host key verification failed

MickaelDelanoe, 2021-11-22:
Hello,

I want adding a git project with a personal recipes.

```
...
SRCREV = "87782cad64659b70dcd8422545a54f3ddfbaae6e"
SRC_URI += "git://git@git.aquassay.net/tools/opwee.git;protocol=ssh"
LICENSE = "CLOSED"
...
```
As I want clone the repo by SSH, so I generated and do the necessary to able to clone the repo.
But I encounter an error during the build : 

```
```

I do not understand why the key verification enter in trouble, because I tried to clone manually my git repo into `build/download/test` and it works well.

I tried to increase the `id_rsa` right, but not effects ...

Do you have an idea ?

Best Regards,

jan-kiszka, 2021-11-22:
You need to pass in a private key into the build env, and that will also skip host-key validation. See https://kas.readthedocs.io/en/latest/command-line.html, looking for `SSH_PRIVATE_KEY`.

Or pass a complete .ssh folder into build env, likely a filter one that only contains the needed keys, but also the right known_hosts file. That is achieved via `--ssh-dir` as kas command line option.

MickaelDelanoe, 2021-11-23:
I used `./kas-container --ssh-dir /home/... build ...` and it's work well.
Thank you.

closed 2021-11-23 by MickaelDelanoe